cantools/cantools #271: database as_dbc_string() TypeError using bitstruct.c

lhh31, 2021-02-05:
**Python Version:** 3.8.5
**Cantools Version** 34.2.0+
**OS:** Windows 10 (1909)

This works fine in cantools version 34.1.0. From the commits it appears there was a change from using bistruct to using bitstruct.c. Reverting this change in 34.2.0 does fix the issue. 

Further analysis shows bitstruct was at version 8.10.0, updating to 8.11.1 can also resolve the issue. 

I am no expert but I think this means that the setup.py needs to have the required bitstruct version upped?

**Test code**
```
import cantools
dbc = cantools.database.load_file("vehicle.dbc",strict=False)
cantools.database.dump_file(dbc,"vehicle2.dbc")
```

**Error Messages**
```
  File "c:\anaconda3\envs\py38-logger-tool\lib\site-packages\cantools\database\__init__.py", line 216, in dump_file
    output = database.as_dbc_string()
  File "c:\anaconda3\envs\py38-logger-tool\lib\site-packages\cantools\database\can\database.py", line 279, in as_dbc_string
    return dbc.dump_string(InternalDatabase(self._messages,
  File "c:\anaconda3\envs\py38-logger-tool\lib\site-packages\cantools\database\can\formats\dbc.py", line 1524, in dump_string
    database = deepcopy(database)
  File "c:\anaconda3\envs\py38-logger-tool\lib\copy.py", line 172, in deepcopy
    y = _reconstruct(x, memo, *rv)
  File "c:\anaconda3\envs\py38-logger-tool\lib\copy.py", line 270, in _reconstruct
    state = deepcopy(state, memo)
  File "c:\anaconda3\envs\py38-logger-tool\lib\copy.py", line 146, in deepcopy
    y = copier(x, memo)
  File "c:\anaconda3\envs\py38-logger-tool\lib\copy.py", line 230, in _deepcopy_dict
    y[deepcopy(key, memo)] = deepcopy(value, memo)
  File "c:\anaconda3\envs\py38-logger-tool\lib\copy.py", line 146, in deepcopy
    y = copier(x, memo)
  File "c:\anaconda3\envs\py38-logger-tool\lib\copy.py", line 205, in _deepcopy_list
    append(deepcopy(a, memo))
  File "c:\anaconda3\envs\py38-logger-tool\lib\copy.py", line 172, in deepcopy
    y = _reconstruct(x, memo, *rv)
  File "c:\anaconda3\envs\py38-logger-tool\lib\copy.py", line 270, in _reconstruct
    state = deepcopy(state, memo)
  File "c:\anaconda3\envs\py38-logger-tool\lib\copy.py", line 146, in deepcopy
    y = copier(x, memo)
  File "c:\anaconda3\envs\py38-logger-tool\lib\copy.py", line 230, in _deepcopy_dict
    y[deepcopy(key, memo)] = deepcopy(value, memo)
  File "c:\anaconda3\envs\py38-logger-tool\lib\copy.py", line 146, in deepcopy
    y = copier(x, memo)
  File "c:\anaconda3\envs\py38-logger-tool\lib\copy.py", line 230, in _deepcopy_dict
    y[deepcopy(key, memo)] = deepcopy(value, memo)
  File "c:\anaconda3\envs\py38-logger-tool\lib\copy.py", line 172, in deepcopy
    y = _reconstruct(x, memo, *rv)
  File "c:\anaconda3\envs\py38-logger-tool\lib\copy.py", line 264, in _reconstruct
    y = func(*args)
  File "c:\anaconda3\envs\py38-logger-tool\lib\copy.py", line 263, in <genexpr>
    args = (deepcopy(arg, memo) for arg in args)
  File "c:\anaconda3\envs\py38-logger-tool\lib\copy.py", line 161, in deepcopy
    rv = reductor(4)
TypeError: cannot pickle 'bitstruct.c.CompiledFormatDict' object
```

JulianWgs, 2021-08-17:
Hello,

I'm having the same issue as I want to use the decode function with multiprocessing for [my project](https://pypi.org/project/CANdas/). The pickling does not work.

It's this [commit](https://github.com/eerimoq/cantools/commit/c6eb73aa030b9e41ba9f2a60751c7f6ddabd45f5#diff-c561bfbab50738f2b78901f39f232bf95dba39e044414bdf6691fab9a3fd967eR6) which introduces the error, especially [these lines](https://github.com/eerimoq/cantools/commit/c6eb73aa030b9e41ba9f2a60751c7f6ddabd45f5#diff-c561bfbab50738f2b78901f39f232bf95dba39e044414bdf6691fab9a3fd967eR6-R10).

For a quick fix I would propose to make it configurable which version is imported (bitstruct vs bitstruct.c). In the current code if it is possible bitstruct.c is always imported.

A long term solution would be to make bitstruct.c pickleable, but I dont know how. Is this possible?

@eerimoq Do you need help implementing any of the changes?

Best regards
Julian

eerimoq, 2021-08-17:
@andlaus is the new maintainer, and I'm sure he appreciates all help he can get.

andlaus, 2021-08-17:
yes, help is always appreciated :)

I guess that in this concrete case, I might need to come back to you if changes to the bistruct or bitstruct.c turn out to be the best solution. My current naive view is that bitstruct.c should be replaced by bitstruct if the performance impact is negligible, or if it is not, all bitstruct.c objects should become pickleable. Let's see, it should not be too hard. (Famous last words.)

JulianWgs, 2021-08-18:
### Pickling support
It is possible to make c extensions pickleable. See more information [here](https://stackoverflow.com/questions/12530333/python-pickling-c-objects). The link is wrong in the post, but the edit is too short to fix it. [Here](https://docs.python.org/3/c-api/structures.html#c.PyMethodDef) is the correct link to PyMethodDef. Unfortunately I still dont quite understand the way forward with this as I never have written a c extension in Python. May be [this code](https://github.com/eerimoq/bitstruct/blob/master/bitstruct/c.c#L230-L239) is relevant?

### Benchmarking

I've just benchmarked the first test case of bitstruct and the c implementation is much faster:
```python
%timeit packed = pack('u1u1s6u7u9', 0, 0, -2, 65, 22)

# bitstruct.c
# 239 ns ± 1.91 ns per loop (mean ± std. dev. of 7 runs, 1000000 loops each)

# bitstruct
# 18 µs ± 336 ns per loop (mean ± std. dev. of 7 runs, 100000 loops each)
```

Also looking at test, but probably more suited for another issue: cantools and bitstruct would be greatly suitable for [hypothesis](https://hypothesis.readthedocs.io/en/latest/).

This is a very low level test, here is a more practical benchmark.

db = cantools.database.Database()
db.add_dbc_string("""
BO_ 257 TestMessage: 6 Vector__XXX
 SG_ TestSignal1 : 16|16@1- (1,0) [0|0] "" Vector__XXX
 SG_ TestSignal2 : 32|16@1- (1,0) [0|0] "" Vector__XXX
 SG_ TestSignal3 : 0|16@1- (1,0) [0|0] "" Vector__XXX
""")

andlaus, 2021-08-25:
hm, this is not exactly what I meant: I am aware that, on its own, `bitstruct.c` is much faster than `bitstruct`, but I was wondering how much impact this has on a holistic level. For this purpose, I've modified the `decode` subparser to measure the relevant timings plus disabling the terminal output and then set it to work on a trace containing ~85000 frames using an ARXML database. Here are the performance results:
```
# bitstruct.c
Setup time: 8.41
Decode time: 2.01

# bitstruct
Setup time: 8.53
Decode time: 3.07
```

The "Setup time" is the time required to parse the database, while "Decode time" is the time required to decode the raw CAN frames via `database.decode_message()` (as you can see from the timing, the database is quite large). From the results above, I conclude that `bitstruct.c` indeed has a significant impact on decoding performance, but also that the regular `bitstruct` module seems to be "fast enough" in most cases, in particular when considering scenarios that involve a "live" bus instead of decoding a trace as fast as possible (the trace used here captures about two minutes of an almost fully loaded 500kbps bus).

I guess that I will give making `bitstruct.c` codecs pickleable a shot in the future anyway, but I will not press very hard on this front...

leunark, 2021-11-16:
I'm having the same problem getting the "can't pickle bitstruct.c.CompiledFormatDict objects" exception when trying to use the cache for faster loading. Also I have huge blfs with a lot of messages that need to be decoded. By utilising multiprocessing the dbc object has to be passed as a parameter but since it is not fully pickable with the bitstruct.c integration, I can't utilise the benefits of multiprocessing. What I've done so far is checking out the repo, and enforcing the use of bitstruct instead of bitstruct.c by reversing [this ](https://github.com/cantools/cantools/commit/c6eb73aa030b9e41ba9f2a60751c7f6ddabd45f5#diff-c561bfbab50738f2b78901f39f232bf95dba39e044414bdf6691fab9a3fd967eR6) commit.

I'm really relying on the dbc object being picklable because although the bistruct.c is faster, with bitstruct the blf files I'm using can be decoded 50% faster. Can we just make the use of bitstruct.c or bitstruct optional?

andlaus, 2021-11-17:
> Can we just make the use of bitstruct.c or bitstruct optional?

bitstruct is pretty essential w.r.t. encoding I agree pickleability is more important than performance, i.e. the default should be switched to `bitstruct`. Mind opening a PR to that end?

leunark, 2021-11-17:
> Mind opening a PR to that end?

Yes, I can do that tomorrow.

eerimoq, 2021-11-18:
If there is a problem with bitstruct, just open an issue on that project explaining the issue.

https://github.com/eerimoq/bitstruct

andlaus, 2021-11-18:
done: https://github.com/eerimoq/bitstruct/issues/24

leunark, 2021-11-18:
Is it then still needed to open a PR to switch the default to `bistruct`? If the pickleable issue for `bitstruct.c` gets fixed soon, this will just lead to switching the default back to `bitstruct.c` with an update of `bitstruct`.
Also, maybe it is smart to integrate a test to check if `can.Database` remains pickleable?

andlaus, 2021-11-18:
> If the pickleable issue for bitstruct.c gets fixed soon, this will just lead to switching the default back to bitstruct.c after updating it.

yes. This hopefully will not cause us any headaches, though...

> Also, maybe it is smart to integrate a test to check if can.Database remains pickleable?

I agree. But for this, it first needs to become pickleable ;)

@eerimoq: Do you intend to fix this issue or shall I give it a go. (the latter will probably take substantially longer.)

eerimoq, 2021-11-19:
@andlaus I'll give it a try during the weekend, so no need for you to work on it right now.

andlaus, 2021-11-22:
since  eerimoq/bitstruct#24, should have addressed that issue, I'm closing it. feel free to re-open if the problem still persists...

closed 2021-11-22 by andlaus

juleq, 2021-11-22:
@andlaus Is the revision of bitstruct pinned or will it update on the next cantools release?

andlaus, 2021-11-22:
requirements.txt says it is unpinned. could you tag a new cantools release? this issue seemed to have affected more people than usual...

juleq, 2021-11-22:
Have released. Has already propagated through for my by now.

juleq, 2021-11-22:
bitstruct did not update when doing pip cantools uninstall/install. Will make another release ad require bitstruct >= 8.12.1.