cantools/cantools #276: New maintainer and/or collaborators needed

eerimoq, 2021-02-09:
Hello!

As I don't use this project myself I have little interest in maintaining it. I think the best way forward is to invite collaborators that all maintain the project together. That is, mainly reviewing and merging pull requests, but also adding new features. Please let me know what you think and if you want to become a collaborator.

Thanks,
Erik

AxelVoitier, 2021-02-17:
Hello,

As we (as a company, advantics.fr) do use cantools and relies on it significantly, we would be happy to contribute and participate in the maintenance of this nice project.
Though, our resources being limited, we would not be able to handle all of it. So, I would also call on other companies that do use cantools actively to step in as well ;-).

And to any passionate individuals also, of course :-).

Cheers,
Axel

ronit-parikh-ad, 2021-03-02:
Hello Erik,

I would like to become a collaborator.

I do take references from cantools to use in my project.

Best Regards,
Ronit

eerimoq, 2021-03-02:
Ok, I'll invite you! =)

oomzay, 2021-03-03:
I would also like to help. I am currently looking at some issues in the CDD parser. This may involve interface changes that would need wider discussion.

eerimoq, 2021-03-03:
@oomzay Invited!

NiklasReisser, 2021-03-10:
Hi, 

we are using your tools too in our projects and plan to extend that in the future - so I'd be happy to help within my limited possibilities. :) 

Best regards

Niklas

matt2005, 2021-03-11:
I'm currently working at adding cantools to [Crankshaft](https://github.com/opencardev/crankshaft)  so I'd also be looking at working on this aswell.

tuna-f1sh, 2021-03-20:
Also happy to help out here. I use the module in a lot of scripting at work so think I can be useful 🙂.

I have a [fork](https://github.com/tuna-f1sh/cantools/blob/master/cantools/subparsers/decode.py) with InfluxDB importing. Needs some tweaks after latest merge but could be useful to others (might make into separate subparser).

tobych, 2022-02-21:
I'm currently on a consulting project this year (2022) that is using this project extensively. I'm beginning spot things I'd like to improve or fix in the package and would be happy to help, perhaps starting by fixing an easy-to-fix bug already in the issues list.