cantools/cantools #278: CANdb format :help wanted:

walzert, 2021-02-16:
Hello together,

thank you for the tool. I working with that tool to transform dbc files to some different formats for further processing.

I did 2 years ago an encoding with the tool from messages that were sent by canoe and that worked good for me. Now we(me and my students at university) trying to send data with  cantools.  

My is question regarding to CANdb++ dbc-Files I playing with some different canoe files and I am not getting values above 255 correctly working. 
For me it seems that values above 255 are encoded in a strange way. And the same for 2 different values in in a byte. 

Is it a error of choosing options in candb++ or do I miss some configuration with cantools. 
Is there an golden way to work with canoe files and cantools. 

Sorry that I don't have more information atm but I could deliver more information. 

Regards,
Thomas

walzert, 2021-02-16:
A bit more context, we are working with Automotive Grade Linux and the idea of using cantools is that my students could develop their own can messages for different sensors/actuators. They should/could do this with CANdb++. Convert them in a pipeline that uses in one step cantools for the right format for AGL processing a can-intgration and they could use their signals in AGL.

eerimoq, 2021-02-16:
Hi, an example that reproduces the issue would be very helpful to understand what is going on.

oomzay, 2021-03-09:
Hi, After some digging I see no alternative but to significantly change the diagnostics.Database interface as outlined in https://github.com/eerimoq/cantools/issues/290. 

Does anyone have a view on this before I wade in?

(I am persitently worried about breaking existing users but also wondering if anyone is actually using it)