cantools/cantools #289: CDD: byte order decode is incorrect and incomplete

oomzay, 2021-03-02:
(At commit 2f66e185eee915ddd2bf9ef1672515c712e9171b)

In .CDD files available to me the byte order encoding is:

- bo='21' = BigEndian
- bo='12' = LittleEndian

The current byte order decode logic in cdd.py is therefore incorrect:

```
   if ctype.attrib['bo'] == '21':
            byte_order = 'little_endian'
```

The correct logic would be:

 ```
    if ctype.attrib['bo'] == '21':
            byte_order = 'big_endian'
     elif ctype.attrib['bo']== '12':
            byte_order = 'little_endian'
     else:
           # Other issues in this module are just logged at debug level but 
            # perhaps better to raise Exception or warning ???
            LOGGER.debug("Ignoring unsupported byte order code '%s'.", bo_code)

```

oomzay, 2021-03-03:
On further investigation I believe the test_diagnostics_database.py test case is substantially flawed due to this endianness error.

Is it known if anyone is successfully using and relying on this CDD parser in its current state?

My planned 'fix' will signficantly change the decoding of all multibyte items!

andlaus, 2021-03-05:
before you implement any sweeping changes: What is your confidence level here? Inspecting `tests/files/cdd/example.cdd` leaves the impression that it was generated using the canonical generator (i.e., CANdela studio, cf the forum post mentioned in the file). There might be a bug in CANdela studio, but that's a thing which we have to live with no matter what?

oomzay, 2021-03-06:
> What is your confidence level here?

Moderate. I would have said "high", but this implies that the CDD module could never have worked for anything but little endian byte-aligned fields. If there is anyone successfully using this CDD module please let us know!

-  CANdela reports that the "ECU Identification.Ident Number Digit (7/6)" has DataType "Bcd (2Byte)"
-  CANdela reports that the "Bcd (2Byte)" Data Type is "HighLow (Motorola)", as are _all_ the multibyte data types in this example file.
- Data Types listed as Motorola by CANdela have bo='21' entries in the CDD
- Data Types listed as Intel by CANdela have bo='12' entries in the CDD
- The current library incorrectly decodes well formed multibyte messages against an OEM CDD

>There might be a bug in CANdela studio

This issue does not imply any bug in CANdela or the example file. The bug is a fundamental mistake in the CDD loader's interpretation of the endianness codes in valid CDD files.

closed 2021-08-16 by juleq