cantools/cantools #290: CDD: Loading DIAGINST from all DIAGCLASSes into the same id space causes id collisions

oomzay, 2021-03-02:
(Commit 2f66e185eee915ddd2bf9ef1672515c712e9171b)

CDD DIAGINST IDs are scoped by DIAGCLASS. The same ID can re-occur in multiple DIAGCLASSes. 

The current cdd.py logic tries to collect them all in one list of dids which results in ID collisions and warnings:

`
2021-03-02 17:47:47,576 : WARNING : cantools.database.diagnostics.database : Overwriting DID 'XXX' with 'YYY' in the identifier to DID dictionary because they have identical identifiers 0x1.
`

Was the intention to collect all DIAGINSTs? In which case, they need grouping/indexing by DIAGCLASS, or was it to just collect the EngineeringParameters? In which case, they need filtering.

The relevant source is from cdd.py:

```
for diag_class in var.findall('DIAGCLASS'):
        for diag_inst in diag_class.findall('DIAGINST'):
            did = _load_did_element(diag_inst,
                                    data_types)
            dids.append(did)
```

oomzay, 2021-03-09:
I propose to drop the global "did" indexes and expose the hierarchical nature of CDD Classes and Instances.

I have come to the conclusion that trying to fix the id collision issue without breaking the current interface is futile. CDD files have the following hierarchical (sub)structure:

    DiagnosticClass (Class) *
        DiagnosticInstance (Instance) *

Instances do not all represent UDS "DID" items. For example some may represent UDS session codes. 

In general there is no uniqueness constraint on the IDs of Instances across Classes. Therefore Instances can only be identified relative to their respective Classes.

Any or all of the Classes and Instances may be of interest to some applications.

The Class names differ across OEMs and do not directly indicate the role of the Instances.

oomzay, 2022-03-16:
I have concluded that a signifigant redesign is needed to support CDD in a robust way. Unfortunately I do not have time for this amount of rework at present.

I have made a partial workaround in my fork of this repo. This workaround allows callers to specify a class filter when loading a CDD and thus avoid these collisions. You are of course free to use or merge that workaround.

Mcthomas777, 2022-03-16:
I did some workaround on local editable version too, actually, I did it even more simple, I just get rid off the if statement (if two dids have the same identifier, overwrite), in my case two dids can have the same identifier and not the same services; e.g. diagnostic session and EcuReset.

JasonGbd, 2022-04-08:
@Mcthomas777 ，do you able to share sample of the code before and after change?

JasonGbd, 2022-04-08:
@oomzay  , I am facing same "overwriting" issue. How do I drop global "did" and is that mean "did" become local only?

JasonGbd, 2022-04-12:
Did anyone also encounter where DID "0xYY" was replaced by "0x00YY"? May I know what is the possible solution to fix such behaviour?