cantools/cantools #298: Missing checks for signal min/max out of range and signal length too small

cindyli-13, 2021-03-18:
I'm using the cantools API to generate a DBC file based on some custom parameters. I noticed that though cantools does check for overlapping signals within a message, it doesn't check whether the specified signal min/max values are within range given the signal scale/offset, or if the specified signal length in bits is enough to cover all the enumerated choices (if provided). Is there a reason that these checks are missing?

andlaus, 2021-03-20:
it could be that the scaling factor and offset is considered for the minimum and maximum values. Be aware that I'm not sure, though...

anphucnguyen, 2021-03-21:
The feature of cantools is parsing dbc file, not generating it. That means a lot of check point for valid dbc may be not covered strictly.
You can try generating a dbc with 2 messages with same CAN ID, it will not return the error.