cantools/cantools #300: decode multiplexed signal, dlc error for bitstruct

zorce, 2021-03-23:
Cannot seem to decode messages with a shorter DLC. The received message is a multiplexed signal:
mux 0 
value 24

```
Exception in thread Thread-1:
Traceback (most recent call last):
  File "C:\Users\dennis.delin\AppData\Local\Programs\Python\Python37-32\lib\threading.py", line 926, in _bootstrap_inner
    self.run()
  File "C:\Users\dennis.delin\AppData\Local\Programs\Python\Python37-32\lib\threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File ".\Library\CAN.py", line 47, in thread_read
    decoded = msg.decode(frame.data)
  File "C:\Users\dennis.delin\AppData\Local\Programs\Python\Python37-32\lib\site-packages\cantools\database\can\message.py", line 819, in decode
    return self._decode(self._codecs, data, decode_choices, scaling)
  File "C:\Users\dennis.delin\AppData\Local\Programs\Python\Python37-32\lib\site-packages\cantools\database\can\message.py", line 782, in _decode
    scaling)
  File "C:\Users\dennis.delin\AppData\Local\Programs\Python\Python37-32\lib\site-packages\cantools\database\utils.py", line 94, in decode_data
    unpacked = formats.big_endian.unpack(bytes(data))
  File "C:\Users\dennis.delin\AppData\Local\Programs\Python\Python37-32\lib\site-packages\bitstruct\__init__.py", line 444, in unpack
    return self.unpack_from(data)
  File "C:\Users\dennis.delin\AppData\Local\Programs\Python\Python37-32\lib\site-packages\bitstruct\__init__.py", line 461, in unpack_from
    return {info.name: v for info, v in self.unpack_from_any(data, offset)}
  File "C:\Users\dennis.delin\AppData\Local\Programs\Python\Python37-32\lib\site-packages\bitstruct\__init__.py", line 461, in <dictcomp>
    return {info.name: v for info, v in self.unpack_from_any(data, offset)}
  File "C:\Users\dennis.delin\AppData\Local\Programs\Python\Python37-32\lib\site-packages\bitstruct\__init__.py", line 301, in unpack_from_any
    len(bits)))
bitstruct.Error: unpack requires at least 56 bits to unpack (got 16)

q
PS C:\SVN\5266_SEMIC\DEVELOPMENT\Test\Robot> python .\Library\CAN.py
wr
Could not decode Timestamp: 1616513953.405618        ID: 0322    S Rx                DLC:  2    00 18                       Channel: 0

```

juleq, 2021-03-23:
Hi. 

Thats probably not enough context to see what is going on. The error message is pretty clear: A signal with an expected length of 56 bit was just 16 bit. Might be a problem in your db definition.

What do message, signal and multiplexor definitions look like? What length was defined for that signal with mux 0?

Best regards

zorce, 2021-03-24:
More on the context, the messages is a set of muxed signals according to the CAN in automation specification. 
Sending a muxed signal mux 0 value 24 in this case, and will receive a response on another frame the one above. 
When sending the muxed signal, DLC will be 7 but the response only 2.
[image]
mux 0x0 has only one signal with 8 byte length in this case
mux 0xA is multiple signal and full hence 56 bit (not sent in this instance)

juleq, 2021-03-24:
8 bytes and the mux will not fit a frame. Maybe you meant 8 bit for the signal in case the mux is 0x0?

What do you use as db format? Do you use dbc, kcd? I am asking because the definition for both messages and all their signals is required as the context.

Having the request and response be different length should work just fine as long as they use different ids.

juleq, 2021-03-24:
I think what might be happening here is that you use the wrong message definition to decode. E.g. trying to decode the response with the request definition. This could happen due to a problem in the dbc definition or, if you set up your db programmatically, probably also a referencing mishap of messages or signals (e.g. reusing the request reference to define the response and thereby overwriting the request or something like that).

zorce, 2021-03-24:
Yes I meant 8 bits not bytes for the signal.
And yes your assumptions is correct, using a DBC database file. The dbc file is defining all the messages with the receiver arbitration id. 
So I use the database message and set the mux, flag and signal, create a can.Message object and send it of with the txid
then later on a thread the response is received and I decode the message

Database cutout with signal names renamed manually and only first mux signal:
```
BO_ 802 Config: 7 SUT
 SG_ mux M : 0|7@1+ (1,0) [0|0] "" Vector__XXX
 SG_ flag : 7|1@1+ (1,0) [0|0] "" Vector__XXX
 SG_ MySignal m0 : 8|8@1+ (1,0) [0|24] "" Vector__XXX
```

Sending:
```
data = msg.encode({'mux': 0, 'flag': 0, 'MySignal': 24})
message = can.Message(arbitration_id=0x302, is_extended_id=False, data=data)
self.bus.send(message)
```

Receiving:
```
try:
    msg = self.db.get_message_by_frame_id(frame.arbitration_id)
    decoded = msg.decode(frame.data)
except Exception as e:
    print('Could not decode', frame)
    print(e)
```

zorce, 2021-03-24:
Maybe if the decoder expects the a buffer of data with 56bits all set to 0 then and the new data ontop and do the decode?
Should not cause any issues right?

juleq, 2021-03-24:
You are sending the 0x302, but you are showing the definition of frame 0x322. Please include the definition of 0x302 (BO 770).

The way I understood you before was that the request is 0x302 (DLC 7) and the response is 0x322 (DLC 2). But you have shown that 0x322 is defined as a frame with DLC 7.

zorce, 2021-03-24:
The definition is the same for both, but I don't have any definition for the tx message in the dbc file they are the same. So using the 0x322 definition for sending aswell.

Did a quick implementation to manually calculate the values and it works for now, but not ideal.
```
                try:
                    msg = self.db.get_message_by_frame_id(frame.arbitration_id)
                    frame_bits = bitarray()
                    frame_bits.frombytes(bytes(frame.data))

                    mux = None
                    flag = None

                    for i, sig in enumerate(msg.signals):
                        signed = sig.is_signed
                        byteorder = 'little' if sig.byte_order == 'little_endian' else 'big'

                        if sig.length > 1:
                            raw = int.from_bytes(frame_bits[sig.start:sig.start+sig.length], byteorder=byteorder, signed=sig.is_signed)
                        #elif frame_bits[0] == 1:
                            # 1 indicates failure
                        #    break
                        else:
                            raw = int(frame_bits[sig.start])
                        phys = (raw * sig.scale) + sig.offset
                        
                        if sig.name == 'mux':
                            mux = raw
                            continue
                        elif sig.name == 'flag':
                            flag = raw
                            continue
                        elif mux is not None and flag is not None:
                            if i == mux + 2:
                                print(sig.name, phys)
                                continue
                        else:
                            print(sig.name, phys)
                            
                except Exception as e:
                    print('Not in database',frame.arbitration_id, frame.data)
                    print(e)
```

juleq, 2021-03-24:
You cannot use the same definition for request and response. You need independently defined messages with their respective signal layouts and DLCs. It is IMHO not worth hacking around that.

zorce, 2021-03-24:
Well even if I make a copy of the existing message definition and call it something else (eg. sufix _rx)
I will end up with the same issue, the received message cannot be decoded correctly as the definition is not interpreted correctly.
The different muxes have different signal lengths and hence not parsed correctly.

Will give it a try to see if I manage to make a fix for it with a pull request.

juleq, 2021-03-24:
Please post your message definitions.

zorce, 2021-03-24:
Here is the full message:
```
BO_ 802 MyMessage: 7 SUT
 SG_ mux M : 0|7@1+ (1,0) [0|0] "" Vector__XXX
 SG_ flag : 7|1@1+ (1,0) [0|0] "" Vector__XXX
 SG_ MySignal1 m0 : 8|8@1+ (1,0) [0|24] "" Vector__XXX
 SG_ MySignal2 m1 : 8|8@1+ (1,0) [0|255] "" Vector__XXX
 SG_ MySignal3 m2 : 8|16@1+ (1,0) [50|500] "" Vector__XXX
 SG_ MySignal4 m3 : 8|16@1- (1,0) [-3600|3600] "" Vector__XXX
 SG_ MySignal5 m4 : 8|16@1+ (1,0) [0|5000] "" Vector__XXX
 SG_ MySignal6 m5 : 8|8@1+ (1,0) [0|255] "" Vector__XXX
 SG_ MySignal7 m6 : 8|8@1+ (1,0) [0|100] "" Vector__XXX
 SG_ MySignal8 m6 : 16|8@1+ (1,0) [0|100] "" Vector__XXX
 SG_ MySignal9 m6 : 24|8@1+ (1,0) [0|100] "" Vector__XXX
 SG_ MySignal10 m6 : 32|8@1+ (1,0) [0|100] "" Vector__XXX
 SG_ MySignal11 m6 : 40|8@1+ (1,0) [0|255] "" Vector__XXX
 SG_ MySignal12 m6 : 48|8@1+ (1,0) [0|255] "" Vector__XXX
 SG_ MySignal13 m7 : 8|8@1+ (1,0) [0|255] "" Vector__XXX
 SG_ MySignal14 m7 : 16|8@1+ (1,0) [0|255] "" Vector__XXX
 SG_ MySignal15 m7 : 24|8@1+ (1,0) [0|255] "" Vector__XXX
 SG_ MySignal16 m7 : 32|8@1+ (1,0) [0|255] "" Vector__XXX
 SG_ MySignal17 m7 : 40|8@1+ (1,0) [0|255] "" Vector__XXX
 SG_ MySignal18 m7 : 48|8@1+ (1,0) [0|255] "" Vector__XXX
 SG_ MySignal19 m8 : 8|8@1+ (1,0) [0|255] "" Vector__XXX
 SG_ MySignal20 m8 : 16|8@1+ (1,0) [0|255] "" Vector__XXX
 SG_ MySignal21 m8 : 24|8@1+ (1,0) [0|255] "" Vector__XXX
 SG_ MySignal22 m8 : 32|8@1+ (1,0) [0|255] "" Vector__XXX
 SG_ MySignal23 m8 : 40|8@1+ (1,0) [0|255] "" Vector__XXX
 SG_ MySignal24 m8 : 48|8@1+ (1,0) [0|255] "" Vector__XXX
 SG_ MySignal25 m9 : 8|8@1+ (1,0) [0|255] "" Vector__XXX
 SG_ MySignal26 m9 : 16|8@1+ (1,0) [0|255] "" Vector__XXX
 SG_ MySignal27 m9 : 24|8@1+ (1,0) [0|255] "" Vector__XXX
 SG_ MySignal28 m9 : 32|8@1+ (1,0) [0|255] "" Vector__XXX
 SG_ MySignal29 m9 : 40|8@1+ (1,0) [0|255] "" Vector__XXX
 SG_ MySignal30 m9 : 48|8@1+ (1,0) [0|255] "" Vector__XXX
 SG_ MySignal31 m10 : 8|8@1+ (1,0) [0|255] "" Vector__XXX
 SG_ MySignal32 m10 : 16|8@1+ (1,0) [0|255] "" Vector__XXX
 SG_ MySignal33 m10 : 24|8@1+ (1,0) [0|255] "" Vector__XXX
 SG_ MySignal34 m10 : 32|8@1+ (1,0) [0|255] "" Vector__XXX
 SG_ MySignal35 m10 : 40|8@1+ (1,0) [0|255] "" Vector__XXX
 SG_ MySignal36 m10 : 48|8@1+ (1,0) [0|255] "" Vector__XXX
 SG_ MySignal37 m11 : 8|8@1+ (1,0) [0|255] "" Vector__XXX
 SG_ MySignal38 m11 : 16|8@1+ (1,0) [0|255] "" Vector__XXX
 SG_ MySignal39 m11 : 24|8@1+ (1,0) [0|255] "" Vector__XXX
 SG_ MySignal40 m11 : 32|8@1+ (1,0) [0|255] "" Vector__XXX
 SG_ MySignal41 m11 : 40|8@1+ (1,0) [0|255] "" Vector__XXX
 SG_ MySignal42 m12 : 8|16@1- (1,0) [0|0] "" Vector__XXX
 SG_ MySignal43 m13 : 8|16@1- (1,0) [0|0] "" Vector__XXX
 SG_ MySignal44 m14 : 8|16@1- (1,0) [0|0] "" Vector__XXX
 SG_ MySignal45 m15 : 8|16@1- (1,0) [0|0] "" Vector__XXX
 SG_ MySignal46 m16 : 8|1@1+ (1,0) [0|1] "" Vector__XXX
 SG_ MySignal47 m16 : 9|1@1+ (1,0) [0|1] "" Vector__XXX
 SG_ MySignal48 m16 : 10|1@1+ (1,0) [0|1] "" Vector__XXX
 SG_ MySignal49 m16 : 11|1@1+ (1,0) [0|1] "" Vector__XXX
 SG_ MySignal50 m16 : 12|1@1+ (1,0) [0|1] "" Vector__XXX
 SG_ MySignal51 m16 : 13|1@1+ (1,0) [0|1] "" Vector__XXX
 SG_ MySignal52 m16 : 14|1@1+ (1,0) [0|1] "" Vector__XXX
 SG_ MySignal53 m16 : 15|1@1+ (1,0) [0|1] "" Vector__XXX
 SG_ MySignal54 m17 : 8|1@1+ (1,0) [0|1] "" Vector__XXX
 SG_ MySignal55 m17 : 9|1@1+ (1,0) [0|1] "" Vector__XXX
 SG_ MySignal56 m17 : 10|1@1+ (1,0) [0|1] "" Vector__XXX
 SG_ MySignal57 m18 : 8|16@1+ (1,0) [0|5000] "" Vector__XXX
 SG_ MySignal58 m19 : 8|16@1- (1,0) [-200|600] "" Vector__XXX
 SG_ MySignal59 m20 : 8|16@1- (1,0) [-200|600] "" Vector__XXX
 SG_ MySignal60 m21 : 8|8@1+ (1,0) [0|5] "" Vector__XXX
 SG_ MySignal61 m22 : 8|16@1- (1,0) [-5000|5000] "" Vector__XXX
 SG_ MySignal62 m23 : 8|16@1+ (1,0) [0|5000] "" Vector__XXX
 SG_ MySignal63 m24 : 8|16@1- (1,0) [-5000|5000] "" Vector__XXX
 SG_ MySignal64 m25 : 8|16@1+ (1,0) [0|5000] "" Vector__XXX
 SG_ MySignal65 m26 : 8|8@1+ (1,0) [4|12] "" Vector__XXX
 SG_ MySignal66 m27 : 8|8@1+ (1,0) [192|200] "" Vector__XXX
 SG_ MySignal67 m28 : 8|16@1+ (1,0) [0|7200] "" Vector__XXX
 SG_ MySignal68 m28 : 24|16@1+ (1,0) [0|7200] "" Vector__XXX
```

zorce, 2021-03-24:
Don't understand the util.py -> decode_data function fully.. as the byte order makes no sense to me
if modifying the incoming bus data (0x0, 0x18) by adding padding, to the end or infront, but it seams like the following will parse the data correctly.. 
```
data = bytearray(8)
data[0] = 0x0
data[1] = 0x0 #mux value is here
data[2] = 0x18 #24 signal value here?
data[3] = 0x0
data[4] = 0x0
data[5] = 0x0
data[6] = 0x0
data[7] = 0x0
```

juleq, 2021-03-24:
Please post your message definitions. You have posted the rx one, but you said you did a tx one as a trial. Post that one please.

zorce, 2021-03-24:
tx one is the same
```
BO_ 770 MyMessage_TX: 7 TUS
 SG_ mux M : 0|7@1+ (1,0) [0|0] "" Vector__XXX
 SG_ flag : 7|1@1+ (1,0) [0|0] "" Vector__XXX
 SG_ MySignal1 m0 : 8|8@1+ (1,0) [0|24] "" Vector__XXX
 SG_ MySignal2 m1 : 8|8@1+ (1,0) [0|255] "" Vector__XXX
 SG_ MySignal3 m2 : 8|16@1+ (1,0) [50|500] "" Vector__XXX
 SG_ MySignal4 m3 : 8|16@1- (1,0) [-3600|3600] "" Vector__XXX
 SG_ MySignal5 m4 : 8|16@1+ (1,0) [0|5000] "" Vector__XXX
 SG_ MySignal6 m5 : 8|8@1+ (1,0) [0|255] "" Vector__XXX
 SG_ MySignal7 m6 : 8|8@1+ (1,0) [0|100] "" Vector__XXX
 SG_ MySignal8 m6 : 16|8@1+ (1,0) [0|100] "" Vector__XXX
 SG_ MySignal9 m6 : 24|8@1+ (1,0) [0|100] "" Vector__XXX
 SG_ MySignal10 m6 : 32|8@1+ (1,0) [0|100] "" Vector__XXX
 SG_ MySignal11 m6 : 40|8@1+ (1,0) [0|255] "" Vector__XXX
 SG_ MySignal12 m6 : 48|8@1+ (1,0) [0|255] "" Vector__XXX
 SG_ MySignal13 m7 : 8|8@1+ (1,0) [0|255] "" Vector__XXX
 SG_ MySignal14 m7 : 16|8@1+ (1,0) [0|255] "" Vector__XXX
 SG_ MySignal15 m7 : 24|8@1+ (1,0) [0|255] "" Vector__XXX
 SG_ MySignal16 m7 : 32|8@1+ (1,0) [0|255] "" Vector__XXX
 SG_ MySignal17 m7 : 40|8@1+ (1,0) [0|255] "" Vector__XXX
 SG_ MySignal18 m7 : 48|8@1+ (1,0) [0|255] "" Vector__XXX
 SG_ MySignal19 m8 : 8|8@1+ (1,0) [0|255] "" Vector__XXX
 SG_ MySignal20 m8 : 16|8@1+ (1,0) [0|255] "" Vector__XXX
 SG_ MySignal21 m8 : 24|8@1+ (1,0) [0|255] "" Vector__XXX
 SG_ MySignal22 m8 : 32|8@1+ (1,0) [0|255] "" Vector__XXX
 SG_ MySignal23 m8 : 40|8@1+ (1,0) [0|255] "" Vector__XXX
 SG_ MySignal24 m8 : 48|8@1+ (1,0) [0|255] "" Vector__XXX
 SG_ MySignal25 m9 : 8|8@1+ (1,0) [0|255] "" Vector__XXX
 SG_ MySignal26 m9 : 16|8@1+ (1,0) [0|255] "" Vector__XXX
 SG_ MySignal27 m9 : 24|8@1+ (1,0) [0|255] "" Vector__XXX
 SG_ MySignal28 m9 : 32|8@1+ (1,0) [0|255] "" Vector__XXX
 SG_ MySignal29 m9 : 40|8@1+ (1,0) [0|255] "" Vector__XXX
 SG_ MySignal30 m9 : 48|8@1+ (1,0) [0|255] "" Vector__XXX
 SG_ MySignal31 m10 : 8|8@1+ (1,0) [0|255] "" Vector__XXX
 SG_ MySignal32 m10 : 16|8@1+ (1,0) [0|255] "" Vector__XXX
 SG_ MySignal33 m10 : 24|8@1+ (1,0) [0|255] "" Vector__XXX
 SG_ MySignal34 m10 : 32|8@1+ (1,0) [0|255] "" Vector__XXX
 SG_ MySignal35 m10 : 40|8@1+ (1,0) [0|255] "" Vector__XXX
 SG_ MySignal36 m10 : 48|8@1+ (1,0) [0|255] "" Vector__XXX
 SG_ MySignal37 m11 : 8|8@1+ (1,0) [0|255] "" Vector__XXX
 SG_ MySignal38 m11 : 16|8@1+ (1,0) [0|255] "" Vector__XXX
 SG_ MySignal39 m11 : 24|8@1+ (1,0) [0|255] "" Vector__XXX
 SG_ MySignal40 m11 : 32|8@1+ (1,0) [0|255] "" Vector__XXX
 SG_ MySignal41 m11 : 40|8@1+ (1,0) [0|255] "" Vector__XXX
 SG_ MySignal42 m12 : 8|16@1- (1,0) [0|0] "" Vector__XXX
 SG_ MySignal43 m13 : 8|16@1- (1,0) [0|0] "" Vector__XXX
 SG_ MySignal44 m14 : 8|16@1- (1,0) [0|0] "" Vector__XXX
 SG_ MySignal45 m15 : 8|16@1- (1,0) [0|0] "" Vector__XXX
 SG_ MySignal46 m16 : 8|1@1+ (1,0) [0|1] "" Vector__XXX
 SG_ MySignal47 m16 : 9|1@1+ (1,0) [0|1] "" Vector__XXX
 SG_ MySignal48 m16 : 10|1@1+ (1,0) [0|1] "" Vector__XXX
 SG_ MySignal49 m16 : 11|1@1+ (1,0) [0|1] "" Vector__XXX
 SG_ MySignal50 m16 : 12|1@1+ (1,0) [0|1] "" Vector__XXX
 SG_ MySignal51 m16 : 13|1@1+ (1,0) [0|1] "" Vector__XXX
 SG_ MySignal52 m16 : 14|1@1+ (1,0) [0|1] "" Vector__XXX
 SG_ MySignal53 m16 : 15|1@1+ (1,0) [0|1] "" Vector__XXX
 SG_ MySignal54 m17 : 8|1@1+ (1,0) [0|1] "" Vector__XXX
 SG_ MySignal55 m17 : 9|1@1+ (1,0) [0|1] "" Vector__XXX
 SG_ MySignal56 m17 : 10|1@1+ (1,0) [0|1] "" Vector__XXX
 SG_ MySignal57 m18 : 8|16@1+ (1,0) [0|5000] "" Vector__XXX
 SG_ MySignal58 m19 : 8|16@1- (1,0) [-200|600] "" Vector__XXX
 SG_ MySignal59 m20 : 8|16@1- (1,0) [-200|600] "" Vector__XXX
 SG_ MySignal60 m21 : 8|8@1+ (1,0) [0|5] "" Vector__XXX
 SG_ MySignal61 m22 : 8|16@1- (1,0) [-5000|5000] "" Vector__XXX
 SG_ MySignal62 m23 : 8|16@1+ (1,0) [0|5000] "" Vector__XXX
 SG_ MySignal63 m24 : 8|16@1- (1,0) [-5000|5000] "" Vector__XXX
 SG_ MySignal64 m25 : 8|16@1+ (1,0) [0|5000] "" Vector__XXX
 SG_ MySignal65 m26 : 8|8@1+ (1,0) [4|12] "" Vector__XXX
 SG_ MySignal66 m27 : 8|8@1+ (1,0) [192|200] "" Vector__XXX
 SG_ MySignal67 m28 : 8|16@1+ (1,0) [0|7200] "" Vector__XXX
 SG_ MySignal68 m28 : 24|16@1+ (1,0) [0|7200] "" Vector__XXX
```

juleq, 2021-03-24:
Ok, whats your cantools version?

I use cantools for db conversion mainly, it has been a while since I have used it for encoding/decoding.

Need to go now but maybe I can manage to try to reproduce this myself later today. It looks like it does not select the correct muxed signal definition for mux group 0x0. I think it is fair to assume that you would not be the first one to stumble over this if it is a general problem with the multiplex implementation. Multiplexed signals are rather common.

zorce, 2021-03-24:
Thanks,
Don't know how common it is to use muxed signals, first time for me and have played with CAN for quite some time.
I run on version 36.2.0 (latest official) when installing with pip

juleq, 2021-03-24:
Trial is looking good. I slapped the two message definitions into a dbc and opened it using CANdb++ to verify I got the syntax right. Works.

Then checking my own cantools version: 36.2.0

Then in Python REPL:
```
import cantools
from pprint import pprint
db = cantools.database.load_file('cantoolstest.dbc')
data = [0, 24, 0, 0, 0, 0, 0, 0]
db.decode_message(0x322, data)
---> {'mux': 0, 'flag': 0, 'MySignal1': 24}
```
Which seems fine. Do you see a difference in your implementation? A notable one is that you use the message member to decode. I will try that, too.

[cantoolstest.dbc.txt](https://github.com/eerimoq/cantools/files/6199882/cantoolstest.dbc.txt)

juleq, 2021-03-24:
Yep, that also works on my end. I just added this to the REPL session from before:
```
msg = db.get_message_by_name('MyMessage')
msg.decode(data)
--->{'mux': 0, 'flag': 0, 'MySignal1': 24}
```
Just to be sure I restarted the REPL session and did just the decode using the messages member. Also works as expected.

zorce, 2021-03-25:
Hey, Maybe abit of a confusion here, but the issue relyes in the data length.
Downloaded your dbc file and copy paste of your code, edited the data to the length I get it from my hardware device. DLC 2

```
import cantools
from pprint import pprint
db = cantools.database.load_file('cantoolstest.dbc')

msg = db.get_message_by_frame_id(0x322)
#data = [0, 24, 0, 0, 0, 0, 0, 0] # <- WORKS FINE (FULL FRAME 8bytes)
data = [0, 24] # <- FAILING EXAMPLE

print(msg.decode(data))

```

That's why I tried to pad the data in the util.py up to 8 bytes if less

juleq, 2021-03-25:
Ok, got it now.

I think its unusual to have basically the same definition for request and response.

Will the HW ever tx more than two bytes? If not, you could simplify the response definition.

I think promising 7 bytes in the definition and then delivering 2 is probably asking for trouble :). I do not know if changing DLCs for a frame is somehow considered in the relevant bus standards.

But I agree that cantools could be more robust in this case. I have a decoder in C# and there I always pad to 8. I have not looked into src yet but I also think that padding to at least the DLC should be done by cantools.

zorce, 2021-03-25:
Ah yes, and the 7 bytes in the definition explains the behaviour I see in the weird padding in utils.
message definition size 7
data = [0x0, 0x18]
add up to message definition at end of data
data = [0x0, 0x18, **0x0, 0x0, 0x0, 0x0, 0x0**]
then add up to 8 bytes in front of data
data = [**0x0**, 0x0, 0x18, **0x0, 0x0, 0x0, 0x0, 0x0**]

juleq, 2021-03-25:
I stepped through the code that sets up the codecs on construction of the db. It will always use the defined DLC as the length. And there is only one DLC for a message, be it muxed or not.

Down the line, when a decode happens, first thing the unpack method in the bitstruct package does is sanity checking the expected number of bits (56) against the passed in one (16).

I would probably either reconsider the rx message definition (define one with DLC 2 if possible) or pad this frame myself before passing it into the decode method. One could think of increasing robustness of the cantools implementation in this regard but the encoding/decoding is rather complex, you need to have some conviction for that :).

zorce, 2021-03-25:
modify the dlc length in rx message will not work as different mux combinations have different length eg mux 0xA uses the full 56 bits. Think this would be a valid robustness fix for the cantools library as this will be a common problem for CiA (CAN in Automation) implementations where you have one message definition and different address between the nodes.

Will pad my data before sending it to cantools for decoding for now. Would be happy to remove that code when this is handled by the library tho :D

Thanks for all the effort!

zorce, 2021-03-25:
Just for the record on my implementation if someone else end up with this issue:
```
# padding up to message length
if len(frame.data) < msg.length:
    diff = msg.length - len(frame.data)
    padding = bytearray(diff)
    frame.data = frame.data + padding
decoded = msg.decode(frame.data)
```

juleq, 2021-03-25:
I do not have a CiA standard text but I think it might be a requirement for CiA nodes to pad up to the DLC. The SDO diagrams on their website seem to suggest that. So if you have a node that responds with DLC 2 when 7 was expected, it might be the original offender :).

But I think the robustness argument is still valid.

Maybe one of the maintainers could add an enhancement tag?