cantools/cantools #301: Standard frame id is more than 11 bits

smpshehan, 2021-03-26:
I used cantools python package to decode canbus message. I used a dbc file created by me for testing. I copied a sample file. When I use can id like 419358976, I get error. But for smaller can ids like 350, it works. Does cantools fail for extended frame ids? how do I get this working? How do I specify to use extended can ids?

my code which is failing for extended id's is as follows:

```
db = cantools.database.load_file('.\\src\\test\\resources\\j1939.dbc')
print(db.decode_message(419358976,b'\xff\xff\xff\xc0\x0c\xff\xff\xff'))
```

error: 
`cantools.database.errors.Error: Standard frame id 0x18fee900 is more than 11 bits in message EEC1.`

juleq, 2021-03-26:
The dbc format has no dedicated field for indicating an id is extended. It uses bit 0x8000000 of the id field for that purpose.

smpshehan, 2021-03-27:
> The dbc format has no dedicated field for indicating an id is extended. It uses bit 0x8000000 of the id field for that purpose.

Hi Jueq,
Ok, but still this doesn't answer my question. Why cantools can not decode my message for can id 419358976?

juleq, 2021-03-27:
Well. It actually does. You enter an ill formatted id field when passing id 419358976 in a dbc. It is missing the 32th bit, which should have been set because the id is extended. cantools does notice that you have passed an id as standard that does not fit in 11 bits, so it rightfully complains.

smpshehan, 2021-03-27:
> 419358976

Hi, I really appreciate your help. Do you mean this can id 419358976 is invalid? I get that id by reading a CAN0 device. It is he actual can id I receive from the device.

juleq, 2021-03-27:
The id itself is perfectly valid. Unfortunately, the author of the dbc file format chose to "misuse" a bit in the id field to indicate if it is extended (instead of adding an atomic boolean field to do that.

cantools uses a dedicated field to represent the std/ext information internally.

So the same id does look different in the dbc file: In the dbc file it is formatted as decimal and 0x80000000 is added if the id is extended.

smpshehan, 2021-03-27:
> look

ah...so he dbc file needs to be fixed. Could you please show me a sample and valid entry for this extended id in the dbc file? what is the valid entry should look like in the dbc file?

juleq, 2021-03-27:
If on Windows: Open calc and switch to programmers view. Select dec. Paste 419358976. Select hex. Add 80000000. Select dec. The numer you see now is what needs to be entered in the text of the dbs file if edited by hand.

smpshehan, 2021-03-27:
> 80000000

Oh... I understood what you meant now. That is really helpful. I will try this on Monday as I don't have access to this now. Thanks a lot. I will update you on Monday :-) I didn't find this information any where. This must be really documented. Where did you find this information?

juleq, 2021-03-27:
Unfortunately Vector Informatik does not publish a dbc file format spec, but you can find a few resources across the internet that detail the format of these files. But I just checked a few and they all did not contain this bit of information :).

The usual way to edit dbc files and check if their syntax is correct is to use Vectors CANdb++ program. That does get shipped with various of their other products and maybe there is also a demo available. Using CANdb++ is also the way for me to know how extended is encoded: You can just change it back and forth and see what it is doing to the id field in the file.

I also like editing dbcs by hand or using generators, doing all via UI can be tedious. But using CANdb++ to check every now and then is helpful.

smpshehan, 2021-03-28:
You are a legend! IT WORKED!
I think this has to be documented properly somewhere

andlaus, 2021-03-29:
@juleq: It would be great if you could make a fix for this quirk of the DBC format. (If you prefer, I can do it for you if you tell me how to test it.)

juleq, 2021-03-29:
@andlaus There is nothing wrong with cantools. It properly parses correctly formatted dbc files. Issue here was that hand-editing dbc files requires some "more esoteric" knowledge of the dbc file format details (the borrowed bit in the id field).

andlaus, 2021-03-29:
ah, good to know. I'm frequently surprised how many file format are/started out just internal formats of some proprietary products -- not just in the automotive industry. IMO "open" formats are not always better, though. (err, I'm looking at you ARXML...)

juleq, 2021-03-29:
Yes. I like Kayak but its tough to go against something as well established across the industry as dbc. And one could argue ingesting the compact text of a dbc is quicker than these verbose xml files :).

smpshehan, 2021-03-29:
As I understood that we need to buy this dbc file from Vector Informatik. Is there a better way to decode CAN messages without using their dbc file? like a open standard?

juleq, 2021-03-30:
You can use the dbc file format. If you want their tooling, you pay. Kayak (kcd) is open source.

smpshehan, 2021-03-30:
Does python cantools support cd file as well? if yes, is there a place to download the full version of kcd file for free? does it has all the extended canid's as same in the dbc file?

juleq, 2021-03-30:
You can just load a dbc using cantools and save it as kcd. cantools has that capability.

I have not used python-can myself. But I think it does not do the encoding/decoding. You would combine python-can and cantools for that.

andlaus, 2021-03-31:
as @juleq said above: you may use the DBC file format however you like, you only need to pay if you want to use Vector Informatik's tools (they are the canonical reference, but as you have noticed they are quite pricey), or -- depending on the licensing conditions of the DBC file -- you need to pay to get access to the file itself.  The situation is exactly like for less exotic file formats like PDF: In order to use a PDF document, you don't need to pay Adobe for an Acrobat license, but you might have to pay the publisher to obtain a PDF copy of e.g. Harry Potter...