cantools/cantools #302: FEATURE REQUEST: javascript implementation

euberdeveloper, 2021-04-02:
It would be beautiful to have a javascript implementation that from a schema generates the code in javascript or typescript for serialize or deserialize the can messages.

zabealbe, 2021-04-02:
Yes please it would be awesome to use THE INDUSTRY STANDARD for AUTOMOTIVE in my NODEJS WEB APP.

euberdeveloper, 2021-04-02:
I do not think that a webapp would be a good usecase for this project.

andlaus, 2021-08-17:
this is out of scope for this project: as far as I'm concerned cantools is going you stay python only. Writing a JavaScript code generator similar to the C one should not be too difficult, though. Closing this issue for now.

closed 2021-08-17 by andlaus

juleq, 2021-08-17:
A note to node.js developers: You can use the fabulous socketcan package by sebi2k1 to encode/decode and send/receive on Linux platforms. It uses kcd as its definition file but you can use cantools to convert dbc->kcd. And if you suffer from boredom you could probably fairly easy do a PR for a js dbc parser for the socketcan project.