cantools/cantools #305: error: no command available when installed with pip -e

erzoe, 2021-04-07:
I want to contribute a new feature to cantools but after a git pull I cannot run any cantools command anymore.

Which ever subcommand I am trying to run it gives the following error message:

> error: Command "decode" is unavailable: "No module named 'cantools.__init__.subparsers'; 'cantools.__init__' is not a package"

I have tried to reinstall cantools:

	$ cd
	$ git clone https://github.com/eerimoq/cantools.git
	$ pip uninstall cantools
	$ rm ~/.local/bin/cantools
	$ pip install --editable ~/cantools

but the problem remains.

Interestingly when installing it with

	$ pip install cantools

it works.

I am using Python 3.7.3 on a Linux system.

I have inserted a few print statements and found that it uses the following import:

```python
from importlib_metadata import distribution
```

Exactly one entry point is found:
```python
EntryPoint(name='cantools', value='cantools.__init__:_main', group='console_scripts')
```

erzoe, 2021-04-07:
I have figured out that the executable ~/.local/bin/cantools is automatically generated and can be configured in setup.py.
There the entry point
```python
'console_scripts': ['cantools=cantools.__init__:_main']
```
is defined.
After removing the `__init__` from that it works again.
```python
'console_scripts': ['cantools=cantools.__init__:_main']
```

Why was that `__init__` in there? Does it break something if I remove it?

andlaus, 2021-04-09:
I don't know why the `__init__` is there (very likely there's a reason). I guess that adding the directory where you cloned cantools to to the `PYTHONPATH` enviroment variable also does the trick?

erzoe, 2021-04-12:
unfortunately no. I have tried adding `export PYTHONPATH="$PYTHONPATH:$HOME"` to my bashrc, reinstalled cantools, and I am back to 

    error: Command "decode" is unavailable: "No module named 'cantools.__init__.subparsers'; 'cantools.__init__' is not a package"

andlaus, 2021-04-14:
try `export PYTHONPATH="$PYTHONPATH:$HOME/cantools` if you really cloned cantools directly into you home directory. (i suppose that you used something equivalent to `cd $HOME; git clone https://github.com/eerimoq/cantools` to clone the repository.)

erzoe, 2021-04-29:
Sorry for the late reply. `export PYTHONPATH="$PYTHONPATH:$HOME/cantools` does not make a difference.

However, I have managed to narrow it down to the commit where this broke.
I have written the following script to test installation. It takes a commit as argument, checks out that commit and reinstalls cantools with that version. Before commit 5a9c65d9e45de36cb6c9007767cc381deac02bb6 this script prints the help of cantools decode, as it should. Starting with that commit it says "Command "decode" is unavailable: "No module named 'cantools.\_\_init\_\_.subparsers'; 'cantools.\_\_init\_\_' is not a package""

```bash
#!/usr/bin/env bash

url='https://github.com/eerimoq/cantools.git'
path='cantools-installation-test'
commit="$1"
commit="${commit:-master}"

# ------- preparation ------

#export PYTHONPATH="$PYTHONPATH:`realpath "$path"`"
echo "PYTHONPATH='$PYTHONPATH'"

if ! [ -d "$path" ]; then
	git clone "$url" "$path"
fi

if ! [ "$commit" = "" ]; then
	cd "$path"
	git checkout "$commit"
	cd -
fi

# ------- uninstall ------

pip3 uninstall -y cantools
rm ~/.local/bin/cantools

# ------- install ------

pip3 install --editable "$path"

# ------- test ------

clear
cantools decode -h
```

Also I have found out that this error seems to occur only when calling cantools directly
```bash
$ cantools decode -h
```

But using
```bash
$ python3 -m cantools decode -h
```
seems to work always.

erzoe, 2021-04-29:
I have inserted a `raise e` in the except block of `_load_subparser` in `cantools/__init__.py` and now I am getting the following traceback:
```
$ cantools decode -h
Traceback (most recent call last):
  File "~/.local/bin/cantools", line 33, in <module>
    sys.exit(load_entry_point('cantools', 'console_scripts', 'cantools')())
  File "~/cantools-installation-test/cantools/__init__.py", line 67, in _main
    _load_subparser('decode', subparsers)
  File "~/cantools-installation-test/cantools/__init__.py", line 45, in _load_subparser
    raise e
  File "~/cantools-installation-test/cantools/__init__.py", line 41, in _load_subparser
    package=__loader__.name)
  File "/usr/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 953, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 962, in _find_and_load_unlocked
ModuleNotFoundError: No module named 'cantools.__init__.subparsers'; 'cantools.__init__' is not a package
```

andlaus, 2021-05-01:
hm, this seems suspiciously to the problem fixed by #299. would be cool if you checked if this PR fixes your issue. (unfortunately  I am currently unable to wield the Green Button due to reasons which you probably don't want to know.)

erzoe, 2021-05-01:
yes, that works. Thanks a lot, I didn't notice that pull request.

closed 2022-01-02 by erzoe