cantools/cantools #309: Signal order with in the messages

malneni, 2021-06-21:
when a DBC files is created,  I noticed that signal  in messages are sorted in alphabetical order. Can I get reverser alphabetical order ?
Although it doe snot make difference. it is useful for me to use beyond compare since the tool I use generate the file is reverser alphabetical order

andlaus, 2021-06-21:
Hm, this is a bit weird; I couldn't find a place in the DBC-writing code where any sorting is done. Could you please check if the messages are already sorted  after you read in your original DBC file? (i.e. is `cantools dump $YOUR_DBC` spitting out the messages in alphabetical order?)

That said, I'm pretty sure that reading a DBC followed by writing the resulting database object with cantools will result in an identical file featuring plenty of other differences if the original DBC file was not written by cantools as well. (Not sure if you want to do something like that.)

juleq, 2021-06-21:
CANdb++ (the dbc OG) uses a chronological order: Massages added last will appear first in the dbc file.

cantools uses Python lists to store messages and lists are an ordered type -> reading and dumping a dbc should retain the original order.

If you want to change that, you can just
- read dbc
- sort the messages list (Google for info on sorting by name property)
- dump dbc

You can do the same for signals.

If you have code control over your other tool: I would recommend mimicking the behavior of CANdb++. It is the original gangster...

malneni, 2021-06-21:
@juleq Thanks for the information. I am thinking to keep original file order while reading itself. may be I should explore mimicking the behavior of CANdb++

fstarnaud, 2021-08-03:
It seems that DBC writing does not sort signals alphabetically within messages, but in reverse list order, at least since this change 
4 years ago: https://github.com/eerimoq/cantools/commit/ad015e25daf420dd07c90d9b8371ebbd1b6c7a87, with this code in ` for signal in message.signals**[::-1]:**` in method `dbc._dump_messages()`

juleq, 2021-08-03:
Off the top of my head: I think it will try to retain the order as read in from dbc. This makes a ton of sense because 1.) CANdb++ orders chronologically and 2.) a lot of tests read in a dbc, dump it out again and verify line by line that the dbcs match. That is a straightforward way to test parser and composer on everything contained in the dbc and it would fail if cantools by default would reorder the messages. But if you want to change the order: See my hint above.

fstarnaud, 2021-08-03:
I tried that, but since `signals` is a property of `message`, I could not replace it. I tried sorting in-place, but that did not work for me. Instead, I took the approach of replacing the `_dump_messages(database)` method implementation using `cantools.database.can.database.dbc._dump_messages = types.MethodType(new_dump_messages, cantools.database.cas.database.dbc)`.

juleq, 2021-08-03:
Sorry, I mixed up messages and signals. The signals are sorted by start bit (internal to cantools and in dbc files).

As for sorting the signals: Since the Python community understands itself as consenting adults, there should be nothing stopping you from sorting the private _signals collection: message._signals.sort(key=name). Have not tried that code but something like that.

fstarnaud, 2021-08-04:
Right, and hence why I want to reverse the **signals** sort order in `dbc._dumpmessages(database)`.

My input DBC has signals sorted by `start` field from LSB to MSB, while cantools dumps the signals from MSB to LSB, in effect reversing the sort and making both DBC files harder to compare side-by-side.

Ah, right! Coming from more restrictive languages, I forget that I can access object private fields in Python, simply ignoring PyCharm's "Access to a protected member _signals of a class" warning.

This works for me and is much cleaner than the `MethodType` approach:
```
for message in new_dbc.messages:
    message._signals.reverse()
new_dbc_string = new_dbc.as_dbc_string()
```

Thanks!

juleq, 2021-08-04:
Welcome :).

`My input DBC has signals sorted by start field from LSB to MSB, while cantools dumps the signals from MSB to LSB, in effect reversing the sort and making both DBC files harder to compare side-by-side.`

I cannot confirm that. Reading in and dumping a dbc that has only seen CANdb++ before does produce identical outputs on my end. Just tried with cantools 36.2.0. If you have a sample representative of yours, I can run it through and check.

andlaus, 2021-08-17:
Seems like this issue has been resolved in `master`? Closing for now, feel free to re-open if it still persists.

closed 2021-08-17 by andlaus