cantools/cantools #310: Namespace issue for SHORT-NAME in ARXML

H4rryK4ne, 2021-06-25:
I get an Exception in Line 454 of arxml.py when parsing an ARXML with this snippet:
```
                <COMPU-METHOD UUID="626a0202-0000-0000-b658-00000006ea56">
                    <SHORT-NAME>MY_SHORTNAME</SHORT-NAME>
                    <COMPU-INTERNAL-TO-PHYS>
                        <COMPU-SCALES/>
                    </COMPU-INTERNAL-TO-PHYS>
                </COMPU-METHOD>

```
There shall be raised a ```ValueError``` but ```<Element '{http://autosar.org/schema/r4.0}SHORT-NAME' at 0x000001608AF9FB88>```
can not be found with:
```            
            if category is None:
                raise ValueError(
                    'CATEGORY in compu method {} does not exist.'.format(
                        compu_method.find('SHORT-NAME').text))
```
since it has a namespace ```{http://autosar.org/schema/r4.0}```

There are two issues:

1. The ```COMPU_METHOD``` has no ```CATEGORY```
2. The ```SHORT-NAME``` has a namespace

Here is a minumum example ARXML to reproduce the error(s): [compu_method_no_category.zip](https://github.com/eerimoq/cantools/files/6714308/compu_method_no_category.zip)

andlaus, 2021-06-25:
ok, I see. I will probably fix these issues next week. your file seems to still exhibit another issue, though:

```
$ python3 -m cantools dump compu_method_no_category.arxml 
error: ARXML: "Encountered dangling reference BASE-TYPE-REF: /BaseType/BT_MY_SIGNAL"
```

H4rryK4ne, 2021-06-25:
@andlaus I expected, that the file may be not working. I deleted as much as could, without suppressing the error, since the file is from our client.

Can you link the branch with the fix, so I can check with the original file.

I have not much knowledge about arxml format. So what is with the missing category. Is this an issue of cantools, or of my arxml?

andlaus, 2021-07-01:
@H4rryK4ne: can you test if #311 does the trick for you? (if possible please also check if the results are correct.)

closed 2021-08-16 by juleq