cantools/cantools #312: Cannot use cache file for database

thbuch, 2021-07-02:
Hi,
I am using Python 3.7.3 and Cantools 36.2.0

I am having problems to create a cache file for the database (.dbc) I load. 
If I load the file without a cache path then it works fine but it takes quite a lot to load the database (around 20s).
However, if I define a cache path, then I got the following error:
'can't pickle bitstruct.c.CompiledFormatDict objects' 

I supose that if cache path is set as argument,the library tries to load the cache file. If not found then loads the database file and creates the cache file. Is it correct?

Can I have a brief example?

Thank you!

juleq, 2021-07-02:
I am on that same version.

This works fine for me.
`
import cantools
db = cantools.database.load_file('vehicle.dbc', cache_dir='./cache')
`
Doing it twice will trigger restauration from cache if dbc is unchanged.

What is the command you are issuing?

thbuch, 2021-07-02:
I am using the following line in a class I created:
self.dbCH1 = cantools.database.load_file(filename= self.dbPath + dbFileName, cache_dir= self.dbPath + dbCacheFolder)

It starts creating the cache file in the specified cache path but after a few seconds i receive the exception:
 'can't pickle bitstruct.c.CompiledFormatDict objects'

juleq, 2021-07-02:
It seems that the diskcache package has trouble serializing the database object. When you google your error, you find an SO question suggesting a  relation to type changes when Python went from version 2 to 3.

I assume this needs fixing upstream in diskcache (if it has not already been fixed). You could try to dig into that.

thbuch, 2021-07-02:
Sorry, I am not sure if I got what you mean.

juleq, 2021-07-02:
The package that is used to store the database object to disk (into the cache) is called diskcache. And this package appears to choke on some dict type that cantools uses.

But since my trial did work, I think your dbc might contain just that bit of data that triggers diskcache to fail. It might be worth isolating what data it is. You could debug and step through the serialization (storage) process until it hits the error.

Unfortunately I do not have the time to dig into this. But it would be a nice exercise to troubleshoot and probably fix it ;).

thbuch, 2021-07-02:
Ok. thanks for that further explanation. What version of diskcache have you installed?

juleq, 2021-07-02:
5.2.1, should be the most recent.

thbuch, 2021-07-02:
Ouch! It was not updated. I had 4.1.0. Now I have updated diskcache and its dependencies but the problem remains.
The strange thing is that some time ago I could create the cache file with another database.
I have tested again with that database and now I am not able to create the cache file.
I did a upgrade all some time ago. I cannot remember the versions installed previously.... :(

juleq, 2021-07-02:
Maybe also update bitstruct.

thbuch, 2021-07-02:
Ok, I will update everything mentioned and I will be back with the results in some days.
I am using Raspian on a Raspberry Pi 3B+ if this has something to do.
Thanks

juleq, 2021-07-03:
Concerning platform: I just mentioned Windows for completeness sake. I don't really expect the issue originating from there. Would be a different story if some fs access had errored out.

In your case, the issue seems to be specific to package/language/type features.

thbuch, 2021-07-03:
I have upgraded the packages to the mentioned versions except python which remains in version 3.7.3. The problem is still the same: I can load the DBC  and work perfectly (it takes up to 20 seconds), but I cannot create it's cache file.

I found this python update a little bit risky for the rest of applications. Do you think it is completely necessary?

juleq, 2021-07-03:
Cannot be ruled out but not highly likely either (they are not far apart). I would probably first try the caching with some other dbcs, maybe the test dbcs from the cantools repo. If these work fine, try debugging (stepping through) cantools with your other dbcs to find the data that it chokes on...

thbuch, 2021-07-04:
Ok, I will check that as soon as possible. I will be a few days out.
Thanks a lot juleq

thbuch, 2021-07-26:
Hi, I have tested some of the dbc from the repo and still got the "can't pickle bitstruct.c.CompiledFormatDict objects" exception.
Same behaviour. It works fine if I don't use the cache's path.

andlaus, 2021-11-22:
since  eerimoq/bitstruct#24, should have addressed that issue, I'm closing it. feel free to re-open if the problem still persists...

closed 2021-11-22 by andlaus