cantools/cantools #313: Create new message with initial values taken from database

thbuch, 2021-07-02:
Hi,
I am using Python 3.7.3 and Cantools 36.2.0

I want to create a message taking the initial values from the database. But I don't now how to access the values.

Now I am creating a message with dataIni instead:
dataIni = [0, 0, 0, 0, 0, 0, 0, 0]

msgLWI_01 = dbCH0.get_message_by_name('LWI_01')
msgLWI_01_Simu = dbCH0.decode_message(msgLWI_01.frame_id, dataIni, decode_choices=False)

Can you help me with that? Thank you.

juleq, 2021-07-02:
I am not sure if there is a way to apply that automatically. You should inspect the contents of the dbc_specifics object that your db has. You will find the initial value in there and can apply it to the signal value manually.

thbuch, 2021-07-02:
Well, I got the idea (get signals from DB message, get initial value from every signal, and then loop every signal in newmessage and set value). I think it may work.
However, will be very useful a 'data' propery in DBMessage returning a list of bytes as used to decode.

Thanks for your help juleq!!

juleq, 2021-07-02:
To check that I understood your last point:

On a separate note, you propose this enhancement: Keep the raw byte array for the message after decoding for some other purpose later on.

Did I understand that the way you meant it?

thbuch, 2021-07-02:
Better an example :)

When I use this line:
msgLWI_01 = dbCH0.get_message_by_name('LWI_01')
I got a messageFrom the database. It would be fine to get the a list with the initial values (raw byte array) doing something like:
dataIni = msgLWI_01.data

so I could:
msgLWI_01_Simu = dbCH0.decode_message(msgLWI_01.frame_id, dataIni, decode_choices=False)
or
msgLWI_01_Simu = dbCH0.decode_message(msgLWI_01.frame_id, msgLWI_01.data, decode_choices=False)

Is more clear now?

juleq, 2021-07-02:
Ah, ok, you want to encode a message using the initial values where available. Yes, that is a reasonable request. As I said, I do not know if automatically applying of initial values is implemented. If not, you could propose an implementation.

Should be around ten lines of code effectively. Can be done either on db level iterating over all signals in all messages or, better, cascade down like: db.ApplyDefaultValuesWhereAvailable() calls that same method on every message and that on every signal.

thbuch, 2021-07-02:
Ok, I will open a request.
Thanks a lot for your help!

thbuch, 2021-07-26:
Well, I don't know how to open a request. I though it was similar to open an issue but it is not.
Anyway, I managed to solve the problem with a small function. I leave it here an example so if someone wants to use it or create a proper function in the library. 

Please, I accept suggestions of better code. And thanks again juleq.

#Example:

#Set initial values from given DB Message to given message
def fSetIniValues(msgDB, msg)
    for signal in msgDB.signals:
        if (signal.initial in not None):
            msg[signal.name] = signal.initial

#Usage:
dataIni = [0, 0, 0, 0, 0, 0, 0, 0]

msgLWI_01 = dbCH0.get_message_by_name('LWI_01')
msgDecoded = dbCH0.decode_message(msgLWI_01.frame_id, dataIni, decode_choices=False
fSetIniValues(msgLWI_01, msgDecoded)
#Set other values (not initial values)
#msgDecoded [signalName] = value
#Encode values to data
data = msgLWI_01.encode(msgDecoded)
#Create CAN message with correct/simulated data
msgLWI_01_Simu = can.Message(arbitrarion_id=msgLWI_01.frame_id, data = data, extended_id = msgLWI_01.is_extended_frame)

andlaus, 2021-07-26:
> Well, I don't know how to open a request. I though it was similar to open an issue but it is not.

it is, but with patches to the code attached. If you're not yet familiar with coding on git/github, https://guides.github.com/introduction/flow/ might help...

thbuch, 2021-07-26:
Thank you andlaus. As I understand, to open a request I need to make a commit first, so I need create a branch or fork the library in order to compare it with the master(for example). Am I right? I find this quite different from openning an issue. 
Obviously I am not familiar with coding on git and sincerelly I think I am still far from creating a proper request. Thank you anyway.

andlaus, 2021-07-26:
> As I understand, to open a request I need to make a commit first, so I need create a branch or fork the library in order to compare it with the master(for example). Am I right? 

basically yes: you first need to fork the repository to your personal github account via the "fork" button on the original repository (i.e., https://github.com/eerimoq/cantools/ ), then you clone that new repository to your local computer, followed by implementing the changes. After that, commit them locally on your machine (if possible into a branch not called `master`), push them to your personal github repository and finally open a pull request against the original repository. That said, note that a "fork" on github is basically equivalent to copying the respective repository to your personal playground, i.e., not what is most people understand when talking about forks in the open source context...

AlexDLSy, 2021-09-15:
Would this solve your request ?
```
def get_data_from_signals(sigs: dict, can_id: int,
                          dbc: database.can.Database):
    """Create frame data using signal dict and dbc.

    Set unspecified signals to their initial values."""
    msg_dbc = dbc.get_message_by_frame_id(can_id)
    for signal in msg_dbc.signals:
        if signal.name not in sigs.keys():
            val = signal.initial if signal.initial else 0
            sigs[signal.name] = val
    # Produce data
    data = dbc.encode_message(can_id, sigs)
    return data
```

This works for me, using cantools 36.4.0 : for each signal I get the initial value from dbc if it exists, else 0.

thbuch, 2021-09-15:
Thanks AlexDLSy! Now we have two different solutions. I will close the issue. Thank you all!

closed 2021-09-15 by thbuch

AlexDLSy, 2021-09-15:
Note that my solution does not work for multiplexed signals, as it will force the multiplexer signal to its initial value or 0.
I will edit my answer once I have a countermeasure for that usecase.