cantools/cantools #318: CDD : Overwriting DID 'XXXX' with 'XXXX' in the identifier to DID dictionary because they have identical identifiers

ashwin-tk, 2021-07-07:
Hi,
I am using Python 3.8 and Cantools 36.2.0

I want Fetch the cdd values so im using this command 
"db1 = cantools.database.load_file('D:\\Users\\f80460c\\Desktop\\FC NORM\\DUCATO.cdd')"

But as the output it overwrites the DID of the similar identifiers

Can you help me with that? Thank you.

ashwin-tk, 2021-07-07:
[DUCATO.zip](https://github.com/eerimoq/cantools/files/6774708/DUCATO.zip)

Mcthomas777, 2022-03-15:
Hi @ashwin-tk, I'm running through the same problem, did you find a way to avoid that ?

oomzay, 2022-03-15:
This looks like same issue as #290. After some investigation i concluded that CDD support needs substantial redesign. I have made a quick and dirty workaround for this in my fork by allowing caller to specify which class(es) they are intetested in to avoid the collisions.