cantools/cantools #323: Incorrect DBC format

kartikarcot, 2021-07-17:
https://github.com/eerimoq/cantools/blob/1bf3e850ddd79eced42548722fe758e9f29ee675/cantools/database/can/formats/dbc.py#L852

I think this has to be just the signal name and not the entire object signature. i.e.
it has to be `signal.multiplexer_signal.name` and not `signal.multiplexer_signal`

juleq, 2021-07-17:
I think you are probably right.

If so, ext multiplexing seems to be lacking test coverage as it slipped through.

I have never used it myself before because Vector requires additional licensing for tool support and so I usually try to stay away.

Would you care to do a PR that fixes it and adds a test that exercises the extended multiplex IO?

andlaus, 2021-08-17:
ok, judging from the comments for #326, it looks like this issue is basically a misunderstanding. I'll close this issue for now, if I did get something wrong, feel free to re-open it...

closed 2021-08-17 by andlaus