cantools/cantools #327: is CANFD supported?

hubiao7, 2021-07-26:
hello togethers:

i used version 33.1.1,but it can't parse canfd type from DBC? Is the newest version supported?

andlaus, 2021-08-17:
I don't know about the DBC support for CAN-FD (and particularly not in an old version like 33.1.1), but I've successfully loaded ARXML files containing descriptions of CAN-FD frames with the latest master version. If you want to interact with modern vehicles, there might be other problems like SecOC (which is out of cantools' scope), though...

messon007, 2022-06-17:
seems dbc parser work if use cbitstruct. however current code use bitstruct.c, which require Integers and booleans must be 64 bits or less.
refer https://pypi.org/project/bitstruct/ for detail.