cantools/cantools #329: how we generate Ros message from "dbc " file

pinnakarahul, 2021-07-30:


simontegelid, 2021-07-30:
I had this exact need a while ago and solved it with mako templates as made possible with #151, it was never merged though. Maybe we should consider including a template tool to handle these issues. What do you think?

pinnakarahul, 2021-07-30:
I have  no idea about  mako. I am at starting level of using the ROS and PI can

juleq, 2021-07-30:
Just one perspective: I would share the maintenance concerns that Eric stated in #151.

I had the same concern when the matplotlib integration came along and broke cantools for some folks.

Maybe there is a way to offer a plugin interface for such developments. Core maintenance might get tough otherwise.

simontegelid, 2021-07-30:
> Just one perspective: I would share the maintenance concerns that Eric stated in #151.
>
> I had the same concern when the matplotlib integration came along and broke cantools for some folks.
>
> Maybe [...]

Yes, I agree with that. Ultimately I would like to break out the core parts (parsing of dbc, arxml, etc) of cantools to its own library and make everything else their own projects. Log parsing is better in python-can, and support for various hardware interfaces is also better in python-can, but what's missing there is the DBC-stuff. Not sure if that's in line with the plans for cantools though.

andlaus, 2021-07-30:
I also agree: IMO cantools should become a collection of stuff which is annoying to implement but required when interacting with CAN busses; i.e., loading of bus description databases and message de-/encoding and -- maybe -- some simple examples of how to use this functionality like `cantools dump`.  Everything else (plotting, bus monitoring, code generation, ...) should ideally go to separate repositories build ontop the core module. 

BTW: personally I would prefer using [jinja](https://jinja.palletsprojects.com/en/3.0.x/) to mako for code generation, but as said above, I don't think that code generation should be covered by the core library in the first place...

> Not sure if that's in line with the plans for cantools though.

As I see it, the plans for cantools are whatever we decide them to be -- apart from stuff vetoed by our [BDFL](https://github.com/eerimoq)?

pinnakarahul, 2021-08-02:
https://github.com/Black-Forest-Formula-Team/bfft_can_bus_msgs_to_ros_topic

is it helpfull...?

simontegelid, 2021-08-02:
It seems like that repo just contains a static ROS message with floats, so it will not help you with unpacking CAN frames as described by a DBC. I plan to write you a gist that you can use to generate messages, havent had the time yet though.

pinnakarahul, 2021-08-30:
> It seems like that repo just contains a static ROS message with floats, so it will not help you with unpacking CAN frames as described by a DBC. I plan to write you a gist that you can use to generate messages, havent had the time yet though.

can you help me...?
how to generate messages..?

closed 2021-09-23 by pinnakarahul