cantools/cantools #330: Database doesn't include some values from DBC

isaacwasserman, 2021-08-03:
When I initialize a database with:
```
db = cantools.database.Database()
db.add_dbc_string(dbcFile.read())
```
`db` doesn't include the signals listed in the `.dbc` with the enumeration type `_val`. For example, after reading in [this dbc](https://raw.githubusercontent.com/jmscslgroup/strym/master/strym/dbc/toyota_rav4_hybrid.dbc), `db` should include an entry for the line `VAL_ 295 GEAR 0 "P" 1 "R" 2 "N" 3 "D" 4 "B";` but instead, the only entry for message ID `295` is: `message('UNKNOWN295', 0x127, False, 7, None)`. Any ideas what's going on? I'm trying to identify when the vehicle is in reverse.

juleq, 2021-08-03:
VAL is the definition of a value table with name GEAR, associated with message 295. In cantools value tables are called choices, you could read up on them in the doc to understand how you use them.

An example for a use case would be: You show a user a UI control to set the value of the signal. But instead of presenting rather meaningless numbers, you use the table to translate to meaningful gear abbreviations.

isaacwasserman, 2021-08-03:
Thanks for your response. My understanding is that messages that have the ID 295 should contain information about the gear status in the form of an integer 0-4. However, when I use `db.decode_message(295, message)`, I get `{'UNKNOWN295_1': 48, 'UNKNOWN295_2': 48, 'UNKNOWN295_3': 49}`. None of these three values seem to indicate the gear status. Am I using the tool incorrectly?

juleq, 2021-08-03:
Yes, you understood that correctly. You got suspicious looking values for suspicious looking signal names in a suspicious looking named message :).

But searching for GEAR throughout your dbc yields:

BO_ 956 GEAR_PACKET: 8 XXX
 SG_ SPORT_ON : 2|1@0+ (1,0) [0|1] "" XXX
 SG_ GEAR : 13|6@0+ (1,0) [0|63] "" XXX

Maybe that is the signal you are looking for?

isaacwasserman, 2021-08-03:
Yes, very suspicious 🙃.
Unfortunately, messages with the ID 956 have a similar problem. Decoding those messages yields: `{'SPORT_ON': 0, 'GEAR': 48, 'SPORT_GEAR': 3, 'SPORT_GEAR_ON': 0, 'DRIVE_ENGAGED': 0, 'ECON_ON': 0}` It seems that the relevant field, `GEAR` is related to `UNKNOWN295` and always seems to return `48` (not 0-4).

juleq, 2021-08-03:
Can you change gears to monitor change in signals? If you trace the vehicle CAN whilst gearing up and down it should be easy to find.

juleq, 2021-08-03:
Just to double check: When you issue db.decode_message(295, message), are you sure that you are passing the right message? Also for 956?

isaacwasserman, 2021-08-03:
> Can you change gears to monitor change in signals?
Unfortunately, I don't have access to the vehicle from which the data was collected.
> are you sure that you are passing the right message?
Yes.

juleq, 2021-08-04:
Maybe it is SPORT_GEAR. If you do not have a dbc that is a valid fit for your vehicle, it comes down to guess/trial work (or maybe googling). But cantools cannot help you with that ;).