cantools/cantools #332: Error executing cantools (arxml.py)

IkerDC, 2021-08-13:
When running a script that uses cantools, right away it gives this error
```
Traceback (most recent call last):

  File "./ea_psb-x_manager.py", line 5, in <module>
    import cantools
  File "/home/pi/.local/lib/python3.5/site-packages/cantools/__init__.py", line 9, in <module>
    from . import database as db
  File "/home/pi/.local/lib/python3.5/site-packages/cantools/database/__init__.py", line 6, in <module>
    from . import can
  File "/home/pi/.local/lib/python3.5/site-packages/cantools/database/can/__init__.py", line 1, in <module>
    from .database import Database
  File "/home/pi/.local/lib/python3.5/site-packages/cantools/database/can/database.py", line 3, in <module>
    from .formats import arxml
  File "/home/pi/.local/lib/python3.5/site-packages/cantools/database/can/formats/arxml.py", line 25
    raise ValueError(f"No XML namespace specified or illegal root tag name '{self._root.tag}'")
                                                                                             ^
SyntaxError: invalid syntax
```
I can't find no one with a similar error. Have already tried installing/unistalling with pip and pip3. For context, I am running it in a RevPi.

juleq, 2021-08-13:
Disclaimer: Never used arxml.

Can you post your arxml? It seems to be invalid. The namespace tag should be the root node in the document (the very first line). And it seems to stumble there already.

juleq, 2021-08-13:
From a sample file:

‘
<?xml version="1.0" encoding="UTF-8"?>
2	
3	<AUTOSAR xmlns="http://autosar.org/schema/r4.0" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xsi:schemaLocation="http://autosar.org/schema/r4.0 AUTOSAR_4-0-3_STRICT.xsd">
‘

IkerDC, 2021-08-13:
Sure, actually the file is quite large, however, this is the first part where it already seems to fail
```
# Load and dump a CAN database in ARXML format.

import re
import logging
from decimal import Decimal

from xml.etree import ElementTree

from ..signal import Signal
from ..signal import Decimal as SignalDecimal
from ..message import Message
from ..internal_database import InternalDatabase

LOGGER = logging.getLogger(__name__)

class SystemLoader(object):
    def __init__(self, root, strict):
        self._root = root
        self._strict = strict

        m = re.match('^\\{(.*)\\}AUTOSAR$', self._root.tag)

        if not m:
            raise ValueError(f"No XML namespace specified or illegal root tag name '{self._root.tag}'")

        xml_namespace = m.group(1)
        self.xml_namespace = xml_namespace
        self._xml_namespaces = { 'ns': xml_namespace }

        m = re.match('^http://autosar\.org/schema/r(4\..*)$', xml_namespace)
        if m:
            # AUTOSAR 4
            autosar_version_string = m.group(1)
        else:
            raise ValueError(f"Unrecognized AUTOSAR XML namespace '{xml_namespace}'")

        m = re.match('^([0-9]*)(\.[0-9]*)?(\.[0-9]*)?$', autosar_version_string)
        if not m:
            raise ValueError(f"Could not parse AUTOSAR version '{autosar_version_string}'")

        self.autosar_version_major = int(m.group(1))
        self.autosar_version_minor = 0 if m.group(2) is None else int(m.group(2)[1:])
        self.autosar_version_patch = 0 if m.group(3) is None else int(m.group(3)[1:])

        if self.autosar_version_major != 4:
            raise ValueError('This class only supports AUTOSAR version 4')

        self._arxml_reference_cache = {}
```

juleq, 2021-08-13:
I meant your arxml file. It seems to be missing the AUTOSAR tag.

IkerDC, 2021-08-13:
Oh yeah I see, the error should be in the arxml file. I however have to find the file it is taking because like I am just using cantools to load a .dbc file and use it, i am not using or loading any arxml myself

juleq, 2021-08-13:
So you do not use arxml at all? Please post your invokation of cantools so that we can see what you are trying to do.

IkerDC, 2021-08-13:
I mean, just doing the import cantools fails, without doing anything else at all. Without import cantools the code works, adding ```import cantools``` it fails

juleq, 2021-08-13:
Could you post your code? I would not expect that part of cantools getting invoked on import at all.

juleq, 2021-08-13:
Also: type python to get a repl, then type import cantools and see if that throws.

IkerDC, 2021-08-13:
Oh it already fails, got:
```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "build/bdist.linux-armv7l/egg/cantools/__init__.py", line 4, in <module>

  File "build/bdist.linux-armv7l/egg/cantools/tester.py", line 15, in <module>
  File "/usr/local/lib/python2.7/dist-packages/python_can-4.0.0.dev0-py2.7.egg/can/__init__.py", line 16
    rc: Dict[str, Any] = dict()
      ^
SyntaxError: invalid syntax
```

juleq, 2021-08-13:
Ah. Invalid syntax. Probably the f string. Try updating python3 please. E.g. I am on 3.8.2.

f strings are supported in Python since 3.6. Thats your issue.

IkerDC, 2021-08-13:
Dang, there it is! I will update the version and it should work. Thanks a lot for you help! :)

closed 2021-08-13 by IkerDC