cantools/cantools #333: initial value and big endian start bit problem.

PUNITPNT, 2021-08-14:
hi,
My Requirement:
I am converting DBC to Excel, according to candbc++ editor excel format using python 3.9.
cantools latest version:

Problem 2:
big endian start bit is different from as it present in vector Candbc++ editor.

Problem 3:
i am getting initial value from cantools as NONE , but initial value present in dbc and visible in  vector Candbc++ editor.

juleq, 2021-08-14:
Problem 1 is not a problem. CANdb++ in its standard setting will display the start of a signal different from what the representation in the dbc file looks like. You can go into the settings and inspect the options there, but I would not recommend changing.

```

// Little endian
//   DBC     Startbit  : Bit numbering -> LSB0, Reference bit -> LSB
//   CANdb++ Startbit  : Bit numbering -> LSB0, Reference bit -> LSB
 
// Big endian
//   DBC     Startbit  : Bit numbering -> LSB0, Reference bit -> MSB
//   CANdb++ Startbit  : Bit numbering -> LSB0, Reference bit -> LSB
 
// Convert bit numbering (both ways)
// b = b - (b % 8) + 7 - (b % 8)
 
// Convert reference bit LSB to MSB (when bit numbering is MSB0)
// b = b + 1 - length
 
// Convert reference bit MSB to LSB (when bit numbering is MSB0)
// b = b + length - 1

```

For problem 2: Some code and a dbc would probably required to help.

PUNITPNT, 2021-08-14:
hi 

Thanks for information for problem 1 .
I need  to convert cantools big endian start bit (msb to lsb ) to match vector Candbc++ editor value.

problem 2.

some of initial value of signal , I get as none type,but in candbc editor initial value is present.
when value will be of None type ? 

initial value of signal from cantools not match with Candbc++ editor .

how to match can tools value to Candbc++ editor value ?

how to convert physical to raw, considering all types in can tools ?

[image]

andlaus, 2021-08-14:
@juleq probably wanted to say that we would appreciate if you boiled down your DBC file to something which shows the problem  and that can be shared freely (i.e., containing no confidential data). If you do so, I volunteer to try to fix the issue and to also create a unit test so that it will not re-appear in the future. (see #311 for an example of how this is supposed to work.)

juleq, 2021-08-14:
Yes. Further, since you are already debugging, you could inspect the dbc_specifics object. You should find the attribute GenSigStartValue associated with your signal in the attributes there. If so, the issue would be that the initial value attribute is properly read when parsing the dbc, but not applied to the respective signal property when building up the internal db. Please check if that might be the case here.

PUNITPNT, 2021-08-14:
@juleq There is not attribute defined for the signal.
[image]

but GenSigStartValue is present in attribute_definitions.
[image]

in attribute definition it contain default values.

@andlaus I will try to create sample dbc and try to reproduce the same issue.

PUNITPNT, 2021-08-15:
hi

Initial value : None
sample dbc attached. I have reproduced the same issue. 
please check I am getting initial value as None. 
[SampleDBC.zip](https://github.com/eerimoq/cantools/files/6988380/SampleDBC.zip)

start bit :
can tools will give startbit as per below
Big endian = > DBC     Startbit  : Bit numbering -> LSB0, Reference bit -> MSB

for conversion as per CANdb++ Startbit , what exactly formula need to use ?

b = startbit from cantools ?

b = b - (b % 8) + 7 - (b % 8)
b = b + length - 1

this not matching with the value of CANdb++ Startbit

juleq, 2021-08-15:
Try this function to convert the start bit representation from dbc to CANdb++. It is C# but easy to adapt:

```
private static uint GetStartBitInVectorCANdbppStyle(Signal signal)
{
  uint startbit = (uint)signal.Start;
  if (signal.IsBigEndian)
  {
    startbit += 7 - 2 * (startbit % 8);
    startbit += (uint)signal.Length - 1;
    startbit += 7 - 2 * (startbit % 8);
  }
  return startbit;
}
```
Maybe I get a chance to look into your dbc tomorrow.

juleq, 2021-08-16:
The dbc does contain an attribute definition for the signal type (SG_) that is of value type FLOAT and has a default value of 0. However, there is no signal in the dbc that has this attribute set to it. So either your example is missing that last step or everything is fine. It might be that Vector tools fall back to this default if there is nothing specificly defined for a signal. But I wonder what their logic would be in this case since the signal you have there is an integer type but the default is FLOAT...

[image]

PUNITPNT, 2021-08-16:
> Try this function to convert the start bit representation from dbc to CANdb++. It is C# but easy to adapt:
>
> ```
> private static uint GetStartBitInVectorCANdbppStyle(Signal signal)
> {
>   uint startbit = [...]

Thanks for the information , I understood the formula.

PUNITPNT, 2021-08-16:
> The dbc does contain an attribute definition for the signal type (SG_) that is of value type FLOAT and has a default value of 0. However, there is no signal in the dbc that has this attribute set to [...]

hi ,
Thanks for information.

if GenSigStartValue attribute not set to the signal, 
then Cantools will make initial value as None.

if None then vlaue will be 0: raw =0

so iv =  (raw * factor ) + offset

so iv will become equal to offset 

Is my understanding Correct ?

juleq, 2021-08-16:
Do you mean when encoding a message without setting the signal value? I could walk through the code to answer that but it is probably easier you try it or even step through the encoding process with your debugger.

PUNITPNT, 2021-08-16:
@juleq 
yes I will  try with debugger, 

but I want to understand at what condition cantools return initial value as None  ?

PUNITPNT, 2021-08-16:
I want to write all signal data into excel.

As i am getting signal.initial  value as None. 

[image]

so because of None , nothing is written in excel sheet for that particular signal.

juleq, 2021-08-16:
I would argue that this is a reasonable behaviour. Vector seems to fall back to the generic attribute which is FLOAT 0, but I think that that is a surprising choice since the signal is of integer type. I don´t really get their reasoning behind that. But it might also be that I am misinterpreting what I saw in CANdb++ and Vector will default to uint 0 when so generic initial integer value is defined.

If you also want to default to 0, just check if value is None and replace that with 0 on export.

PUNITPNT, 2021-08-16:
@juleq 
ya I understood your explanation and I agree to it.

I just created sample dbc for one signal, as i will try changing the FLOAT and check.

PUNITPNT, 2021-08-18:
hi @juleq 

I changed the value FLOAT to INT , still i got None .

what I observed is when "GenSigStartValue"  attribute not set to signal, that time cantools return value as None .

and with respect to vector tool , when  "GenSigStartValue"  attribute not set to signal, it fall back to default value defined in attribute definitions.

So whenever I get initial value as none from cantools , then I taking default value of GenSigStartValue from attribute defintion.

juleq, 2021-08-18:
Sounds reasonable to do so.

But for cantools in general, I would argue that such a generic fallback would not be mandatory. When you do a residual bus simulation you should probably initialize the signal values or define the initial value for each specific signal to avoid surprises due to a generic fallback. But it is a design choice to do one or the other...

PUNITPNT, 2021-08-23:
Hi,
@juleq with respect to current requirement to display vector data in excel sheet without use of vector tool ,I think this approach it will work.
Thanks for your information.

I have additional requirement to this.
REQ : To get Japanese character as it is from dbc and write Japanese character to excel .

as my dbc will contain some Japanese character in comment attribute.
when I read and write comments attribute  in excel , it will display some unreadable characters for Japanese letter.

Can you tell How to handle this Japanese characters ?

juleq, 2021-08-23:
I think cantools has a parameter to define encoding if that is an issue. I assume it should be UTF-8 bit check that using a capable editor that you can load your dbc into to find out what encoding it uses.

I would assume that cantools uses utf-8 internally, but have not checked.

I do not know how you output, but I think you could define encoding there to. Your mileage may vary on output, not sure if you use text-based output or some OLE / ODBC stuff to get to excel. 

It might be painful but having a recent Python and its packages at your disposal will certainly help if you need to fix stuff along the way (e.g. I vaguely remember using iconv for some encoding matters...).

PUNITPNT, 2021-08-23:
hi @juleq
Coding language : python 3.9
editor : Pycharm

I read dbc using cantool package and store in python dictionary.
I use openpyxl to write dictionary data to excel .

Ex:
worksheet.cell(row=i, column=j).value = "Data"

if Cantools use utf 8, then it able to read Japanese character ?

I am getting this from cantools:
[image]

same thing it is written to excel.

juleq, 2021-08-23:
Looks like you need to specify the encoding when reading in. Check the docs for cantools to see how to do that.

Find out what encoding to use by opening dbc in editor.

As far as I see utf-8 does contain most Japanese chars, but not all (?). So it might mostly work out of the box as long as you get the encoding parameters right. 

A good resource might be googling for handling of Japanese text with Python in general. You will come across usual encodings and packages for handling such stuff that way.

PUNITPNT, 2021-08-23:
hi @juleq 

I gone through cantools doc,  I changed encoding to cp932 for Japanese Characters  while reading DBC.

self.db = cantools.database.load_file(self.path,  encoding ="cp932")

Great Cantools took care of encoding part also.

Thanks for your Quick Reply and suggestions.

PUNITPNT, 2021-08-25:
hi @juleq 

when I testing my dbc to excel script, I observed that,
cantools returns None for Min and Max value ?

at what condition Cantools return None for min and max value ?

EX: For this signal
 SG_ xxxx : 232|5@1+ (1,0) [0|0] "NA"  abc

Note : Vector gives min =0 max =0

juleq, 2021-08-25:
Cantools is not specific to any db format.

The issue sou are seeing is that dbc represents None with 0 in this case, because its syntax has no other means to represent None.

You can always look into the dump function for dbc to see the way cantools does dbc specific replacements.

PUNITPNT, 2021-08-25:
hi @juleq 

it means ,
if Cantools return None for min and max value , then its ok to consider 0 
None = 0 

is this ok ?

I observed in vector min = 0  max =0
I got none in cantools,

I observed in vector min = 1  max =1
can tools give min = 1 , max =1

juleq, 2021-08-25:
Yes, if min and max are 0 that is the dbc files way of saying None (not defined).

PUNITPNT, 2021-08-25:
hi @juleq 

thanks for clarification . 

I will write 0 to excel if I got min and max as none.