cantools/cantools #334: Doesn't support comment symbols in DBC file

vkaser, 2021-08-17:
I am trying to use a dbc file with comment characters '`(* *)`' and `cantools` stop working with the error:

`error: DBC: "Invalid syntax at line 5113, column 3: "( >>!<<*""`

juleq, 2021-08-17:
What _exactly_ does the comment look like?
What command are you issuing that is failing?

cantools matches strings in comments as:
```
r'"(\\"|[^"])*?"'
```

You can check if that would fail by entering into regex101.com using the Python flavor.

vkaser, 2021-08-18:
> What _exactly_ does the comment look like?

I added comments at arbitrary locations in a DBC file using `(*` ... `*)` notation, but as it turned out this symbols are used only for comments in the [documentation describing the DBC file syntax](https://www.dropbox.com/s/eds308znwazljs5/DBC_File_Format_Documentation.pdf). I realized that I could  put explanatory comments in the comments section of the database itself, using the `CM_` notation. And it works.

> What command are you issuing that is failing?

`$ cantools monitor file_with_comments.dbc`

closed 2021-08-18 by vkaser