cantools/cantools #337: Decode response based on the dbc type?

IkerDC, 2021-08-19:
Hi,
I'm using a cantools to communicate with a RevPi. Is it possible to decode a message/signal returned based on the .dbc file format?
For instance, I have a message called ```Query_Object``` that contains a Multiplexor called ```Query_Mux```, now, lets say I want to read the value of one of the signals named ```Actual Current``` that in the Multiplexor has a reference of 508. I encode the message:
```
data = example_message.encode({'Query_Mux': 508})
message = can.Message(arbitration_id=example_message.frame_id, data=data)
``` 
As in the examples. Then I capture the response, however if I decode the message following the example:
```
message = can_bus.recv()
db.decode_message(message.arbitration_id, message.data)
```
It decodes  ```{Query_Mux: Actual Current}```. Is there a way of decoding what the RevPi returns, i.e. the numeric value of ```Actual Current``` and using the format that the signal has on the .dbc. In other words, the are some signals where the data types have a more complex representation such as floats, so can you decode somehow saying this is the kind of object you are decoding?

juleq, 2021-08-19:
Hi.

Just to ensure: You are using separate ids for the query message and for the response message, right?

The way you do it should already return you the signal value.

When it does not, I can think of two causes:

1.) Same id for query and response -> It is not a good idea in CAN networks to send the same id from two nodes.

2.) Something is wrong with your response message definition. To look into this, you could post your query and response definitions (assuming dbc that would be the BO_ and their SG_ enries. If query and response objects are large, you could strip out the signal definitions that are not relevant (signals multiplexed with other mux value).

IkerDC, 2021-08-19:
Actually I managed to kinda figure it out, the ID were correct, I had 2 for read and 1 for query, however there was some other problems that I finally managed to avoid: 
* First is that when a captured the message, it was of type can.message.Message and not can.Message and thus raised an error. To solve this I just retrived the data and called the decode with the known id for Read (i.e. 2) and the data as a bytearray.
*  Then, for some reason there are mismatches between the DLC of the messages in the .dlc and what the RevPi returns, thus it usually returned less bits than expected as a consequence it raised ```Short data```. To solve this I manually added zeros (not beautiful but seems to work).

I guess I was just getting those and other errors and thought that it may wasn't possible to decode that ways, but no it is correct. Thank you!

closed 2021-08-19 by IkerDC