cantools/cantools #339: how to write python script which read the DBC file

pinnakarahul, 2021-08-31:
My dbc file is this
VERSION "1.0"
NS_ : 
    NS_DESC_
    CM_
    BA_DEF_
    BA_
    VAL_
    CAT_DEF_
    CAT_
    FILTER
    BA_DEF_DEF_
    EV_DATA_
    ENVVAR_DATA_
    SGTYPE_
    SGTYPE_VAL_
    BA_DEF_SGTYPE_
    BA_SGTYPE_
    SIG_TYPE_REF_
    VAL_TABLE_
    SIG_GROUP_
    SIG_VALTYPE_
    SIGTYPE_VALTYPE_
    BO_TX_BU_
    BA_DEF_REL_
    BA_REL_
    BA_DEF_DEF_REL_
    BU_SG_REL_
    BU_EV_REL_
    BU_BO_REL_
    SG_MUL_VAL_

BS_:

BU_: PCM1 FOO

BO_ 496 ExampleMessage: 8 PCM1
 SG_ Temperature : 0|12@0- (0.01,250) [229.52|270.47] "degK"  PCM1,FOO
 SG_ AverageRadius : 6|6@0+ (0.1,0) [0|5] "m" Vector__XXX
 SG_ Enable : 7|1@0+ (1,0) [0|0] "-" Vector__XXX

CM_ BO_ 496 "Example message used as template in MotoHawk models.";

VAL_ 496 Enable 0 "Disabled" 1 "Enabled" ;

### 

****I need to get output as following like
float64 Temperature
float64 AverageRadius****

oaoren, 2021-09-02:
Iterate the Whole file, each line starting with "SG_ " is a signal. The second string is the name of the signal.

Something like this:

`    
file_obj = open(file, 'r', errors='ignore')
Lines = file_obj.readlines()

    for line in Lines:
        if line.startswith(' SG_'):
            line = line.replace("SG_", "", 1)
            list = line.split(' ')
            sig_name = list[2]
            sig_name_plain = sig_name
            sig_name = msg_name+"__"+sig_name
            datatype = 'float64' # or whatever logic you need
            print(f'{datatype} {sig_name}', file=out_file)
`

pinnakarahul, 2021-09-06:
*can you please help me with the following:*

*My dbc file is this*
VERSION "1.0"
NS_ :
NS_DESC_
CM_
BA_DEF_
BA_
VAL_
CAT_DEF_
CAT_
FILTER
BA_DEF_DEF_
EV_DATA_
ENVVAR_DATA_
SGTYPE_
SGTYPE_VAL_
BA_DEF_SGTYPE_
BA_SGTYPE_
SIG_TYPE_REF_
VAL_TABLE_
SIG_GROUP_
SIG_VALTYPE_
SIGTYPE_VALTYPE_
BO_TX_BU_
BA_DEF_REL_
BA_REL_
BA_DEF_DEF_REL_
BU_SG_REL_
BU_EV_REL_
BU_BO_REL_
SG_MUL_VAL_

BS_:

BU_: PCM1 FOO

BO_ 496 ExampleMessage: 8 PCM1
SG_ Temperature : ***@***.*** (0.01,250) [229.52|270.47] "degK" PCM1,FOO
SG_ AverageRadius : ***@***.***+ (0.1,0) [0|5] "m" Vector__XXX
SG_ Enable : ***@***.***+ (1,0) [0|0] "-" Vector__XXX

CM_ BO_ 496 "Example message used as template in MotoHawk models.";

VAL_ 496 Enable 0 "Disabled" 1 "Enabled" ;

I need to get output as following like read and print from dbc file
float64 Temperature
float64 AverageRadius

pinnakarahul, 2021-09-06:
sorry. i did get you fully.

can you please say me how i can do that

oaoren, 2021-09-06:
First let me see if I understand you correctly:

You want to extract the name of all the "signals" in the file and print them. Is that right?

(Another question is why you wa to print float64 next to the signal name. The signals you refer to is not encoded as 64bit floats so I'll just assume you want to treat them as 64floats in some application.)

pinnakarahul, 2021-09-06:
Yes, I need to extract the signals from DBC file.
 Then the message file will created.

From that message file I am going use in ROS network

oaoren, 2021-09-06:
if your dbc file is called dbc.dbc and you run the following script in the same folder then you should get an output file corresponding to your request. 

`
out_file = open('./output.txt', 'w', errors='ignore')
file_obj = open('./dbc.dbc', 'r', errors='ignore')

Lines = file_obj.readlines()

for line in Lines:
    if line.startswith(' SG_'):
        line = line.replace("SG_", "", 1)
        sig_name = line.split(' ')[2]
        datatype = 'float64' # or whatever logic you need
        print(f'{datatype} {sig_name}', file=out_file)

`

pinnakarahul, 2021-09-06:
Thank you.

from code last line I am getting invalid syntax

juleq, 2021-09-06:
Update to Python >= 3.6 to get f string support or change the line to use classic strings.

pinnakarahul, 2021-09-06:
I am using in linux 
and Thonny

oaoren, 2021-09-06:
I updated my comment so that it should work for you.

pinnakarahul, 2021-09-09:
Thank you once again.
but still in last line it was showing suntax error

oaoren, 2021-09-09:
What version of python are you using?

pinnakarahul, 2021-09-09:
Thonny python IDE

pinnakarahul, 2021-09-09:
Sorry I got it 
Thank you so much for your help...

Have a nice day...:)

pinnakarahul, 2021-09-21:
hello again. I need one help

how I can get CAN open...?

juleq, 2021-09-21:
1.) That is a different issue. You should open a new one.
2.) Try formulate your question in your language and translate that using Google
3.) If you are referring to the CANopen protocol: cantools has no support for it
4.) Usually you purchase a CANopen stack for your embedded system (generic ANSI C stack) and integrate that. There might be open implementations. I think you would have to have a membership to CiA to obtain a copy of the specification for your own implementation
5.) Since cantools has no support for CANopen, you might be asking at the wrong location

closed 2021-09-21 by pinnakarahul