cantools/cantools #344: CANopen

pinnakarahul, 2021-09-22:
Hello , 

How can I get the CANOpen..?
Where should I learn that....?

Thank you.

AlexDLSy, 2021-09-23:
CANOpen protocol is similar to CAN protocol, but it has more subtlety than regular CAN. 

- In a CANOpen network, each node has a communication ID, called COB-ID
- The lowest COB-ID is the network master : other nodes are not slaves but they cannot do all of what the master can do
- The CAN ID of the CANOpen frames contain the COB-ID as last bits
- The network nodes do not know about the frames/message list  --> At the beginning of every node power up, the master must send them some information about the messages they will receive or should send (through SDO).
- There are 3 main protocols : LSS, SDO and PDO 
    - LSS is for communication setup 
    - SDO is Service Data Object : it is used at the beginning of the transfer to setup communication between nodes
    - PDO is Protocol Data Object : it is used to send information, based on an object dictionary. Each object in the dictionary is a variable with an ID.
- There are many other things which are special to CANopen (I cannot explain everything here)

Currently, cantools does not support CANOpen : if you want to use CANOpen, you must add your own code on top of cantools in order to make the adaptation.
You could maybe create a dbc file with a CANOpen template using CANdb++ software (if you have it)  : it would help by creating some protocol signals automatically, then you could use cantools as a way to fill each signal.

If you want to learn about CANOpen, please look at the following websites :
[Wikipedia](https://en.wikipedia.org/wiki/CANopen)
[CiA website](https://www.can-cia.org/canopen/#:~:text=CANopen%20is%20a%20CAN%2Dbased,systems%2C%20such%20as%20handling%20systems.)
[CSS Electronics](https://www.csselectronics.com/pages/canopen-tutorial-simple-intro)

andlaus, 2021-11-22:
I suppose this feature is not applicable to cantools in its current form. closing...

closed 2021-11-22 by andlaus