cantools/cantools #348: Multiplexer signals with the same name are not supported

yesralph, 2021-09-29:
Hello,

The following scenario seems to be normal but would end up with error:
```
BO_ 1111 DebugMessage: 2 ECU
 SG_ ECU_DBG_Multiplexer M : 0|4@1+ (1,0) [0|0] "" DBG
 SG_ ECU_DBG_m0s0 m0 : 4|4@1+ (1,0) [0|0] "" DBG
 SG_ ECU_DBG_timestamp m0 : 8|8@1+ (1,0) [0|0] "" DBG
 SG_ ECU_DBG_m1s0 m1 : 4|2@1+ (1,0) [0|0] "" DBG
 SG_ ECU_DBG_timestamp m1 : 6|8@1+ (1,0) [0|0] "" DBG
```
Error:
```
cantools.database.errors.Error: The signals ECU_DBG_timestamp and ECU_DBG_m0s0 are overlapping in message DebugMessage.
```

The reason is that `Message.get_signal_by_name()` only returns the first signal found in `Message._signals` and it mistakes `ECU_DBG_timestamp m0` for `ECU_DBG_timestamp m1`.

It would be nice if it can be supported.

Thank you!

juleq, 2021-09-30:
You are right, cantools requires signal names to be unique within the signal collection of a message since the name is used as key.

Is using the same name multiple times within a message a common use case for you?

Have you checked that CANdb++ does support that? I have not yet.

How would you propose to distinguish between the signals?

yesralph, 2021-10-01:
Hi @juleq 

I found `Signal.multiplexer_ids` is a list type and assumed a multiplexed signal can be mapped to multiple multiplexer IDs.

> Is using the same name multiple times within a message a common use case for you?
Not really. I just came across the situation to include timestamp in all multiplexed signals.

> Have you checked that CANdb++ does support that? I have not yet.
Here is the warning I got from CANdb++. It is supported but probably not recommended.
[image]

> How would you propose to distinguish between the signals?
We can pass the multiplexer ID to `get_signal_by_name()`. As for the comments or attributes, I would have to agree with CANdb++ warning message.

AlexDLSy, 2021-10-01:
In CANdb++ it is possible to have non-multiplexed signals inside a multiplexer frame.
Here is the example I have :
[image]

I don't know how cantools handles that case, but you should be able to avoid the CANdb++ warning and keep unique signal names with such a mapping.

juleq, 2021-10-01:
@yesralph So CANdb++ does warn in that case. I think this to be expected because the dbc format is not fit to handle such situation: Comments, attributes and the like reference their signal by message id and signal name within the file. They do not contain further information on multiplexers. Cantools basically follows that strategy.

I would also argue that if the CANdb++ consistency check reveals issues, they might be worth fixing.

yesralph, 2021-10-01:
I agree there are issues regarding signal comments and attributes and it's better to avoid such name collision. Having said that, why is `Signal.multplexer_ids` a list instead of a `None`/`int`? In which scenario a signal can have multiple multiplexer IDs within a message?
```
commit 57e7cfe9da509d184f91937baf390e6d9847f49d
Author: Erik Moqvist <erik.moqvist@gmail.com>
Date:   Fri Jan 5 14:45:03 2018 +0100

    Allow multiple multiplex ids per signal.
```

Also, if unique signal name is somewhat required by `cantools`, I think we should detect it and throw out warnings/errors if signals with the same name are found within a message. Currently, it only reports error is the signal overlaps with other signals. For example, the follow situation was not detected and handled, even though these two signals with the same name are different in offset and length:
```
BO_ 1111 DebugMessage: 2 ECU
 SG_ ECU_DBG_Multiplexer M : 0|4@1+ (1,0) [0|0] "" DBG
 SG_ ECU_DBG_m0s0 m0 : 4|4@1+ (1,0) [0|0] "" DBG
 SG_ ECU_DBG_timestamp m0 : 8|8@1+ (1,0) [0|0] "" DBG        <---------- X
 SG_ ECU_DBG_m1s0 m1 : 4|2@1+ (1,0) [0|0] "" DBG
 SG_ ECU_DBG_timestamp m1 : 10|16@1+ (1,0) [0|0] "" DBG      <---------- X
```

meessage object:
```
message('DebugMessage', 0x457, False, 2, None)
  signal('ECU_DBG_Multiplexer', 0, 4, 'little_endian', False, None, 1, 0, None, None, 'None', True, None, None, None, None)
  signal('ECU_DBG_m0s0', 4, 4, 'little_endian', False, None, 1, 0, None, None, 'None', False, [0], None, None, None)
  signal('ECU_DBG_m1s0', 4, 2, 'little_endian', False, None, 1, 0, None, None, 'None', False, [1], None, None, None)
  signal('ECU_DBG_timestamp', 8, 8, 'little_endian', False, None, 1, 0, None, None, 'None', False, [0], None, None, None)
  signal('ECU_DBG_timestamp', 10, 16, 'little_endian', False, None, 1, 0, None, None, 'None', False, [1], None, None, None)
```

juleq, 2021-10-01:
You can have multiple mux ids for a signal if the _same_ signal at the same position is transmitted for multiple mux values.

Yes, warning on ambiguities (like CANdb++) would be a good idea. A PR would be welcome :).

yesralph, 2021-10-01:
> You can have multiple mux ids for a signal if the same signal at the same position is transmitted for multiple mux values.

That was one of my assumption but it does not do it :(  Still two separate messages even if they are of the same position and length.
```
message('DebugMessage', 0x457, False, 2, None)
  signal('ECU_DBG_Multiplexer', 0, 4, 'little_endian', False, None, 1, 0, None, None, 'None', True, None, None, None, None)
  signal('ECU_DBG_m0s0', 4, 4, 'little_endian', False, None, 1, 0, None, None, 'None', False, [0], None, None, None)
  signal('ECU_DBG_m1s0', 4, 2, 'little_endian', False, None, 1, 0, None, None, 'None', False, [1], None, None, None)
  signal('ECU_DBG_timestamp', 8, 8, 'little_endian', False, None, 1, 0, None, None, 'None', False, [0], None, None, None)   <---
  signal('ECU_DBG_timestamp', 8, 8, 'little_endian', False, None, 1, 0, None, None, 'None', False, [1], None, None, None)   <---
```

juleq, 2021-10-01:
You mean two separate signals?

yesralph, 2021-10-01:
> You mean two separate signals?

Yes, and the `multiplexer_ids` still have single M ID

juleq, 2021-10-01:
How do you set up this message and how do you inspect it? Dbc and dump? Have you inspected the message in the debugger?

yesralph, 2021-10-01:
I should probably have looked into the code how `multiplexer_ids` are filled. It's just from the use case, it does not seem to follow my expectation.

> How do you set up this message and how do you inspect it? Dbc and dump? Have you inspected the message in the debugger?

Here is my setup and debug output:
```
BO_ 1111 DebugMessage: 2 ECU
 SG_ ECU_DBG_Multiplexer M : 0|4@1+ (1,0) [0|0] "" DBG
 SG_ ECU_DBG_m0s0 m0 : 4|4@1+ (1,0) [0|0] "" DBG
 SG_ ECU_DBG_timestamp m0 : 8|8@1+ (1,0) [0|0] "" DBG      <--- same signal at 8|8
 SG_ ECU_DBG_m1s0 m1 : 4|2@1+ (1,0) [0|0] "" DBG
 SG_ ECU_DBG_timestamp m1 : 8|8@1+ (1,0) [0|0] "" DBG      <--- same signal at 8|8
````
debug output
````
>>> import cantools
>>> db=cantools.db.load_file("test.dbc")
>>> msg=db.get_message_by_name("DebugMessage")
>>> sig=msg.get_signal_by_name("ECU_DBG_timestamp")
>>> sig.multiplexer_ids
[0]
>>> print(f"{msg.signals[3].name}: {msg.signals[3].multiplexer_ids}")
ECU_DBG_timestamp: [0]
>>> print(f"{msg.signals[4].name}: {msg.signals[4].multiplexer_ids}")
ECU_DBG_timestamp: [1]
```

andlaus, 2021-11-22:
is this issue still relevant, or is it WONTFIX and shall be closed?

juleq, 2021-11-22:
Iwould vote for WONTFIX. CANdb++ warns when you map two signals with the same name into one message and the consistency check also complains. Doing that is asking for trouble.

[image]

[image]

andlaus, 2021-11-22:
ok. let's close this issue for now...

closed 2021-11-22 by andlaus

bessman, 2024-10-31:
Chiming in with a use case I have for same-named signals in a multiplexed message.

I'm working with CCP (CAN Calibration Protocol). A typical CCP session contains just two frame IDs: One going from the host to the ECU, and one in the other direction (these are called Command Receive Objects (CRO) and Data Transmission Objects (DTO) in the protocol specification).

Since there is only one frame ID in each direction, the frame is necessarily heavily multiplexed. Many commands, especially, share signal names across mux IDs. The signal tree of the CRO frame is as follows:

```
[{'command_code': {1: ['1_station_address'],
   2: ['2_mta', '2_extension', '2_address'],
   3: ['3_size', '3_data'],
   4: ['4_size'],
   6: ['6_mode',
    '6_daq_list_number',
    '6_last_odt_number',
    '6_event_channel',
    '6_rate_prescaler'],
   7: ['7_permanent', '7_station_address'],
   12: ['12_status_bits'],
   20: ['20_daq_list_number', '20_dto_id'],
   21: ['21_daq_list_number', '21_odt_number', '21_element_number'],
   22: ['22_size', '22_extension', '22_address'],
   23: ['23_device_info'],
   27: ['27_major', '27_minor'],
   33: ['33_action_service_number', '33_parameters']}},
 'ctr']
```
Here, I have prepended the mux ID to each signal name to work around the name collisions which would otherwise occur. This is of course a perfectly decent workaround. But if multiplexed signals could share names, I could directly pass through function keywords to the encoder :

```python
def connect(ecu):
    ecu._exchange(command_code=CONNECT, station_address=node.station_address)

def disconnect(ecu, permanent=1):
    ecu._exchange(command_code=DISCONNECT, permanent=permanent, station_address=node.station_address)

class ECU:
    ...
    def _exchange(self, **signals):
        cro = self.db.encode_message(self.cro_id, data=signals)
        self.send(cro)
        ...
```

Since this was closed "for now", I'm curious what the outlook is for enabling this use case?

juleq, 2024-10-31:
Hi,

Vector tools like for the GL logger seem to be capable of generating CCP/XCP dbc files. I would be curious how they get around this limitation in the dbc spec:

[image]

My guess is: They do not. They either do not run into the issue in the first place or they handle it by "uniquifying" the name collisions. But I do not have the time to find out. Maybe you can have a look?

So as it stands, this limitation is rooted in the dbc spec and outputting non-compliant files is asking for trouble is still my assessment.

If tools like the logger indeed do this "uniquifying", you could consider creating a PR that adds an option to resolve all such naming collisions in the same way to spare yourself the manual prefixing.

PS: The origin (tool) of your files would be good to include here and also if you know if the Vector tools choke on them (e.g. consistency check in CANdb++).

Best regards.

bessman, 2024-10-31:
> The origin (tool) of your files would be good to include here and also if you know if the Vector tools choke on them

I'm actually not working with any database files as origin. I'm creating it from scratch using cantools. So the signal names can be whatever I want them to be, I just thought it would be convenient to have them match the command parameter name from the CCP spec.

I've changed the message layout to instead prepend the command name:

```
[{'command_code': {1: ['CONNECT:station_address'],
   2: ['SET_MTA:mta', 'SET_MTA:extension', 'SET_MTA:address'],
   3: ['DNLOAD:size', 'DNLOAD:data'],
   4: ['UPLOAD:size'],
   33: ['ACTION_SERVICE:action_service_number', 'ACTION_SERVICE:parameters'],
   6: ['START_STOP:mode',
    'START_STOP:daq_list_number',
    'START_STOP:last_odt_number',
    'START_STOP:event_channel',
    'START_STOP:rate_prescaler'],
   7: ['DISCONNECT:permanent', 'DISCONNECT:station_address'],
   12: ['SET_S_STATUS:status_bits'],
   20: ['GET_DAQ_SIZE:daq_list_number', 'GET_DAQ_SIZE:dto_id'],
   21: ['SET_DAQ_PTR:daq_list_number',
    'SET_DAQ_PTR:odt_number',
    'SET_DAQ_PTR:element_number'],
   22: ['WRITE_DAQ:size', 'WRITE_DAQ:extension', 'WRITE_DAQ:address'],
   23: ['EXCHANGE_ID:device_info'],
   27: ['GET_CCP_VERSION:major', 'GET_CCP_VERSION:minor']}},
 'ctr']
```

This is more readable and still lets me programmatically create a dictionary for encoding rather easily.