cantools/cantools #356: standard and extended id signals overwritten

mitchdetailed, 2021-10-31:
when I load a dbc, if I have a standard arbitration id message that's the same value as an extended arbitration id, 1 is being overwritten.

is there a way to check if the message is standard or extended for decoding?

juleq, 2021-10-31:
Could you paste the message definition from the dbc (BO_)?

This usually happens when hand editing is done and one misses to set the highest bit of the id. The highest bit signifies that the id is of extended type.

So Id 0x100 as std would show as 256 in the dbc file while ext would show as 2147483904 (0x80000100).

mitchdetailed, 2021-10-31:
code <>

BO_ 0 Standard0: 8 Vector__XXX
 SG_ Std_0_sig : ***@***.***+ (1,0) [0|0] "" Vector__XXX

BO_ 2147483648 Extended0: 8 Vector__XXX
 SG_ Ext_0_sig : ***@***.***+ (1,0) [0|0] "" Vector__XXX
code</>

mitchdetailed, 2021-10-31:
```

BO_ 0 Standard0: 8 Vector__XXX
 SG_ Std_0_sig : ***@***.***+ (1,0) [0|0] "" Vector__XXX

BO_ 2147483648 Extended0: 8 Vector__XXX
 SG_ Ext_0_sig : ***@***.***+ (1,0) [0|0] "" Vector__XXX

```

juleq, 2021-10-31:
Having the same id for standard and extended is not supported at the moment. The messages collection uses the id (and not the id + dbc bit) as its key and the second message would just replace the first one.

I know that Vector does support that. Vector also supports name collisions. I would rather question that approach. Is there a reason that you cannot change your id scheme?

mitchdetailed, 2021-10-31:
it isn't an issue at present time, but is something I'm interested in
addressing if possible.

juleq, 2021-10-31:
I think supporting that would be traded against error potential during development. Developers would always have to watch out to use the correctly formatted id instead of just the id.

I have two other implementations that take the Vector approach. But I am the sole developer on those.

I think the original author purposefully chose to implement it that way and I am not sure that this use case would warrant a change.