cantools/cantools #357: Increase the usage of augmented assignment statements

elfring, 2021-11-02:
:eyes: Some source code analysis tools can help to find opportunities for improving software components.
:thought_balloon: I propose to [increase the usage of augmented assignment statements](https://docs.python.org/3/reference/simple_stmts.html#augmented-assignment-statements "Augmented assignment statements") accordingly.

```diff
diff --git a/cantools/database/can/c_source.py b/cantools/database/can/c_source.py
index b3ac68e..a688633 100644
--- a/cantools/database/can/c_source.py
+++ b/cantools/database/can/c_source.py
@@ -964,7 +964,7 @@ def _format_pack_code_level(message,
                                      variable_lines,
                                      helper_kinds)
 
-    body_lines = body_lines + muxes_lines
+    body_lines += muxes_lines
 
     if body_lines:
         body_lines = [''] + body_lines + ['']
@@ -1109,7 +1109,7 @@ def _format_unpack_code_level(message,
     if muxes_lines:
         muxes_lines.append('')
 
-    body_lines = body_lines + muxes_lines
+    body_lines += muxes_lines
 
     if body_lines:
         body_lines = [''] + body_lines
```

simontegelid, 2021-11-02:
Great idea! Can you provide a PR with the patch above?

elfring, 2021-11-05:
:thought_balloon: Can the presented changes be integrated also directly?

juleq, 2021-11-05:
Hi. Sure, just make a PR. When that passes testing, I can merge your PR.

elfring, 2021-11-05:
:thought_balloon: Would anybody else get into the mood to pick such an update suggestion up (by another pull/merge request)?

ilanbiala, 2025-11-20:
Closing as stale, this code cleanup is not critical and can happen as necessary along with other changes.