cantools/cantools #358: EncodeError for multiplexed Message and strict=True

zariiii9003, 2021-11-11:
I have a dictionary that contains all signals of a multiplexed message. It does not contain any other signals.
When calling the `Message.encode` method with `strict=True` i get an EncodeError.

This breaking change was introduced in #350.

After [this line](https://github.com/cantools/cantools/blob/f52c7d657dd9c128a04594f1e18109aa43c53bd0/cantools/database/can/message.py#L445) the variable `all_signals ` only contains the signals that belong to the current value of the multiplexor signal.

Then in [this line](https://github.com/cantools/cantools/blob/f52c7d657dd9c128a04594f1e18109aa43c53bd0/cantools/database/can/message.py#L449) the `data` dictionary (that contains all signals of the message) is compared to `all_signals` (which only contains the signals for the current value of the multiplexor).

I would need to use a separate dictionary for each value of the multiplexor to avoid this error.

juleq, 2021-11-11:
What file format do you use?
Can you post a sample?
What is your use case for cantools?

zariiii9003, 2021-11-11:
It is a DBC file but the OEM wouldn't be happy if i show the example.
The use case is a restbus simulation.

Let's say i have a multiplexed message with a multiplexor signal named 'N_MUX' .
So depending on the value of the multiplexor i have 3 messages:

N_MUX=0, N_1, N_2, N_3
N_MUX=1, N_4, N_5, N_6
N_MUX=2, N_7, N_8, N_9

Now when i call `encode` with a `data` dict that contains N_MUX and all of N_1 up to N_9 i get an EncodeError. But these signals N_1 to N_9 are all contained in the same message in the DBC file. So in my opinion this should not fail.
This worked fine before and now it does not.

juleq, 2021-11-11:
One multiplex value has one collection that has three enries. So three collections in total.

This is the usual way to implement it. 

Imagine if you had 3x 10ms frames for the muxes. The encoder would have to determine the subset of signals for a message with a given mux on each encode if you do not partition in advance. Thats a lot of unnecessary operations.

andlaus, 2021-11-11:
> I would need to use a separate dictionary for each value of the multiplexor to avoid this error.

Is there anything which prevents you from encoding your message in non-strict mode? (Because strict mode is -- err -- strict...)

juleq, 2021-11-11:
The decision to make that fail in strict was taken because not checking it might leave simple and common mistakes (typos) fail silently. Thats not what I would expect from a software that knows about the error, even more so in strict.

juleq, 2021-11-11:
Ahh, good evening :).

andlaus, 2021-11-11:
my pleasure. nice to see you ;)

zariiii9003, 2021-11-11:
But the failing check happens AFTER encoding. It does not have any performance benefit. [See here.](https://github.com/cantools/cantools/blob/f52c7d657dd9c128a04594f1e18109aa43c53bd0/cantools/database/can/message.py#L449)

It is fine when it fails for Signals that are not part of the Message in the DBC file. But here i'm only using the Signals from this Message instance.

It was convenient to have one `data` dictionary per message and get the correct encoded bytes just based on the value of the multiplexor value. That was an elegant soulution.

zariiii9003, 2021-11-11:
> > I would need to use a separate dictionary for each value of the multiplexor to avoid this error.
> 
> Is there anything which prevents you from encoding your message in non-strict mode? (Because strict mode is -- err -- strict...)

Well i like the range checking in some places of my code. But it is deactivated mostly.

juleq, 2021-11-11:
Ahh. Ok. Now I got you. I would agree that selecting by just setting the mux signal is the more convenient option. @andlaus, what is your take?

andlaus, 2021-11-11:
> But the failing check happens AFTER encoding. It does not have any performance benefit

if you want better performance, go for non-strict mode ;)

> Ahh. Ok. Now I got you. I would agree that selecting by just setting the mux signal is the more convenient option. @andlaus, what is your take?

I see why one would want to do something like this, but I think allowing the dictionary to contain signals that are only relevant for other multiplexer values will lead to difficult to find errors, e.g. if one modifies a signal in the dictionary and the encoded result won't change. I also think that using non-strict mode for such a use case would not be such a big deal. I'm not very dogmatic about this though, so if you @juleq prefer the other behavior, I'll prepare a patch for it soonish...

juleq, 2021-11-11:
I have a different implementation in C# that kind of behaves like @zariiii9003 proposes. 

My approach was to have all the signals, be it rx or tx, exposed to the consumer. When I want to send a specific mux, I set the signals (including the mux) and call encode on the message. Encode would check if the message is muxed and use the signal collection associated with the mux. So this subset collection is hidden and the consumer just plays with the signals.

I like that approach because it provides good abstraction. But I also get that one would appreciate a warning when a passed signal cannot be included because it is not associated with that mux.

Maybe we need strictness levels:
- I’m Too Young to Die
- Hurt Me Plenty
- Ultra Violence
- and Nightmare
have been popular choices once upon a time :).

Or maybe there is a compromise: Issue a warning instead of erroring out?

zariiii9003, 2021-11-11:
I think it would be better to split the `strict`-Parameter into range checks / missing signals / unknown signals. But that might break a lot of code...

andlaus, 2021-11-12:
> My approach was to have all the signals, be it rx or tx, exposed to the consumer.

I thought a bit about this. I concluded that @zariiii9003 is basically a less generic form of putting all signals of the whole system into the dictionary, and encoding frames based on this. I still think that if in doubt the `strict` parameter of the  `encode()` method should make it err on the side of being too strict, but I think it would be possible to introduce a `gather_message_signals(global_dict)` method to `Message` which given a larger dictionary produces one that only contains the ones relevant for `encode()`. Would this fix your problem @zariiii9003?

zariiii9003, 2021-11-13:
1. I would keep range checking and the signal completeness checking separated. I think the user might want one but not the other.
2. For the signal completeness i would add a `_signal_set: Set[str]` attribute to the `Message` class. That attribute would be calculated during the Database parsing and it should contains all the signal names of the Message. During the `encode()` call i would check the signal completeness with `self._signal_set == set(data.keys())`. This should be more efficient than the current implementation.

andlaus, 2021-11-14:
> I would keep range checking and the signal completeness checking separated. I think the user might want one but not the other.

I agree that it would be nice to split the strictness checks into `message.ensure_signal_ranges(signal_dict)`, `message.ensure_signal_completeness(signal_dict)` and `message.ensure_no_unused_signals(signal_dict)` and just call all of them if `strict=True` is called for `encode()`. Maybe I've got a bit of spare time to hack this together next week...

> [...] This should be more efficient than the current implementation.

good point, but I don't think it will be quite as easy once multiplexers get involved, in particular nested ones will be potentially ugly. is the current encoding performance actually a limiting factor for you?

AlexDLSy, 2021-11-15:
Hi there @andlaus @juleq,
Actually the follow-up of this conversation might lead to what I was suggesting in #342 .
Or is it just that the feature from #350 actually resolves issue #342 ? And @zariiii9003 's data sources is not compliant with that formalism ?

juleq, 2021-11-15:
@AlexDLSy Its the latter. @zariiii9003 Wants to be able to pass a collection of signal values that is not limited to the mux value that is set. Doing so currently raises an encode error in strict, which it did not do before.

AlexDLSy, 2021-11-15:
OK, so shall we close #342 then ?

closed 2022-02-24 by juleq