cantools/cantools #369: Not able to decode properly floats sent over CAN between Arduino Nano IoT and RPI

AndPatr, 2021-11-30:
Hi,
My setup is made of the following components: 
- A RPI with Ubuntu 20.04 Server installed on it and the Seedstudio CAN hat mounted on it (successfully tested and validated)
- An Arduino Nano IoT connected to the CAN SPI Click board from Mickroelectronica (successfully tested and validated) and the Adafruit BNO085 IMU fusion sensor

I have generated a suitable.dbc file using the tools provided by cantools.
Let us look at an example for simplicity, i.e. sending a quaternion through CAN. Each component of the quaternion is 32 bit, so I have to split the data into two messages, one containing the real part and the first component and then a second message containing the rest.
On the RPI side, I use the Python utilities of cantools to read messages (successfully).
On the Arduino side, I use the c-code generator tool of cantools for generating the packing functions.
I want to stress that I have successfully tested the functioning of the Arduino CAN library which I use for sending data and works perfectly fine.

The resulting packaged message is sent over CAN successfully, but the data sent does not correspond to the original floats.
There is probably some problem with how floats are handled on the Nano Iot or in the pack functions.
As an example, these are the frames I receive on the RPI side:

(2934.944351)  can1  RX - -  001   [8]  00 00 C0 9C 3D 00 00 A8    --------------> q_w=-1.2705494208814505e-21

(2934.944368)  can1  RX - -  002   [8]  00 C0 9C 3D 00 00 A8 3A   -------------->  q_i=-7.105479026610784e-15

where the message with ID 1 is meant to be the first half of the quaternion, while the ID 2 is associated with the rest.
As you can see, there is a strange pattern bewteen these two frame (C0 9C 3D, for example, appears again is the second message, left shifted by 1 byte).

The decoding works fine and it is coherent with the sent data package; it is the packaged data frame which is wrong, and I do not know why.
Any help is much appreciated.

juleq, 2021-11-30:
Are you sure that you are packing the actual float/double value and not its address (pointer)?

Keep in mind that IEEE754 does not specify byte or word order. That is another catch when working with floats.

I would try to pack values that you know the byte representation of. 0.0, 1.0 and the like. Might make understanding what happens more easy. 

You could also post code snips.

AndPatr, 2021-12-01:
Thanks @juleq for your reply. 
I'll try to send known values to see what happens.
To pack the data frame, I am using the auto-generated pack functions.
For example, this is the one for packing the first half of the quaternion:

```
   int quat_first_half_pack(
      uint8_t *dst_p,
      const struct quat_first_half_t *src_p,
      size_t size)
  {
    uint32_t q_i;
    uint32_t q_w;

    if (size < 8u) {
        return (-EINVAL);
    }

    memset(&dst_p, 0, 8);
        
    memcpy(&q_w, &src_p->q_w, sizeof(q_w));
    dst_p[0] |= pack_left_shift_u32(q_w, 0u, 0xffu);
    dst_p[1] |= pack_right_shift_u32(q_w, 8u, 0xffu);
    dst_p[2] |= pack_right_shift_u32(q_w, 16u, 0xffu);
    dst_p[3] |= pack_right_shift_u32(q_w, 24u, 0xffu);
    memcpy(&q_i, &src_p->q_i, sizeof(q_i));
    dst_p[4] |= pack_left_shift_u32(q_i, 0u, 0xffu);
    dst_p[5] |= pack_right_shift_u32(q_i, 8u, 0xffu);
    dst_p[6] |= pack_right_shift_u32(q_i, 16u, 0xffu);
    dst_p[7] |= pack_right_shift_u32(q_i, 24u, 0xffu);

    return (8);
}
```
and this is the function I use for sending the orientation over CAN:
```
void sendOrientationOverCAN(uint8_t *dst_p_first_half, uint8_t *dst_p_second_half,
                            struct whee_b_bot_imu_can_quat_first_half_t *src_p_first_half,  struct whee_b_bot_imu_can_quat_second_half_t *src_p_second_half,
                            struct orient_quat_t *quaternion) {
  
     src_p_first_half->q_w=quaternion->q_w;
     src_p_first_half->q_i=quaternion->q_i;
     src_p_second_half->q_j=quaternion->q_j;
     src_p_second_half->q_k=quaternion->q_k;
     
     whee_b_bot_imu_can_quat_first_half_pack(dst_p_first_half,src_p_first_half,64u);
     whee_b_bot_imu_can_quat_second_half_pack(dst_p_second_half,src_p_second_half,64u);

     CAN.sendMsgBuf(0x001, CAN_STDID, 8, dst_p_first_half); 
     CAN.sendMsgBuf(0x002, CAN_STDID, 8, dst_p_second_half);

}
```
I checked that src_p_first_half actually contains the right float values, and indeed it does.
I really do not know if here I am doing something wrong with pointers/addresses.

juleq, 2021-12-01:
The packing function looks good to me. Direct queries of member addresses avoid alignment issues you might have when working with offsets in structs.

Why does the send function take the dest buffers as parameters? Since they are probably not used outside, I would just make them static locals of the send function to minimize their scope. Or is that also generated?

As a next step I would overwrite the quaternion components with known values at the beginning of the send function and see whats happening on the bus. Next, I would print the dest bytes after sending each of the frames to verify that it equals the bytes that are being sent on the bus.

AndPatr, 2021-12-01:
Yes, you are right; I can make them static locals for sure (and I will).
As soon as I can, I will try to overwrite those values assigning for example, 1.0 to q_w and 0.0 to q_i and I will see if they are packaged properly.
I've already perfomed the second check that you mentioned and the bytes received on the bus are exactly the same ones which are packaged in the dest object, so I would exclude problems on the CAN library. Just to be sure, I will perform that check again.

I will let you know what happens very soon.
Thanks for the help.

Ps. Could it also be a problem of how floats are handled on the Nano IoT (it has a SAMD21 Cortex®-M0+ 32bit low power ARM MCU processor)?

juleq, 2021-12-01:
It is a modern core, little endian, without FPU and it will use a modern math lib to do the floats. I would not expect surprises there.

AndPatr, 2021-12-03:
So, there is an update. 
There was an error in my code: changing '''memset(&dst_p, 0, 8); ''' to ''' memset(dst_p, 0, 8);''' or ''' memset(dst_p[0], 0, 8);''' I am now able to send known floats in a reliable way.
There are some errors appearing when trying to decode numbers like, for instance, 0.0, which might appear something like 1e-38 on the RPI side (there is a conversion from float to double happening).
The problem is that, when sending the actual readings from the IMU (either quaternions, euler angles, gyro, etc...) I always get only part of the message right; for example, if sending a quaternion, I might get only a couple of components right, while the others are off. Sometimes some of the values become NaN.
This is quite strange, since I plot those values over serial right before packing and they are ok.
Any idea of what might be happening? 
To prevent issues related to the timing I am doing all these tests plotting and sending values at 1 Hz.

juleq, 2021-12-03:
What types are used in source, bus and destination?

What does the change look like?

E.g.
srcVal1 -> dstVal1
srcVal2 -> dstVal2
etc.

So what does one corrupted set look like at src and at dst?

AndPatr, 2021-12-03:
In the source structures the used data type is float (the Nano IoT is a 32 bit board). The destination variables use uint8_t, which is then "expanded" to 32 bits with `memset()` .
The function which sends on the bus uses the following inputs `sendMsgBuf(byte status, unsigned long id, byte ext, byte rtrBit, byte len, volatile const byte *buf) `.
For example, to send the first half of the quaternion:  `CAN.sendMsgBuf(msgIDs->quat_first_half_ID, 0, 8, dst_p_quat_first_half);` where `uint8_t *dst_p_quat_first_half;`.
For reference, I have uploaded the code at [https://github.com/AndPatr/WheeBBot-v1.0.0/blob/devel/Arduino/test_scripts/IMU_data_over_CAN_test/IMU_data_over_CAN_test.ino](url). This repo also contains the library I am using for sending messages over the bus.

AndPatr, 2021-12-03:
In the weekend I will perform more detailed tests to better check the transition between the source variables and the destination ones

AndPatr, 2021-12-03:
As a side note, I should point out that I have modified the Seed_Arduino_CAN library to work with the SPI-CAN shield I am using on the Nano IoT, since that chip uses a 10 MHz oscillator, which is relatively uncommon.
Basically, I added suitable bit timing register configs, which I computed using a tool from Kvaser at [https://www.kvaser.com/support/calculators/bit-timing-calculator/](url). This tools does not have the MCP2515 option, so I computed the bit timings using the MCP2510. However, I don't think the problem lies there, since I am able to send known values correctly.
Here is the added configuration, taken from the mcp2515_can_dfs.h file:

#define MCP_10MHz_1000kBPS_CFG1 (0x00)
#define MCP_10MHz_1000kBPS_CFG2 (0x80)
#define MCP_10MHz_1000kBPS_CFG3 (0x01)

#define MCP_10MHz_500kBPS_CFG1 (0x00)
#define MCP_10MHz_500kBPS_CFG2 (0x92)
#define MCP_10MHz_500kBPS_CFG3 (0x02)

#define MCP_10MHz_250kBPS_CFG1 (0x00)
#define MCP_10MHz_250kBPS_CFG2 (0xB6)
#define MCP_10MHz_250kBPS_CFG3 (0x04)

#define MCP_10MHz_200kBPS_CFG1 (0x00)
#define MCP_10MHz_200kBPS_CFG2 (0xBF)
#define MCP_10MHz_200kBPS_CFG3 (0x07)

#define MCP_10MHz_125kBPS_CFG1 (0x01)
#define MCP_10MHz_125kBPS_CFG2 (0xB6)
#define MCP_10MHz_125kBPS_CFG3 (0x04)

#define MCP_10MHz_100kBPS_CFG1 (0x04)
#define MCP_10MHz_100kBPS_CFG2 (0x92)
#define MCP_10MHz_100kBPS_CFG3 (0x02)

#define MCP_10MHz_50kBPS_CFG1 (0x04)
#define MCP_10MHz_50kBPS_CFG2 (0xB6)
#define MCP_10MHz_50kBPS_CFG3 (0x04)

#define MCP_10MHz_40kBPS_CFG1 (0x04)
#define MCP_10MHz_40kBPS_CFG2 (0xBF)
#define MCP_10MHz_40kBPS_CFG3 (0x07)

#define MCP_10MHz_20kBPS_CFG1 (0x18)
#define MCP_10MHz_20kBPS_CFG2 (0x92)
#define MCP_10MHz_20kBPS_CFG3 (0x02)

#define MCP_10MHz_10kBPS_CFG1 (0x18)
#define MCP_10MHz_10kBPS_CFG2 (0xB6)
#define MCP_10MHz_10kBPS_CFG3 (0x04)

#define MCP_10MHz_5kBPS_CFG1 (0x31)
#define MCP_10MHz_5kBPS_CFG2 (0xB6)
#define MCP_10MHz_5kBPS_CFG3 (0x04)

juleq, 2021-12-03:
> Basically, I added suitable bit timing register configs, which I computed using a tool from Kvaser at https://www.kvaser.com/support/calculators/bit-timing-calculator/. This tools does not have the MCP2515 option, so I computed the bit timings using the MCP2510. However, I don't think the problem lies there, since I am able to send known values correctly.

Yeah, I am familiar with the China MPCs ;). Should be no issue. You would expect error frames if that was a problem.

juleq, 2021-12-03:
"The destination variables use uint8_t, which is then "expanded" to 32 bits with memset()"

Could you show the code? Keep in mind that memcpy and memset work on a byte basis. So you could shoot yourself in the foot in two ways: Alignment in memory (compiler is free to align struct members on machine boundaries efficiently without telling you) and setting e.g. each byte in e.g. an int32_t to a certain value where you just meant to set one value to the overall int32_t.

AndPatr, 2021-12-03:
Yes, you will find the complete and updated version of my arduino code at the repository link [ https://github.com/AndPatr/WheeBBot-v1.0.0/blob/devel/Arduino/test_scripts/IMU_data_over_CAN_test/IMU_data_over_CAN_test.ino](url).
Hopefully it is organized and commented well enough to be easily readable. If not, I can send here parts of it which might be significant or even make a condensed version of it with a single message, instead of the whole database.

AndPatr, 2021-12-04:
I have now removed most of the code and simplified it to better debug the issue.
Now I am only sending the first half of the quaternion (the code I am now using at https://github.com/AndPatr/WheeBBot-v1/tree/main/Arduino/test_scripts/IMU_data_over_CAN_test_simplified).
Source q_w and HEX:

Source q_w:	`0.25677490`	source q_i:	`0.00390625`
Packed HEX on the Nano IoT:	`00-78-83-3E | 00-00-80-3B`

Read msg on the RPI: ` (6430.142733)  can1  RX - -  001   [8]  00 78 83 3E 00 00 80 3B`

`q_w:0.25677490234375
q_i:0.00390625`

As you see, I am getting the values right, even if I move rapidly my IMU.

As soon as I try to add the second half, the values are completely wrong (there is also a NaN appearing sometimes). You can see the script I am using for this second case at https://github.com/AndPatr/WheeBBot-v1/tree/main/Arduino/test_scripts/IMU_data_over_CAN_test.

juleq, 2021-12-04:
I just looked through the driver that you are using. You are using a wrapper that not only waits for copying the data into the MCP message box, but also for the box to be transmitted. So thats a defensive implementation, would not expect problem there.

Please post both frames, send and receive. So:

Sent bytes, frame 1 -> Received bytes, frame 1
Sent bytes, frame 2 -> Received bytes, frame 2 

Do you use an MCP with the RPI, too? I assume the mainline socketcan driver for the MCP?

juleq, 2021-12-04:
Also test the following:
- Does sending only the 2nd half work?
- Do both sends fail when both are enabled?
- If only the last one fails: What is the behavior when you reverse order?
- Does adding a delay in
between change outcome?

AndPatr, 2021-12-04:
I will perform the tests you suggest right now.
In the meantime, the shield I am using on the RPI is https://wiki.seeedstudio.com/2-Channel-CAN-BUS-FD-Shield-for-Raspberry-Pi/, which uses the MCP2517FD or MCP2518FD (right now I do not remember which one).

AndPatr, 2021-12-04:
Results of sending both frames:
FIRST FRAME->
Source: `q_w: 0.25445557;  q_i:-0.06903076 `
Sent frame: `0-48-82-BF | 3C-60-ED-FF`
Received frame (on RPI): `001   [8]  00-48-82-BF | 3C-60-ED-FF`
Decoded value(on RPI): `q_w: -1.017822265625; q_i: nan`
SECOND FRAME->
Source:  `q_j: 0.01995850; q_k:	0.96441650`
Sent frame: `48-82-BF-3C | 60-ED-FF-3F`
Received frame (on RPI): `002   [8]  48-82-BF-3C | 60-ED-FF-3F`
Decoded value(on RPI): `q_j: 0.02337755262851715; q_k: 1.9994316101074219`

AndPatr, 2021-12-04:
Results of the tests you suggested:

- "Does sending only the 2nd half work?"-> Yes, it works
- "Do both sends fail when both are enabled?"->As you see in the previous message, both sends fail (meaning the values are wrong)
- "If only the last one fails: What is the behavior when you reverse order?"-> both fail

AndPatr, 2021-12-04:
For the last one, i.e. the delay, there is something interesting which happens. 

- "Does adding a delay in
between change outcome?"-> Apparently it does not (I tried different amount of delay in different positions).

However, if I change my packing function from:
```
 int orient_quat_pack(
    uint8_t* dst_p_first,
    uint8_t* dst_p_second,
    const struct whee_b_bot_imu_can_quat_first_half_t* src_p_first,
    const struct whee_b_bot_imu_can_quat_second_half_t* src_p_second)
{
    uint32_t q_i;
    uint32_t q_w;
    uint32_t q_j;
    uint32_t q_k;

    memset(dst_p_first, 0, 8);
    memset(dst_p_second, 0, 8);

    Serial.print("\n source q_w:\t"); Serial.print(src_p_first->q_w,8); Serial.print("\t"); Serial.print("source q_i:\t"); Serial.print(src_p_first->q_i,8); Serial.print("\n"); // checking if source package is read correctly
    Serial.print("\n source q_j:\t"); Serial.print(src_p_second->q_j,8); Serial.print("\t"); Serial.print("source q_k:\t"); Serial.print(src_p_second->q_k,8); Serial.print("\n"); // checking if source package is read correctly

    memcpy(&q_w, &(src_p_first->q_w), sizeof(q_w));
    dst_p_first[0] |= pack_left_shift_u32(q_w, 0u, 0xffu);
    dst_p_first[1] |= pack_right_shift_u32(q_w, 8u, 0xffu);
    dst_p_first[2] |= pack_right_shift_u32(q_w, 16u, 0xffu);
    dst_p_first[3] |= pack_right_shift_u32(q_w, 24u, 0xffu);
    
    memcpy(&q_i, &(src_p_first->q_i), sizeof(q_i));
    dst_p_first[4] |= pack_left_shift_u32(q_i, 0u, 0xffu);
    dst_p_first[5] |= pack_right_shift_u32(q_i, 8u, 0xffu);
    dst_p_first[6] |= pack_right_shift_u32(q_i, 16u, 0xffu);
    dst_p_first[7] |= pack_right_shift_u32(q_i, 24u, 0xffu);
       
    
    
    memcpy(&q_j, &(src_p_second->q_j), sizeof(q_j));
    dst_p_second[0] |= pack_left_shift_u32(q_j, 0u, 0xffu);
    dst_p_second[1] |= pack_right_shift_u32(q_j, 8u, 0xffu);
    dst_p_second[2] |= pack_right_shift_u32(q_j, 16u, 0xffu);
    dst_p_second[3] |= pack_right_shift_u32(q_j, 24u, 0xffu);
    
    memcpy(&q_k, &(src_p_second->q_k), sizeof(q_k));
    dst_p_second[4] |= pack_left_shift_u32(q_k, 0u, 0xffu);
    dst_p_second[5] |= pack_right_shift_u32(q_k, 8u, 0xffu);
    dst_p_second[6] |= pack_right_shift_u32(q_k, 16u, 0xffu);
    dst_p_second[7] |= pack_right_shift_u32(q_k, 24u, 0xffu);
    
    return (8);
}
```

To :

```
int orient_quat_pack(
    uint8_t* dst_p_first,
    uint8_t* dst_p_second,
    const struct whee_b_bot_imu_can_quat_first_half_t* src_p_first,
    const struct whee_b_bot_imu_can_quat_second_half_t* src_p_second)
{
    uint32_t q_i;
    uint32_t q_w;
    uint32_t q_j;
    uint32_t q_k;

    memset(dst_p_first, 0, 8);

    Serial.print("\n source q_w:\t"); Serial.print(src_p_first->q_w,8); Serial.print("\t"); Serial.print("source q_i:\t"); Serial.print(src_p_first->q_i,8); Serial.print("\n"); // checking if source package is read correctly
    Serial.print("\n source q_j:\t"); Serial.print(src_p_second->q_j,8); Serial.print("\t"); Serial.print("source q_k:\t"); Serial.print(src_p_second->q_k,8); Serial.print("\n"); // checking if source package is read correctly

    memcpy(&q_w, &(src_p_first->q_w), sizeof(q_w));
    dst_p_first[0] |= pack_left_shift_u32(q_w, 0u, 0xffu);
    dst_p_first[1] |= pack_right_shift_u32(q_w, 8u, 0xffu);
    dst_p_first[2] |= pack_right_shift_u32(q_w, 16u, 0xffu);
    dst_p_first[3] |= pack_right_shift_u32(q_w, 24u, 0xffu);
    
    memcpy(&q_i, &(src_p_first->q_i), sizeof(q_i));
    dst_p_first[4] |= pack_left_shift_u32(q_i, 0u, 0xffu);
    dst_p_first[5] |= pack_right_shift_u32(q_i, 8u, 0xffu);
    dst_p_first[6] |= pack_right_shift_u32(q_i, 16u, 0xffu);
    dst_p_first[7] |= pack_right_shift_u32(q_i, 24u, 0xffu);
       
    memset(dst_p_second, 0, 8);
    
    memcpy(&q_j, &(src_p_second->q_j), sizeof(q_j));
    dst_p_second[0] |= pack_left_shift_u32(q_j, 0u, 0xffu);
    dst_p_second[1] |= pack_right_shift_u32(q_j, 8u, 0xffu);
    dst_p_second[2] |= pack_right_shift_u32(q_j, 16u, 0xffu);
    dst_p_second[3] |= pack_right_shift_u32(q_j, 24u, 0xffu);
    
    memcpy(&q_k, &(src_p_second->q_k), sizeof(q_k));
    dst_p_second[4] |= pack_left_shift_u32(q_k, 0u, 0xffu);
    dst_p_second[5] |= pack_right_shift_u32(q_k, 8u, 0xffu);
    dst_p_second[6] |= pack_right_shift_u32(q_k, 16u, 0xffu);
    dst_p_second[7] |= pack_right_shift_u32(q_k, 24u, 0xffu);
    
    return (8);
}
```
Then what happens is that I get the whole second frame right. The only change was to put the second` memset(dst_p_second, 0, 8);` after the `memcpy()` of the first frame.
Simply adding a delay between them is not sufficient to make the second frame right.
A really curious behaviour I'd say.

AndPatr, 2021-12-04:
I have also tried to split the packaging into a function for the first half and a function for packing the second, but I still get only the second frame right.
I have also tried to remove all instances of serial prints (just to try), but I still get a completely off first frame.

juleq, 2021-12-04:
Try the following on both hosts:

uint32_t test = 0x004882BF;
float ftest = *(float*)&test;
print ftest;

This is just setting the bytes for 0.25445557 directly in an unsigned type and reinterpret them as float.

Given that:
- Your endianness
- Your math.h float representation 
- And your printing works…

…you should get the same results on both machines. Please verify that.

PS: You can also cast the frame bytes:
byte buffer[] = {0x00, 0x48, 0x82, 0xBF};
float ftest = *(float*)buffer;

To be clear: Such reinterpreting casts are not recommended because of their portability issues and the surprises that arch/compiler might have in store for you, but its a good test to see whats happening.

AndPatr, 2021-12-04:
I just tried ```
uint32_t test = 0x004882BF;
float ftest = *((float*)&test);
``` on the Nano IoT.
When I print to serial, I only get 0.00000000.
However, the second method using the buffer works and gives me an output of 1.01782227

AndPatr, 2021-12-04:
If I try the first on the RPI, I get `6.65906e-39`.
As of now, I was not able to execute the second command because it gives me compilation issues ( error: cannot convert ‘int’ to ‘std::byte’).

juleq, 2021-12-04:
Sorry, mixed up BE and LE in the test variable:

    uint32_t valueLE = 0x3e824800;
    float ftest = *((float*)&valueLE);
    
    printf("%f\n", ftest);
    printf("%08x\n", *((uint32_t*)&ftest));

    uint8_t bytesLE[] = {0x00, 0x48, 0x82, 0x3E};
    ftest = *((float*)bytesLE);
    
    printf("%f\n", ftest);
    printf("%08x\n", *((uint32_t*)&ftest));
    
    uint8_t bytesBE[] = {0x3E, 0x82, 0x48, 0x00};
    ftest = *((float*)bytesBE);
    
    printf("%f\n", ftest);
    printf("%08x\n", *((uint32_t*)&ftest));

juleq, 2021-12-04:
The memset is curious indeed. It should have no effect at all since it is unconditionally overwritten by memcpy. Maybe comment them out for now.

juleq, 2021-12-04:
Ok, on a big screen now. Found it. Your comment is lying :)

    uint8_t dst_p_quat_first_half;// 8 byte data destination variable for CAN msgs
    uint8_t dst_p_quat_second_half;// 8 byte data destination variable for CAN msgs

You are only reserving one byte of data and using it as if you have reserved 8. Mem corruption.

Should be:

uint8_t dst_p_quat_first_half[8]; // 8 byte data destination variable for CAN msgs
uint8_t dst_p_quat_second_half[8]; // 8 byte data destination variable for CAN msgs

Be sure to just use dst_p_quat_first_half when passing it around (and not &dst_p_quat_first_half) since it already is a pointer.

AndPatr, 2021-12-04:
Wow, that was it.
Now it all works flawlessly. 
Your really saved me. Thanks for your help and support.
If my project will ever work, that will also be thanks to you and the awesome cantools utility.

juleq, 2021-12-04:
Cool. Pitch your project here once its up and running. Happy hacking.

juleq, 2021-12-04:
I always suffer when I use the Arduino IDE.

One recommendation: You really should try Platform.io with VS Code some time. It makes it easy to set up your project into modules to separate your systems concerns. And VS Code, code completion and error checking are awesome.

It supports a lot of platforms.

closed 2021-12-04 by juleq