cantools/cantools #370: Wrong compare in _check_signals_ranges_scaling

fabiano67, 2021-12-01:
Hi,
I have the error

`File "... cantools\database\can\message.py", line 368, in _check_signals_ranges_scaling raise EncodeError(
cantools.database.errors.EncodeError: Expected signal 'xxx' value less than or equal to 65.535 in message 'yyy', but got 65.535.`

while importing a dbc file.
The signal is defined in the following way

[image]

The problem is in the float numbers compare, as reported in the following picture

[image]

It seems that the error is related to the internal number representation.
In the picture
value: Decimal('65.535')
signal.maximum: 65.535
The two number are float type but the internal representation is different and the compare fail.

[image]

juleq, 2021-12-01:
Have you tried setting your initial value to the next lower value representable in float? Setting the initial value that way and enabling the checks almost assumes that the counterpart uses a fixed-point representation.

fabiano67, 2021-12-01:
Hi,
I can try but the dbc is from a big Swedish truck manufacturer and formally
I can't touch it.

By the way I will have a try as soon as possible.

Thanks

juleq, 2021-12-01:
You might have another problem. That OEM sets the value like this so the (fixed point) consumer can test for availability of that value. If it is 0xffff, it is not available. You cannot test like that when your internal representation is float.

For cantools there would probably be two options:

 - Harmonize the calculations and conversions such that maximum and the value are equal
 - Consider an epsilon when testing the bounds - but has to be in a way that ensures encoding does not fail

Both won't solve that availability test issue.

Alternatively you could check the behavior when the bounds check is disabled.

fabiano67, 2021-12-01:
If I disable the 'bounds check' the problem disappears.
I agree with you that in cantools the calculations and conversions should be harmonized.

closed 2025-11-20 by ilanbiala