cantools/cantools #373: Wrong compare type in _check_signals_ranges_scaling()

triqtruong, 2021-12-16:
Hi, I got an error while running cantool version 37.0.0 
`Error: '<' not supported between instances of 'NamedSignalValue' and 'int'`

This seems to be bug where the signals with type `NamedSignalValue` are not skipped before being compare with an actual number (in this case, an `int`). Bug seems to be introduced in version `36.4.0`
[image]

andlaus, 2021-12-16:
hm, this probably means that your DBC specifies a minimum value for the signal plus some named values and the value which you try to encode is one of the named ones. I agree that this is a legit use case and that this is a bug. I'll have a look...

juleq, 2021-12-16:
@andlaus could you also add a test to ensure we don't regress? 

I have my merge happy gloves on this week, wink and we should be able to release a fix ^^.

@triqtruong Thank you for reporting.

andlaus, 2021-12-17:
@triqtruong: can you please verify that your issue is gone in the latest master version?

triqtruong, 2021-12-17:
@andlaus The issue is gone now. Thank you! 
Would it be possible if you could put this as a hotfix version (something like 37.0.1), so it'll be a bit easier to update via pip? Thanks a lot!

juleq, 2021-12-17:
Will release when I get home today :).

andlaus, 2021-12-17:
great, let's close this issue then...

closed 2021-12-17 by andlaus