cantools/cantools #375: arxml.py file issue. I can't load arxml database.

linpanusst, 2021-12-20:
I have some problem with load the arxml database file in the python object. The error message is :
`  File "C:\ProgramData\Anaconda3\lib\site-packages\cantools\database\can\formats\arxml.py", line 87, in parse_int_string
    return int(in_string, 0) # autodetect the base
ValueError: invalid literal for int() with base 0: '0.0'`

I have checked the input. The input is in_string variable is "0.0". It seams that int(in_string, 0) can't decode "0.0". However "0" works fine.
I checked that "21.0" also failed. "21" works fine.

andlaus, 2021-12-20:
it seems like the code expects an integer but a floating point value is specified by your file. I don't have any file which does this (e.g. fractional scaling works by providing integers for the numerator and the denominator), but if  you provide a test case which validates against the ARXML XSD, I'll fix this...

linpanusst, 2021-12-21:
Ok, I will prepare some simple test case via your e-mail which is open on your github home page before 2021.12.23. Thank you for  your time.

linpanusst, 2022-01-05:
> it seems like the code expects an integer but a floating point value is specified by your file. I don't have any file which does this (e.g. fractional scaling works by providing integers for the [...]

Hi, I have e-mail you some test data. Sorry for the delay due to some health problem. Thank you for your time.

closed 2022-01-12 by simontegelid