cantools/cantools #376: default cycle time not applied to message

solarjoe, 2021-12-22:
Hello,

I am not sure if this is a bug or the correct behavior, but the default cycle time is not assigned to a message.

For the DBC below, `Message_2` has no cycle time assigned and I would have expected it to use the default value 0 stated as  `BA_DEF_DEF_  "GenMsgCycleTime" 0;`. This is also the behaviour I see with CANdb++. But cantools states it is `None`.

```
import cantools
from pathlib import Path

SCRIPT_DIR = Path(__file__).parent
INPUT_DBC_PATH = SCRIPT_DIR.joinpath('cycletime_default.dbc')

db = cantools.db.load_file(INPUT_DBC_PATH)

ct1 = db.get_message_by_name('Message_1').cycle_time

ct2 = db.get_message_by_name('Message_2').cycle_time

print(ct1)
print(ct2)
```

```
VERSION ""

NS_ : 
	NS_DESC_
	CM_
	BA_DEF_
	BA_
	VAL_
	CAT_DEF_
	CAT_
	FILTER
	BA_DEF_DEF_
	EV_DATA_
	ENVVAR_DATA_
	SGTYPE_
	SGTYPE_VAL_
	BA_DEF_SGTYPE_
	BA_SGTYPE_
	SIG_TYPE_REF_
	VAL_TABLE_
	SIG_GROUP_
	SIG_VALTYPE_
	SIGTYPE_VALTYPE_
	BO_TX_BU_
	BA_DEF_REL_
	BA_REL_
	BA_DEF_DEF_REL_
	BU_SG_REL_
	BU_EV_REL_
	BU_BO_REL_
	SG_MUL_VAL_

BS_:

BU_: XYZ Control

BO_ 1 Message_1: 8 XYZ
 SG_ Signal_1 : 8|4@1+ (1,0) [0|15] ""  Control

BO_ 2 Message_2: 8 XYZ
 SG_ Signal_2 : 48|14@1+ (16,0) [0|262128] ""  Control

BA_DEF_ BO_  "GenMsgCycleTime" INT 0 65535;
BA_DEF_DEF_  "GenMsgCycleTime" 0;

BA_ "GenMsgCycleTime" BO_ 1 500;
```

andlaus, 2021-12-23:
it seems like the problem is that the file specifies a default of 0 which is interpreted as there being no cycle time at all. Looking at the [code](https://github.com/cantools/cantools/blob/master/cantools/database/can/formats/dbc.py#L1376), it looks to me like the default for this attribute is considered correctly?

andlaus, 2021-12-23:
(the issue boils down to what a cycle time of 0 means: e.g., the bus could only feature messages of this type in order to make the cycle time as small as possible, it could mean that this is an aperiodic message or it could mean something else entirely. Currently the second interpretation is used by cantools and to avoid having to add a case distinction in the user code, the `cycle_time` property of such messages is set to `None`.)

solarjoe, 2021-12-23:
Ok, that's fine then. Thanks for the explanation.

closed 2021-12-23 by solarjoe