cantools/cantools #384: ModuleNotFoundError: No module named 'typing_extensions'

Andreasdahlberg, 2022-01-23:
**Steps to reproduce:** 
Run `cantools generate_c_source --no-floating-point-numbers --database-name NAME --output-directory OUT_DIR DBC_FILE`

**Output:**
```
Traceback (most recent call last):
  File "/usr/local/bin/cantools", line 5, in <module>
    from cantools.__init__ import _main
  File "/usr/local/lib/python3.8/dist-packages/cantools/__init__.py", line 13, in <module>
    from . import database as db
  File "/usr/local/lib/python3.8/dist-packages/cantools/database/__init__.py", line 8, in <module>
    from . import can
  File "/usr/local/lib/python3.8/dist-packages/cantools/database/can/__init__.py", line 1, in <module>
    from .database import Database
  File "/usr/local/lib/python3.8/dist-packages/cantools/database/can/database.py", line 11, in <module>
    from .formats import arxml
  File "/usr/local/lib/python3.8/dist-packages/cantools/database/can/formats/arxml.py", line 11, in <module>
    from ..signal import Signal, NamedSignalValue
  File "/usr/local/lib/python3.8/dist-packages/cantools/database/can/signal.py", line 5, in <module>
    from cantools.typechecking import Comments
  File "/usr/local/lib/python3.8/dist-packages/cantools/typechecking.py", line 3, in <module>
    import typing_extensions
ModuleNotFoundError: No module named 'typing_extensions'
```

**OS:** Ubuntu 20.04.3 (running on Github Actions)
**Version:** 37.0.2 (installed with pip)

juleq, 2022-01-23:
@zariiii9003 Can you fix that and add a test that covers it? Thanks.

juleq, 2022-01-25:
Release is out, please try. Pypi will have it in a few minutes.

closed 2022-01-25 by juleq

Andreasdahlberg, 2022-01-25:
Works fine now with 37.0.3, thanks!

PR #385 (2022-01-25): Fix #384
Closes #384 

Currently the CI installs the dependencies with `pip install -r requirements.txt` and ignores the dependencies in `setup.py`.
This wouldn't have happened if cantools used tox. Tox actually tests the packaged and installed code.
commit bbe72f57 (2022-01-25): fix #384