cantools/cantools #387: Change default back to prune_choices=False?

erzoe, 2022-01-27:
Less than 2 months ago (in version 37.0.0, commit da19f1e12addaa223b48480580429181eca2b13d) a new feature has been added to cantools which abbreviates the names of choices.
This feature can be quite useful but I have some concerns about enabling it by default because:

1. To me it seems very unintuitive that cantools is using different names *by default* than what is stored in the file and what other tools like CANdb++ are displaying. Always needing to remember that names might need to be shortened or explicitly disabling this feature seems error prone.
2. This breaks existing code. For me it was easy enough to just add a `prune_choices=False` because I already stumbled upon this feature in a pull request but if I hadn't known about this it would have cost me some hours of debugging until finding that the "error" is a change in cantools and not our code or the dbc file.

	However, adding a `prune_choices=False` breaks the code for a colleague of mine who is not willing to update cantools because of not backward compatible changes like this.

	Therefore a simple 

	```python
	db = database.load_file(fn_db)
	```

	became a

	```python
	def cantools_version_greater_or_equal(major, minor, patch):
		return tuple(int(v) for v in cantools.__version__.split('.')) >= (major, minor, patch)
	
	kw = {}
	if cantools_version_greater_or_equal(37, 0, 0):
		kw['prune_choices'] = False
	db = database.load_file(fn_db, **kw)
	```

	which seems overly complicated.

Therefore I would like to raise the question whether it would make sense to change the default back to `prune_choices=False`.

simontegelid, 2022-01-27:
Totally agree. Default should be 'false'.

andlaus, 2022-01-27:
I'm not opposed to this either. Since I like the pruned names much better without the visual clutter, and nobody made any noise when I asked this in #372, I set the default to `True`. In retrospect, this was a mistake...

juleq, 2022-01-29:
Released

closed 2022-01-29 by juleq