cantools/cantools #393: Enable Floats in C Source Generation

samc1213, 2022-02-01:
Right now, `generate_c_source` only generates encode and decode functions with `double`s. Some floating point units may only have hardware support for single precision, so it might be nice to enable a flag like `generate_c_source --use-float`, which would use `float` instead of double.

@juleq Does this proposal make sense or do you forsee any issue? I'm happy to implement if you think this would be a worthwhile change.

juleq, 2022-02-01:
Disclaimer: I have not used the feature (yet).

I assume it has been selected to output double because no/little concerns with respect to precision loss should arise for having double all the way.

Since even a lot of modern lower- to mid-end MCUs do have no or just an SP FPU, I think the option would be appreciated. And it also helps when you have strong size constraints.

If you choose to give it a shot, do not forget the f postfix on constants to prevent unwanted double promotion.

samc1213, 2022-02-01:
Sounds good, I will take a stab at it. My MCU just has an SP MCU, and DP operations are implemented in software / with more instructions. Good call on the f postfix, thank you.

timbitss, 2022-02-03:
I was about to make the same issue last week! I created a [fork](https://github.com/timbitss/cantools/commit/a4b3c9390f3c71cd17dd76444a60fb2ac6a4ffb5) just to convert doubles to floats for factors other than 1. A flag option would be nice. Hope to see this resolved soon

samc1213, 2022-02-07:
@timbitss if you have time, please install the version in #394 and see if it works with your project/system. Please let me know your thoughts.

PR #394 (2022-02-12): Add --use-float flag to generate_c_source
* If you add --use-float, any floating point
  decoding/encoding will be done using the
  float type instead of the double type
* This is to be used on ECUs that only
  have a single-precision floating point
  unit
* NOTE: If a database file is using a type
  called Double (as in, the Signal in the DBC
  file specifies Double), then this autogenerated
  code will still use the double type, since there
  is no other valid way to handle this

closes #393
closed 2022-02-12 by juleq