cantools/cantools #395: db.add_arxml_file error

shuyuanw, 2022-02-09:
Got this error when using add_arxml() function:
[image]
Not sure why this happens? Any help is appreciated

andlaus, 2022-02-09:
this is an error raised by the XML parser. Are you sure that your data is valid XML?

andlaus, 2022-02-09:
(judging from your function names, you probably want to use `dbc_db.load_dbc_file("MyFileName.dbc")` not `db.add_arxml_file()')

shuyuanw, 2022-02-09:
Pretty sure the arxml file is valid. It is a requirement for me to work with arxml to decode blf file. Not really a choice for me.

andlaus, 2022-02-09:
I recommend to double check this, e.g., open the file with a web browser or pass it through an XML linter. the error you outlined above is not anything cantools can do about; the XML parser basically complains that the file you gave it is not XML. since this parser is part of the python standard library, it should be *extremely* well tested...

shuyuanw, 2022-02-10:
Searched a little bit, saw that the error is from root = ElementTree.fromstring(string), the string is the filename I parsed in. However, according to https://stackoverflow.com/questions/39861806/error-trying-parsing-xml-using-python-xml-etree-elementtree-parseerror-syntax, this is just parsing the string of filename, instead of the content of the file, should be  ET.fromstring(open('filename.xml').read()). Is this a bug or is this intended?

andlaus, 2022-02-10:
If this is the case, it's a bug and we would be grateful if you could make a PR that fixes this. I just have a hard time understanding this:

according to the stack trace above, you called [`add_arxml_file()`](https://github.com/cantools/cantools/blob/master/cantools/database/can/database.py#L145), which on [line 155](https://github.com/cantools/cantools/blob/master/cantools/database/can/database.py#L155) opens a file and uses this to call [`add_arxml()`](https://github.com/cantools/cantools/blob/master/cantools/database/can/database.py#L137) which just reads all its contents and calls `add_arxml_string()`. I am thus baffled about how the file name passed to `add_arxml_file()` could end up as an argument to `add_arxml_string()`...

shuyuanw, 2022-02-11:
the arxml file is broken, closing the comment

closed 2022-02-11 by shuyuanw