cantools/cantools #401: Multiplexed messages with variable data length

andyinno, 2022-02-22:
Hi!

I am using a dbc file that uses a multiplexed value on some ID.

While using the dbc file I noticed that the datalen of the CAN message changes depending on the MUX it is dealing with.

For example:
Mux 0: message frame is 8 bytes of payload
Mux 1: message frame is 2 bytes of payload

while printing the values from candump and piping them into cantools as the readme example cantools complains with "Short data" also if I defined that Mux1 does not have any valid data after the frame length.

Am I missing something or there is something that I can do for allow cantools to interpret that frame too?

thank you in advance

andlaus, 2022-02-22:
> Am I missing something or there is something that I can do for allow cantools to interpret that frame too?

I guess that it is most likely a bug in cantools' DBC parser and/or your DBC file is slightly incorrect. If the latter is the case, you should IMO get a warning. @juleq what do you think?

@andyinno: can you provide us with a stripped down example to reproduce this issue and which we could use to make a unit test?

juleq, 2022-02-22:
Have not looked into it but since the dbc file format uses only one length attribute for a message, be it muxed or not, cantools is likely wrong to reduce the length to payload. Strange. I guess it must be in decode since the parsing and storing of the length attribute should be straightforward.

andyinno, 2022-02-22:
Hi!
thank you for your fast reply.

The DBC file, I'm writing it by myself.
Here's the data that I am trying to decode and to add to the DBC file.

As you can see, the payload is varying, but I need to set a length in the DBC file. If I set the minimum length, then the editor highlights the exceeding data. If I set it to 8, then I can add the different data to the different muxes.
From the excerpt here below I defined two muxes in the same message. 

First mux on bits 6,7 
There I observed possible values as 80, C0, 33, 43.
80 and C0 are a ISP-TP frame part.
second mux in the frame is on bytes 33 => C4, CD, CE, CF,  sends different values of different sensors D2, D5, D6, D7 too.
```
  can0  F10   [8]  80 08 33 C3 38 07 E7 F9
  can0  F10   [5]  C0 80 00 00 60
  can0  F10   [3]  33 C4 0A
  can0  F10   [3]  33 CD 10
  can0  F10   [5]  33 CE 77 00 50
  can0  F10   [3]  33 CF 00
  can0  F10   [8]  80 07 33 D0 73 00 73 00
  can0  F10   [4]  C0 73 00 00
  can0  F10   [5]  33 D2 01 AA 00
  can0  F10   [8]  33 D5 00 00 02 00 00 00
  can0  F10   [8]  33 D6 90 01 00 00 00 00
  can0  F10   [7]  33 D7 94 8A 09 00 00
```

andlaus, 2022-02-22:
it seems like you got something a bit confused, because all the can frames above are for id 0xf10. are you trying to do diagnosis over ISO-TP or do you want to do the standard (for automotive) signal based communication? As far as I know, both worlds cannot really be mixed  (i.e., they need to use frames exhibiting disjoint arbitration IDs) and they use a drastically different communication model (signals with LRU semantics vs. datagrams)

andlaus, 2022-02-22:
(cantools is basically only good for the signals based communication model.)

andyinno, 2022-02-22:
Hi again...

Well... Canbus frames are not mine.

I'm just trying to decode them.
While decoding them, I am adding them to a dbc file for easy access.

andlaus, 2022-02-22:
if you are actually trying to decode a conversation over ISO-TP, you should use an ISO-TP module for python (for fairly recent linux systems, I can recommend [`can-isotp`](https://pypi.org/project/can-isotp/)), whilst for decoding the payload of the telegrams, you need to either write custom code or use something like the [`obd`](https://pypi.org/project/obd/) module (for plain OBD-II), or something like [`odxtools`](https://github.com/mercedes-benz/odxtools) if you need to do UDS and have created your diagnostic description files using a tool like CANdela studio...

andyinno, 2022-02-22:
Thank you for the links.

Actually, please ignore my comment about the iso-tp frames.

Messages that starts with 0x33 in byte0, has byte1 as mux.

What I want to do is read those sensors using the c code generated by cantools and it works pretty well.
What I can't do is

candump vcan0 | python3 -m cantools decode --single-line tests/files/dbc/motohawk.dbc

Some 0x33 frames has lenght 8 byes others are shorter. 

The decode function outputs "Short data" instead of the value that the bytes contains. 

The generated c functions can parse correctly those messages

andlaus, 2022-02-22:
maybe it is possible to set a `strict` parameter to `False` in your case. I'm not familiar with the C code generator, but in the case of `Message.decode()` it might help...

juleq, 2022-02-22:
The sensor should pad with 0x00 or 0xff if a multiplex variant of the base message is shorter. But since you are likely not in control of that, trying @andlaus proposal is your best option. This should be just a sanity check of the received data length vs. the expected.

If strict does not get you around that, you could consider issuing a PR that makes the aforementioned sanity check conditional to strict.