cantools/cantools #407: Variable DLC causes "ValueError: Short data" in db.decode()

frysyz, 2022-03-09:
Thanks for the awesome tool!

Certain messages have variable DLCs based on how much information is available, which differ from the DLC in the DBC file. It looks like this property is not handled in the db.decode() function. Everything is fine if the actual DLC is higher than the DBC DLC, but if the actual DLC is lower, the below error trace happens (tested with DBC-DLC: 2, actual-DLC: 1, also with DBC-DLC: 22, actual-DLC: 8).

Ideally, the return key-value dict would only include the signals that are decode-able from the actual data, but filling the unavailable data with their maximum decodable value to represent an error would also work if all the keys need to be present.

...
  File "/usr/local/lib/python3.7/dist-packages/cantools/database/can/database.py", line 469, in decode_message
    return message.decode(data, decode_choices, scaling)
  File "/usr/local/lib/python3.7/dist-packages/cantools/database/can/message.py", line 853, in decode
    return self._decode(self._codecs, data, decode_choices, scaling)
  File "/usr/local/lib/python3.7/dist-packages/cantools/database/can/message.py", line 736, in _decode
    scaling)
  File "/usr/local/lib/python3.7/dist-packages/cantools/database/utils.py", line 107, in decode_data
    unpacked = formats.big_endian.unpack(bytes(data))
ValueError: Short data.

samc1213, 2022-03-10:
@frysyz - Are you able to add a test case for this?

For example, there's already a test for DLC of 0. Note a simple DBC file was added here:

https://github.com/cantools/cantools/blob/c70aebf898c4d4dcbd95ebe57710d2ab1a43f46a/tests/files/dbc/message-dlc-zero.dbc#L1-L12

And then a test was added here, asserting the encoding and decoding works as expected:

https://github.com/cantools/cantools/blob/c70aebf898c4d4dcbd95ebe57710d2ab1a43f46a/tests/test_database.py#L5504-L5508

If you can add a test with a specific DBC file and desired functionality of decode, that helps ensure any fix covers your use case. 

I'm not sure how this should be fixed, I think you would need to update this function:

https://github.com/cantools/cantools/blob/c70aebf898c4d4dcbd95ebe57710d2ab1a43f46a/cantools/database/utils.py#L119

This could be a significant change from what I can tell, but would love someone who is more familiar with the `bistruct` module and how this part of cantools works to opine.

andlaus, 2022-03-10:
> This could be a significant change from what I can tell, but would love someone who is more familiar with the bistruct module and how this part of cantools works to opine.

I don't think that variable-length DLCs are intended with normal signal frames (in contrast, container frames only make sense with a variable DLC), but I agree that having a "non-strict decoding" feature would be quite useful because sometimes ECUs might save a few bytes in certain scenarios (e.g. if multiplexer frames are involved). I'll have a look, but it might take a while...

frysyz, 2022-03-12:
@andlaus One example where this happens is in J1939, if an ECU has no active trouble codes, it can send a regular 8 byte frame of the diagnostic message. But if there are multiple trouble codes, it will send the diagnostic message with >8 bytes in multiple frames, using J1939 transport protocol. Since the DBC file usually covers the >8 byte message, the db.decode() function crashes when it sees the single frame version of the message.

@samc1213 Info below. Running the test cases, I get the same short data error only when the DBC length is higher than the actual data. For a janky workaround, I have been checking the length of the message and padding out the raw bytes to match the DBC length before decoding.
`db.get_message_by_name('TestMessage1').length`
This produces fake data rather than the desired behavior of only returning the decode-able bytes. I'll have a closer look at the internal functions to see if there's a cleaner way to do it.

DBC:
```
VERSION ""
BS_:
BU_: Test

BO_ 2164195584 TestMessage1: 3 Test
 SG_ Signal1 : 0|8@1+ (1,0) [0|0] ""  Test
 SG_ Signal2 : 8|8@1+ (1,0) [0|0] ""  Test
 SG_ Signal3 : 16|8@1+ (1,0) [0|0] ""  Test

BO_ 2164195840 TestMessage2: 10 Test
 SG_ Signal4 : 0|15@1+ (1,0) [0|0] ""  Test
 SG_ Signal5 : 15|15@1+ (1,0) [0|0] ""  Test
 SG_ Signal6 : 30|15@1+ (1,0) [0|0] ""  Test
 SG_ Signal7 : 45|15@1+ (1,0) [0|0] ""  Test
 SG_ Signal8 : 60|15@1+ (1,0) [0|0] ""  Test

BA_DEF_ BO_  "VFrameFormat" INT 0 3;
BA_DEF_DEF_  "VFrameFormat" 3;
```

New test cases:
```
    def test_decode_dlc_variable(self):
        db = cantools.database.load_file('tests/files/dbc/variable_dlc.dbc')
        decoded_message1_full = {
            'Signal1': 1,
            'Signal2': 2,
            'Signal3': 3,
        }
        decoded_message1_limited = {
            'Signal1': 1,
            'Signal2': 2,
        }
        decoded_message2_full = {
            'Signal4': 4,
            'Signal5': 5,
            'Signal6': 6,
            'Signal7': 7,
            'Signal8': 8,
        }
        decoded_message2_limited = {
            'Signal4': 4,
            'Signal5': 5,
            'Signal6': 6,
            'Signal7': 7,
        }
        #message 1 (dlc 3 match, dbc lower, dbc higher)
        self.assertEqual(db.decode_message('TestMessage1', b'\x01\x02\x03'), decoded_message1_full)
        self.assertEqual(db.decode_message('TestMessage1', b'\x01\x02\x03\x04'), decoded_message1_full)
        self.assertEqual(db.decode_message('TestMessage1', b'\x01\x02'), decoded_message1_limited)
        #message 2 (dlc 10 match, dbc lower, dbc higher)
        self.assertEqual(db.decode_message('TestMessage2', b'\x04\x80\x02\x80\x01\xe0\x00\x80\x00\xf8'), decoded_message2_full)
        self.assertEqual(db.decode_message('TestMessage2', b'\x04\x80\x02\x80\x01\xe0\x00\x80\x00\xf8\x01'), decoded_message2_full)
        self.assertEqual(db.decode_message('TestMessage2', b'\x04\x80\x02\x80\x01\xe0\x00\x80\x00'), decoded_message2_limited)
```

juleq, 2022-03-15:
@frysyz I think the transport layer of the J1939 device you are interacting with should always use a DLC of 8 and pad with 0xff to fill it up. If you do not have the SAE J1939/21 document, I can recommend [this Autosar spec](https://www.autosar.org/fileadmin/user_upload/standards/classic/4-3/AUTOSAR_SWS_SAEJ1939TransportLayer.pdf#page23) for details.

That said, cantools does not support J1939. If you require more complex protocol decoders, there might be alternatives available elsewhere.

But maybe @andlaus proposal would be sufficient to allow for some "DM1 light" support?

frysyz, 2022-03-30:
@andlaus @samc1213 Please let me know if I should create a pull request for the new test cases and dbc. Also let me know if the workaround I posted above is too janky, I can spend some more time looking for something cleaner.

andlaus, 2022-04-01:
I hope to get the `bitstruct`/`bitstruct.c` stuff required for this feature ready by the end of next week. I suppose your code above is a bit too hacky to include something like it in the master branch, but I suppose that I can possibly make a unit test  based on it.

frysyz, 2022-04-01:
@andlaus thanks again for looking into this. Feel free to use the DBC and test cases I posted above and let me know if I can help.

closed 2022-05-31 by juleq