cantools/cantools #412: Overlapping signals when changing definition to big endianness

TomaszBorczyk, 2022-03-24:
Hi,

today I switched my dbc frame definition from little endian to big endian (previously I thought `1+` meant big endian, as official documentation says so, but I learned I was using little endian all the time)

So I modified dbc definitions to use `0_+` instead `1+`, just replacing the entries.

from:
```
BO_ 176 Values1: 8 XXX
  SG_ Value_0 : 0|12@1+ (1,0) [0|4095] " " Vector_XXX
  SG_ Value_1 : 12|12@1+ (1,0) [0|4095] " " Vector_XXX
```

it was changed to:
```
BO_ 176 Values1: 8 XXX
  SG_ Value_0 : 0|12@0+ (1,0) [0|4095] " " Vector_XXX
  SG_ Value_1 : 12|12@0+ (1,0) [0|4095] " " Vector_XXX
```
I load it via script:
```
import cantools

if __name__ == '__main__':
    filename = "./dbc_frame_example.dbc"  # path might need to be modified to absolute depending how you run the script
    cantools.database.load_file(filename)
```

With little endianness (`1+`) it works without error, but with big endianness (`0+`) it throws:
`cantools.database.errors.Error: The signals Value_1 and Value_0 are overlapping in message Values1.`

Could anyone help me understand this message? I don't see how changing endianness of signal makes them overlapping - isn't endianness just indicating bit order relevance?

andlaus, 2022-03-24:
DBC -- and by extension cantools -- uses a (IMO weird) "sawtooth" bit numbering, cf. https://github.com/cantools/cantools/blob/master/cantools/database/can/signal.py#L154

juleq, 2022-03-24:
@TomaszBorczyk That can be tough to get right in a text editor (byte order, bit numbering and LSB or MSB as reference for signal start has to be taken into account). You have a few options for tool support, e.g.:

- Write a python script that imports cantools and programmatically set up your messages and signals, then dump it into a dbc
- Install a Vector CANalyzer demo version. I think that ships with CANdb++, which is the Vector tool to create dbc files

Please tell us what worked for you.

Best regards

TomaszBorczyk, 2022-03-24:
@andlaus 
I've just read and analysed that bit numbering chart that you've sent, and my brain hurts a bit...

so if I have two signals, big endianness and 12 bit each, should they look like this?
```
BO_ 176 Values1: 8 XXX
  SG_ Value_0 : 7|12@0+ (1,0) [0|4095] " " Vector_XXX
  SG_ Value_1 : 12|12@0+ (1,0) [0|4095] " " Vector_XXX
```
This does not seem correct to me

TomaszBorczyk, 2022-03-24:
@juleq I will try to do it programatically, thanks

andlaus, 2022-03-24:
> so if I have two signals, big endianness and 12 bit each, should they look like this?

Unfortunately, I think so. The reason for the sawtooth numbering is probably that can sends the most significant bit of each byte first on the wire. if you now think about where a bit ends up if you shift it over a byte boundary, you get the bit ordering madness that is DBC (probably)...

TomaszBorczyk, 2022-03-28:
@juleq 
So I've tried generating the dbc file programatically, and I am confused.
This is minimal working example that will describe my problem:
```
import cantools

def generate_frame_definition() -> None:
    endianness = 'big_endian'

    example_value_0 = cantools.database.can.Signal("ExampleValue_0", 12, 12, endianness)
    example_value_1 = cantools.database.can.Signal("ExampleValue_1", 16, 12, endianness)

    example_values_1 = cantools.database.can.Message(
        0x001, "ExampleValue1", 8, [
            example_value_0,
            example_value_1,
        ]
    )
    
    db = cantools.database.can.Database([
        example_values_1
    ])
    cantools.database.dump_file(db, "generated_frame.dbc")

if __name__ == '__main__':
    generate_frame_definition()

```
I am using interactive editor from csselectronics for cross validation: https://docs.google.com/spreadsheets/d/1X6VQBkqRX0a6QLd4oFbAos_ps3zrpVc86CVGd67DoSs/edit (taken from https://www.csselectronics.com/pages/can-dbc-file-database-intro)

Using big endianess, I am having contradicting result between `cantools` generated frame definition and above spreadsheet.
For big endianess, we have kind of 'reversed' counting. When I define my signals as:
```
    example_value_0 = cantools.database.can.Signal("ExampleValue_0", 12, 12, endianness)
    example_value_1 = cantools.database.can.Signal("ExampleValue_1", 16, 12, endianness)
```
the output in DBC file is:
```
BO_ 1 ExampleValue1: 8 Vector__XXX
 SG_ ExampleValue_1 : 16|12@0+ (1,0) [0|0] "" Vector__XXX
 SG_ ExampleValue_0 : 12|12@0+ (1,0) [0|0] "" Vector__XXX
```
No operations are made on starting bits, they are taken `as is`

Though when I compare it to the tool that I posted above, definitions are completely different (and more logical, since counting is reversed):
[image]

What is correct here?

TomaszBorczyk, 2022-03-28:
Also, two different approaches generate without error:
1. when I treat starting bit as LSB:
```
    example_value_0 = cantools.database.can.Signal("ExampleValue_0", 12, 12, endianness)
    example_value_1 = cantools.database.can.Signal("ExampleValue_1", 16, 12, endianness)
    example_value_2 = cantools.database.can.Signal("ExampleValue_2", 36, 12, endianness)
```
2. when I treat starting bit as MSB:
```
    example_value_0 = cantools.database.can.Signal("ExampleValue_0", 7, 12, endianness)
    example_value_1 = cantools.database.can.Signal("ExampleValue_1", 11, 12, endianness)
    example_value_2 = cantools.database.can.Signal("ExampleValue_2", 31, 12, endianness)
```
For big endian, starting bit should be MSB, is that correct in `cantools` ?

andlaus, 2022-03-28:
I  would be a bit skeptical about the correctness of third-party tools such as the one which you used: As @juleq mentioned, the authoritative source for the DBC file format is Vector's candb++ (so much so that their bugs become "the standard", i.e. it does not matter what's logical and what's not...)

juleq, 2022-03-29:
I have little time to look into this. These are the default settings as found in CANdb++. 

[image]

You can also get CANdb++ by downloading the Vector Drivers disk. It also contains the CANdb++ setup files.

TomaszBorczyk, 2022-04-03:
> the authoritative source for the DBC file format is Vector's candb++ (so much so that their bugs become "the standard", i.e. it does not matter what's logical and what's not...)

Seeing how convoluted this standard is, and seeing comments like this, I think I'd rather stick to old parsers written by my team internally than using a tool that we don't trust. The issue would be scaling and generalization, but until then we might come to some solutions regarding dbc format.

Thanks for all the support, I will verify my results with CANdb++ once I have some time

andlaus, 2022-04-03:
> Seeing how convoluted this standard is, and seeing comments like this, I think I'd rather stick to old parsers written by my team internally than using a tool that we don't trust.

Sure. Be aware that in this case the format which you are going to use most likely is "DBC like", not DBC...

andlaus, 2022-04-19:
This issue does not be caused or fixable by cantools. Let's close it for now...

closed 2022-04-19 by andlaus