cantools/cantools #435: Divide by zero if signal scale is 0

martin-ejdestig, 2022-05-30:
Currently working in a project that loads a database that seems to have signals with scale set to 0. I have no idea if this is allowed or not or if the database is erroneous, but it results in a divide by zero when trying to use it:

```
  ...
  File "<libpath>/python3.8/site-packages/cantools/database/can/message.py", line 922, in encode
    self._encode(self._codecs, data, scaling)
  File "<libpath>/python3.8/site-packages/cantools/database/can/message.py", line 760, in _encode
    encoded = encode_data(data,
  File "<libpath>/python3.8/site-packages/cantools/database/utils.py", line 89, in encode_data
    unpacked = {
  File "<libpath>/python3.8/site-packages/cantools/database/utils.py", line 90, in <dictcomp>
    field.name: _encode_field(field, data, scaling)
  File "<libpath>/python3.8/site-packages/cantools/database/utils.py", line 58, in _encode_field
    value = (Decimal(value) - Decimal(field.offset)) / Decimal(field.scale)
decimal.DivisionByZero: [<class 'decimal.DivisionByZero'>]
```

Worked around by checking if scale is 0 and setting it to 1 after loading the database at the moment. I have no idea if this may cause some unwanted side effects.

andlaus, 2022-05-30:
> I have no idea if this is allowed or not or if the database is erroneous, but it results in a divide by zero when trying to use it:

I'm not sure either. If a scale of zero is technically allowed, it is certainly against the spirit of what this attribute is supposed to do. I suppose that your fix is sound and I'm not aware of any adverse side effects...

juleq, 2022-05-30:
CANdb++ will not allow me to enter a factor of 0 (Invalid factor). Since that program is authoritative for the dbc format, I would trust that :). I would also fix it to 1 in your case and inform the supplier of this dbc.

juleq, 2022-05-31:
I will close for now. If there is someone to take the time to do validity and consistency checks that provide more concise information on the issue found during loading, they could pick this up, too.

closed 2022-05-31 by juleq