cantools/cantools #440: Question marks in encoded message bytearray

melav99, 2022-06-08:
**Background:**
I am sending CAN messages from a power module to a controller. 

**Problem:**
I want to send an encoded message and then decode it to be used in another function. However, it seems to encode my message incorrectly. 

**Input:**
```
#!/usr/bin/env python

import can
import cantools

db = cantools.database.load_file('CAN_v2.2.kcd')

def send_one():
    """Sends a single message."""

    with can.interface.Bus() as bus:
        bus = can.interface.Bus(bustype='socketcan', channel='can0', bitrate=500000)
        
        Power_Modules_Status_data = db.encode_message(
            'Power_Modules_Status', {
            'Present_Voltage' :  350.0, 'Present_Current' : 70.0, 
            'Power_Modules_Temperature' : 35.0, 'Enclosure_Temperature' : 35.0, 
            'System_Enable' : 'Not_Allowed', 'Insulation_Resistance' : 480.0
            }, 
            padding = True ,
            strict = True
        )

        msg = can.Message(
            arbitration_id=0x1060010, data=Power_Modules_Status_data, is_extended_id=True
        )

        try:
            bus.send(msg)
            print(Power_Modules_Status_data)
            print(db.decode_message(0x1060010, Power_Modules_Status_data))

        except can.CanError:
            print("Message NOT sent")

if __name__ == "__main__":
    send_one()
```

**Output:**

Data from PCAN-View: 
ID=17170448x, Type=D, Length=8, Data=AC 0D BC 02 4B 4B 00 F0

```
b'\xac\r\xbc\x02KK\x00\xf0'

Traceback (most recent call last):
  File "/usr/lib/python3.9/site-packages/cantools/database/can/database.py", line 385, in decode_message
KeyError: 17170448

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/root/send_one.py", line 42, in <module>
    send_one()
  File "/root/send_one.py", line 36, in send_one
    print(db.decode_message(0x1060010, Power_Modules_Status_data))
  File "/usr/lib/python3.9/site-packages/cantools/database/can/database.py", line 387, in decode_message
KeyError: 17170448

```

**CAN_v2.2.kcd:**

        <Message id="0x0060010" length="8" name="Power_Modules_Status" interval="100" format="extended">
            <Notes><![CDATA[
                Periodic message reporting the current status of power modules. This message should
                be sent all the time when power modules are running.

                [IMPORTANT]
                ====
                Controller implements a timeout on the reception of this message. It is used as a
                source of interlock condition. As such, this interlock source is evaluated only in
                states where power modules should be alive.

                Therefore, it is NOT evaluated when waiting for a PEV to plug-in, or when
                negotiating the connection with the PEV.

                Power modules should be alive from the moment <<New_Charge_Session>> is sent, until
                <<Charge_Session_Finished>> is sent. At any other time, power modules are allowed to
                sleep and not send this message.
                ====
            ]]></Notes>

            <Producer>
                <NodeRef id="6"/>
            </Producer>

            <Signal name="Present_Voltage" offset="0" length="16">
                <Notes>Voltage measurement at the output of the charger</Notes>
                <Value type="unsigned" slope="0.1" unit="Volts" />
            </Signal>

            <Signal name="Present_Current" offset="16" length="16">
                <Notes>Current measurement at the output of the charger</Notes>
                <Value type="signed" slope="0.1" unit="Amps" />
            </Signal>

            <Signal name="Power_Modules_Temperature" offset="32" length="8">
                <Notes>Top temperature sensed in power modules.</Notes>
                <Value type="unsigned" slope="1" intercept="-40" min="-40" max="215" unit="°C"/>
            </Signal>

            <Signal name="Enclosure_Temperature" offset="40" length="8">
                <Notes>Top temperature sensed in the enclosure.</Notes>
                <Value type="unsigned" slope="1" intercept="-40" min="-40" max="215" unit="°C"/>
            </Signal>

            <Signal name="System_Enable" offset="48" length="8">
                <Notes><![CDATA[
                        Tells if charging is allowed.

                        When <<New_Charge_Session>> is sent, the controller waits for this signal
                        to be 1 to continue the charge sequence.

                        If 0 is emitted between <<New_Charge_Session>> and before the charging loop
                        starts, the charge controller tells the vehicle the charging service is
                        unavailable.

                        If 0 is emitted during the charging loop, this is interpreted as an
                        emergency shutdown requested by power modules.
                ]]></Notes>
                <LabelSet>
                    <Label name="Not_Allowed" value="0" />
                    <Label name="Allowed" value="1" />
                </LabelSet>
            </Signal>

            <Signal name="Insulation_Resistance" offset="56" length="8">
                <Notes>
                    Reports the current insulation resistance, in multiple of 2 kOhms.

                    TIP: If the RCD sensor only gives a boolean value, then 255 correspond to a __Valid__
                    insulation, and 0 to an __Invalid__ insulation.

                    IMPORTANT: Whenever a power function is used after the insulation test, if this
                    signal reports an insulation resistance below the limit of 100 Ohms/V, then this
                    is considered as an interlock condition and an emergency stop is carried out.
                </Notes>
                <Value type="unsigned" slope="2" unit="kOhms" />
            </Signal>
        </Message>

zariiii9003, 2022-06-08:
The message id in the database is 0x0060010, you are using 0x1060010 in your script.

Also, this:
```python3
    with can.interface.Bus() as bus:
        bus = can.interface.Bus(bustype='socketcan', channel='can0', bitrate=500000)
```
does not make any sense.
You are using the context manager, which calls `bus.shutdown()` on exit. But then you create a new bus object. The context manager has no effect on the new object.

melav99, 2022-07-25:
Thank you! I somehow missed the message id typo. The above function now works but once I alter it to read, decode, encode and send signals periodically, I get an error. 

The function is now: 
``` python
import cantools
import can

bus = can.interface.Bus(channel='can0', bustype='socketcan', bitrate='500000')
db = cantools.database.load_file('CAN_v2.2.kcd')

def Power_Modules_Status():
    """          Power_Modules_Status         """
    Current_Out = None
    Voltage_Out = None
    Temp_Rectifier_Out = None
    Temp_Bridge_Out = None
    Temp_Transformer_Out = None
    Max_Temp_Out = None
    Insulation_Resistance_Out = None
    arbitration_ids = [0x38004, 0x38005, 0x3800a]

    for LLC in bus:  
        db.refresh()
        LLC_decoded = db.decode_message(LLC.arbitration_id, LLC.data)
        
        for arbitration_id in arbitration_ids:          
            
            # extract the Current_Out and store to encode
            Current = LLC_decoded.get('Current_Out_3')
            if Current is not None and Current <= 75 and len(str(Current)) < 6:
                Current_Out = Current
            
            # extract the Voltage_Out and store to encode               
            Voltage = LLC_decoded.get('Voltage_Out_3')
            if Voltage is not None and Voltage <= 750 and len(str(Voltage)) < 6:
                Voltage_Out = Voltage
            
            # extract separate temperatures of the LLC components 
            Temp_Rectifier = LLC_decoded.get('Temp_Rectifier_3')
            if Temp_Rectifier is not None :
                Temp_Rectifier_Out = Temp_Rectifier
                
            
            Temp_Bridge = LLC_decoded.get('Temp_Bridge_3')
            if Temp_Bridge is not None :
                Temp_Bridge_Out = Temp_Bridge
                           
            Temp_Transformer = LLC_decoded.get('Temp_Transformer_3')
            if Temp_Transformer is not None :
                Temp_Transformer_Out = Temp_Transformer
                     
            # extract the Insulation_Resistance of the LLC and store to encode 
            Insulation_Resistance = LLC_decoded.get('InsulationResistance_3')
            if Insulation_Resistance is not None and len(str(Insulation_Resistance)) < 6:
                Insulation_Resistance_Out = Insulation_Resistance
                
            if Temp_Bridge_Out and Temp_Rectifier_Out and Temp_Transformer_Out:
                Max_Temp_Out = max([Temp_Bridge_Out, Temp_Rectifier_Out, Temp_Transformer_Out])

            Power_Modules_Status_Out = [Current_Out, Voltage_Out, Max_Temp_Out, Insulation_Resistance_Out]

            Power_Modules_Status_data = db.encode_message(
                'Power_Modules_Status', {
                'Present_Voltage' :  Power_Modules_Status_Out[0], 'Present_Current' : Power_Modules_Status_Out[1], 
                'Power_Modules_Temperature' : Power_Modules_Status_Out[2], 'Enclosure_Temperature' : Power_Modules_Status_Out[2], 
                'System_Enable' : 'Not_Allowed', 'Insulation_Resistance' : Power_Modules_Status_Out[3]
                }
            )

            Power_Modules_Status = can.Message(arbitration_id = 0x1060010, data = Power_Modules_Status_data, is_extended_id = True)
            
            msg = can.Message(
                arbitration_id=0x1060010, data=Power_Modules_Status_data, is_extended_id=True
            )

            try:
                bus.send(msg)
                print(Power_Modules_Status_data)
                print(db.decode_message(0x1060010, Power_Modules_Status_data))

            except can.CanError:
                print("Message NOT sent")

if __name__ == "__main__":
    Power_Modules_Status()
```

But now I get this error:
```
# python3 test.py
Traceback (most recent call last):
  File "/root/test.py", line 86, in <module>
    Power_Modules_Status()
  File "/root/test.py", line 62, in Power_Modules_Status
    Power_Modules_Status_data = db.encode_message(
  File "/usr/lib/python3.9/site-packages/cantools/database/can/database.py", line 431, in encode_message
    return message.encode(data, scaling, padding, strict)
  File "/usr/lib/python3.9/site-packages/cantools/database/can/message.py", line 914, in encode
    self.assert_signals_encodable(data, scaling=scaling)
  File "/usr/lib/python3.9/site-packages/cantools/database/can/message.py", line 631, in assert_signals_encodable
    used_signals = self.gather_signals(input_data)
  File "/usr/lib/python3.9/site-packages/cantools/database/can/message.py", line 519, in gather_signals
    raise EncodeError(f'The signal "{signal.name}" is '
cantools.database.errors.EncodeError: The signal "Present_Voltage" is required for encoding.
```

closed 2022-08-02 by melav99