cantools/cantools #445: ARXML problem: decode data are not what expected.

linpanusst, 2022-06-24:
Hi, Miss or Mr.
I have a question when I decode the can files via "arxml". Here are the descriptions.
[image]
Please see this picture. The decoded values which are **False** and **Disabled**. They should be uint8 or uint16 values. Eg. They should be 0, 1 or 2...
This picture is the definition of one signal.
[image]
It seams that the code put the "Description value" in the value column.
Could you please help me to fix this problem?
If you need more description or pictures. Please contact my e-mail of linpan_usst@163.com
Looking forward to hearing from you.

andlaus, 2022-06-24:
I suppose that you need to ask the provider of the tool which you are using; cantools seems to be completely unrelated to it. I'm closing the issue for this reason. Feel free to reopen it if I made a mistake...

andlaus, 2022-06-24:
(I can have a closer look if you provide a sample data file.)

closed 2022-06-24 by andlaus

linpanusst, 2022-06-25:
Hi, I am glad to provide sample data. Could I send such data to this address: [github@poware.org](mailto:github@poware.org)? 
Or could you please send me hello via the e-mail address linpan_usst@163.com? Then I can direct send you sample data.

linpanusst, 2022-06-25:
To provide clean sample data. I need about 3 days. Thank you so much.

andlaus, 2022-06-25:
> Hi, I am glad to provide sample data. Could I send such data to this address: [github@poware.org](mailto:github@poware.org)?

yes, that would work. Alternatively, you can create a [gist](https://docs.github.com/en/get-started/writing-on-github/editing-and-sharing-content-with-gists/creating-gists) and post the link to it here...

> To provide clean sample data. I need about 3 days. Thank you so much.

don't worry, I'll probably need a few days to inspect it thereafter anyway...

linpanusst, 2022-06-27:
Hi, I have sent you the test data and its related arxml format database. Thank you for your time.

linpanusst, 2022-06-29:
Hi, here is the gist link. However, the test data which is xxx.blf can not be uploaded. I have sent you via e-mail [github@poware.org](mailto:github@poware.org). Please check.
On the gist, it is my python can decode code. Its ugly.

andlaus, 2022-06-29:
thanks for the data. I'll have a look, but since I'm pretty busy at the moment it might take a while before I can have a look...

linpanusst, 2022-06-30:
Ok, thanks a lot. Feel free to e-mail if you need more information.

Lit-fish-auto, 2022-10-11:
> Ok, thanks a lot. Feel free to e-mail if you need more information. [testdata.zip](https://github.com/cantools/cantools/files/9017920/testdata.zip)

I use the following code：

db = cantools.database.load_file('abc.arxml')
all_message = db.messages
print(al_lmessage)

The above code reads frame in _arxm_l, and can only read the signal directly hung under message.(message.signal_trees)
In fact, many signals in arxml are hung under the pdu, and these signals cannot be read.
I don't know whether the above is a bug or There is a problem with my way of reading load_file

can you help me?

Lit-fish-auto, 2022-10-11:
Generally, the arxml format of CANFD is   Frame :: Container-PDU :: I-signal-PDU :: Signal
If arxml is in the above format, thenThe method of  load_ file（abc.arxml） cannot read the signal.

Above, how should this form of arxml be read? or  is the bug ？