cantools/cantools #458: SYM Decoder Cannot Handle Leading A-F in Hexadecimal Numbers

samc1213, 2022-07-22:
I have added a unit test [here](https://github.com/RemoraCarbon/cantools/commit/1b1a45364e62391c6cd0fefb7c261dad893180bb) to represent the behavior I would like to implement.

I have found a bug in the SYM parser where, if a hexadecimal number does not begin with a digit (i.e., `0-9`), then the parser breaks. This is due to the token regex here:

https://github.com/cantools/cantools/blob/5b43fd0985acdd5cc7c6243e75d1aef017329519/cantools/database/can/formats/sym.py#L105

That regex requires that a digit (`\d`) lead the number. However, if I update the regex to something simpler, say, `[0-9A-F]?(h)`, then the parser breaks in a variety of other ways. I feel this is a limitation of the way the textparser uses a large regex to tokenize, before matching to the file grammar. Can someone with more experience with the parser suggest a fix please?

samc1213, 2022-07-22:
@eerimoq / @juleq I think I'm going to need some help from you on this. Please let me know your thoughts if you have time to take a look.

juleq, 2022-07-22:
Regexes are my kryptonite :).

Have you tried

-?[\dA-F]+\.?[0-9A-F]*([eE][+-]?\d+)?(h)

Or

-?[0-9A-F]+\.?[0-9A-F]*([eE][+-]?\d+)?(h)

Both should be equivalent. I think your approach might change the number of matching groups or something.

samc1213, 2022-07-22:
Thank you for your quick reply. I implemented both of those, and a unit test is still breaking. I tried something similar to this as well - the problem (I think) is it will detect random characters inside of strings as a HEXNUMBER. For example, the string `Ch` is being treated as a HEXNUMBER I believe, hence the error:

```
======================================================================
ERROR: test_signal_types_6_0_sym (tests.test_database.CanToolsDatabaseTest)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "c:\users\sam\documents\textparser\textparser.py", line 889, in parse
    return Grammar(self.grammar()).parse(tokens, token_tree)
  File "c:\users\sam\documents\textparser\textparser.py", line 684, in parse
    raise GrammarError(tokens.peek_max().offset)
textparser.GrammarError: Invalid syntax at offset 158.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\Sam\Documents\cantools\tests\test_database.py", line 1937, in test_signal_types_6_0_sym
    db = cantools.database.load_file('tests/files/sym/signal-types-6.0.sym')
  File "C:\Users\Sam\Documents\cantools\cantools\database\__init__.py", line 193, in load_file
    return load(fin,
  File "C:\Users\Sam\Documents\cantools\cantools\database\__init__.py", line 287, in load
    return load_string(fp.read(),
  File "C:\Users\Sam\Documents\cantools\cantools\database\__init__.py", line 384, in load_string
    return load_can_database('sym')
  File "C:\Users\Sam\Documents\cantools\cantools\database\__init__.py", line 357, in load_can_database
    db.add_sym_string(string)
  File "C:\Users\Sam\Documents\cantools\cantools\database\can\database.py", line 286, in add_sym_string
    database = sym.load_string(string, self._strict, sort_signals=self._sort_signals)
  File "C:\Users\Sam\Documents\cantools\cantools\database\can\formats\sym.py", line 993, in load_string
    tokens = Parser60().parse(string)
  File "c:\users\sam\documents\textparser\textparser.py", line 891, in parse
    raise ParseError(text, e.offset)
textparser.ParseError: Invalid syntax at line 11, column 10: "Sig=Char >>!<<char /ln:"CharLong""
```

juleq, 2022-07-22:
Ah, ok. It cannot distinguish between string and hex anymore. Requiring at least one whitespace (\n\t\r or space) would be the hacky way out (with ugly limitations). But to to it properly you would have to use some stateful parsing that knows what type to expect after the preceding token. The parser should already do it like that since it has a grammar. Not sure why he seems to be forgetting the token type to expect at that point. Thats probably indeed a question that @eerimoq might want to take. I am still under siege for the foreseeable future and cannot offer to get deeper into that.

juleq, 2022-07-22:
So it tokenizes first and I assume since hex has priority over the catchall string because of its position in the spec list. So this approach would require to sufficiently specify hex to make sure it does not match on arbitrary strings. But I do not see a way that keeps this approach as is while distinguishing between Ah, the hex number and Ah, the string without accepting there might be edge cases.

The hex number in question has a whitespace before and after, while the string has an equal sign before. Maybe requiring the absence of an equal sign works better, but that in turn requires that all other hex numbers do also not have an equal sign before…

This is just how far my understanding goes, I could be wrong.

samc1213, 2022-07-23:
I agree, I think the root issue is that the parser naively tokenizes first using regex, without using the grammar to aid its parsing. And you're right about the priority order (things listed first in the list come first in the large OR regex statement, so they match first). Thanks for thinking about it more, I also came to the conclusion a larger change to the parser might be required to properly fix this. Hoping @eerimoq might have some ideas as well.

eerimoq, 2022-07-27:
Yeah I realized that hexadecimal numbers with a letter first are problematic when I implemented in parser. I simply ignored the issue at that point. I don't have a good suggestion of how to fix it. I guess removing the hexnumber token could work and check for number later, or simply implement the parser using another library or regex.