cantools/cantools #462: C code generation - encode function saturation on overflow

dzid26, 2022-08-06:
I think it would be good for the _encode(...) functions to saturate value to protect against overflow when converting from a floating-point number to an integer. 
I think the saturation shall be according to a number of bits for the signal and not the specified signal range.
I think using saturation should be a default for the generated code.

juleq, 2022-08-07:
Sounds reasonable. That overflow is never the wanted outcome. I think range test is probably best done using some bitwise ops to keep encode fast. PR welcome.

andlaus, 2022-08-07:
I also agree in principle, but I'm not sure if an overflow can actually happen here because when converting physical (float) to internal (integer) values, the physical values need to be "unscaled" *before* the type cast. (I might  miss something, though.)

dzid26, 2022-08-07:
@juleq I think the range test would have to be done while still in float. 
It would have to be compared to 2^bits for unsigned and +/- 2^(bits-1) for signed signals.
For example (signed 8bit signal):
```
int8_t Msg_steering_status_steering_speed_encode(float value)
{
    return (int8_t)(value / 0.015625f);
}
```

Would become something like this:
```
int8_t Msg_steering_status_steering_speed_encode(float value)
{
value = value / 0.015625f;
if (value > 127)
{
   value = 127;  //or return (int8_t) 127;
}
if (value < -128)
{
   value = -128;
}
    return (int8_t)(value);
}
```
If the signal were stored in some weird 7bit, it would have to be compared to 63 and -64.

juleq, 2022-08-07:
Yes, you are right. Not that familiar with the code gen. You naturally can just render out the constants to check against which makes it simple enough. I was thinking about encoding it into a generic int type and then masking it down to target width.

dzid26, 2022-08-07:
@juleq No, because `double` maximum value is a massive 1.79e+308. There would always be a possibility to overflow even int64. Neither would this be efficient. 

Unless I misunderstood.

juleq, 2022-08-07:
@andlaus Can still overflow.

@dzid26 Yes, but not fitting any intermediate integer format is a generic encoding error not specific to the signal. But your proposal is the best way to do it. I’d go with that.

andlaus, 2022-08-07:
> @andlaus Can still overflow.

even if the value of the physical signal is in range [signal.minimum, signal.maximum]?

juleq, 2022-08-07:
@andlaus I see what you are getting at. That depends: 

- Are the min and max correct? Is there code in place to enforce that min/max (phy) do fit the raw data type?
- If they are correct: Is the value being in range (min, max) being enforced?

I have not looked into the generator. @andlaus comment does raise a question though: Maybe the safeguards before scaling and typecasting are sufficient (if in place).

If min and max are checked for validity and encode is aborted for values outside (min, max), overflows do not happen.

andlaus, 2022-08-07:
> Are the min and max correct? Is there code in place to enforce that min/max (phy) do fit the raw data type?

possibly not. one could scale `0` and `(1<<signal.length) -1` to get physical values that always ought to work, though? (if the scaling factor and offsets are not something ridiculous like NaN or Inf...)

dzid26, 2022-08-08:
> > @andlaus Can still overflow.
> 
> even if the value of the physical signal is in range [signal.minimum, signal.maximum]?

C code doesn't have access to these values currently. They are not generated. 
Nor do I think the range values should be used as value limits to prevent overflow. They can be higher than the true signal range.
There is no safeguard currently for the float to int casting. 

There is also a _range() test function being generated but it is expected to be used on integer values (after being converted from float). 

For reference here example of what is available:
```
/**
 * Signals in message STEERING_STATUS.
 *
 * All signal values are as on the CAN bus.
 */
struct Msg_steering_status_t {
    /**
     * Steering speed calculated from motor position sensor
     *
     * Range: -128..127 (-2..1.984375 rev/s)
     * Scale: 0.015625
     * Offset: 0
     */
    int8_t steering_speed;
};

int8_t Msg_steering_status_steering_speed_encode(float value)
{
    return (int8_t)(value / 0.015625f);
}

bool Msg_steering_status_steering_speed_is_in_range(int8_t value)
{
    (void)value;

    return (true);
}
```
_range() test always returns true, when the whole signal range is used. I would say range test in this case shouldn't even be generated, but that's how it is now.

andlaus, 2022-08-08:
I see, I guess this is a workable solution given the limitations of the generated C code.

> _range() test always returns true, when the whole signal range is used.  I would say range test in this case shouldn't even be generated, but that's how it is now.

I agree. That said, I think a better solution would be to make `Msg_steering_status_steering_speed_is_in_range()` take physical (float) values, and let it check either the limits deriving from the length of the signal or the ones given by the database file (whichever is more restrictive).