cantools/cantools #464: Message getting overwritten

Sharmaranjan, 2022-08-10:
Frame info gets overwritten by new messages in database.py file in function _add_message()
My question is when new message comes to get added in respective dictionary , it overwrites the previous one if current frame is already in dictionary, although it is coming from different bus.

Ex - frame 0x134 is coming from bus_name1 and again it is coming from bus_name2 , it overwrites the previous one. Issue with it is , at the end we get info of frame from one bus only using function get_message_by_frame_id().

Issue with this implementation is , after reading any blf file we get msg from different channels and if we don't have signals name from different channels then mapping is one of the issue like channel1 data should be mapped to this signal or channel2 and if any one if them is mapped what about data from other channels.

From canoe or canape tool we get option to map channel1 data to signal from one bus and channel2 data to same signal from other bus.

Is this a bug ? Or is there any way to handle this?

andlaus, 2022-08-10:
> Is this a bug ?

depends on how you see it: the current data model of cantools only really has one bus per database description file in mind. 

> Or is there any way to handle this?

if you have multiple buses you need to load one database file per channel, a la

```python
db1 = cantools.database.load_file('oem1_bus1.dbc')
db2 = cantools.database.load_file('oem2_bus2.kcd')

raw_msg1 = can1.recv()
raw_msg2 = can2.recv()

print(db1.decode_message(raw_msg1.arbitration_id, msg1.data))
print(db2.decode_message(raw_msg2.arbitration_id, msg2.data))
```

Sharmaranjan, 2022-08-11:
Currently i am using single arxml database only, not multiple databases. In any of the arxml database , a signal comes from multiple buses like speed can come from multiple buses but current implementation in cantools returns only one bus i. e last bus from arxml for this signal.

My expectation is i should get list of all buses from which this particular signal has been transmitted. Only then we will be able to map signal values from multiple channels of blf to respective bus for any signal.

Ex - in arxml speed is coming from bus1, bus2, bus3....
But cantools returns only one of them specifically the last one i. e speed is from bus3.

Now from blf i get speed signal from channel1 , channel2, channel3...
Channel1 data should be mapped to this signal or channel2 or any other and even if i map one of them what about other channels data.

Ideally cantools should have returned list of buses from which speed signal is coming and then we will have flexibility to map dara from each channel to any of the bus i. e one channel to one bus and so on.

Note - i am using 1 arxml database and 1 blf file only . I am not considering multiple databases

andlaus, 2022-08-11:
you're right, but this is a feature which nobody seems to have been using or testing yet. If you provide a patch for the ARXML loader, I will gladly review it. (or -- if you provide such a file for public consumption -- I might even fix it for you...)

Sharmaranjan, 2022-08-16:
I changed the code _add_message() function from database.py to something like this
def _add_message(self, message: Message) ->None:
  name_to_msg_flag = True
  id_to_msg_flag = True
  if message.name in self._name_to_message:
     name_to_msg_flag = False
     prev_msg = self._name_to_message[message.name]
     for msg in prev_msg:
         if msg.bus_name == message. bus_name:
             break
         else:
             prev_msg.append(message)
             self._name_to_message[message.name] = prev_msg

     masked_frame_id = (message.frame_id & self._frame_id_mask)

    if masked_frame_id in self._frame_id_to_message:
       prev_msg = self._frame_id_to_message[masked_frame_id]
       id_to_msg_flag = False
       for msg in prev_msg:
           if msg bhs_name == message.bus_name:
              break
           else:
              prev_msg.append(message)
              self._frame_id_to_message[masked_frame_id] = prev_msg

    if name_to_msg_flag:
      self._name_to_message[message.name] = [message]
    if id_to_msg_flag:
       self._frame_id_to_message[masked_frame_id] = [message]

I am able yo get signals from multiple buses now. But wherever signals are extracted from get_message_frame_id () or other function, that needs to be handled accordingly as now info is one level below. Similarly decode function needs to be modified as well. I am using my own logic for decoding so couldn't check the issues.

andlaus, 2022-08-16:
I propose to add a `bus: Union[None, str, Bus]` argument to the `get_message_by_*()` functions and letting it default to `None` meaning "any bus". This approach would keep compatiblity with the current API and as far a I can see, adapting the infrastructural code, parsers and the test suite would be not very hard, but it would be quite a bit of work. (Also -- besides ARXML -- I'm not sure if any of the other file formats support multiple busses anyway. Any insights @juleq?)

juleq, 2022-08-16:
For dbc, there is an Admin variant of CANdb++ that lets you define Networks (Body, Power Train, etc.) which would probably be the bus equivalent.

That said, for me using multiple instances for multiple busses is sufficient. That is how I also work usually with the Vector tooling: Just add a bus with its data base. Usually I do even have multiple dbcs for one bus because nodes of multiple vendors with their own dbc files are being integrated. 

I would have no objection to extending cantools to support "that hierarchy layer up", but I think @andlaus concern about the amount of work is fair: Not only unobtrusively adding it to the existing API, but also testing and a strategy for how to cope with probable lack of support by other file formats.