cantools/cantools #465: AttributeError: module 'can' has no attribute 'Listener'

ayushmankumar7, 2022-08-10:
[image]

juleq, 2022-08-10:
You should probably add the cantools version that you have installed.

ayushmankumar7, 2022-08-10:
@juleq It's the latest on 37.1.0

juleq, 2022-08-10:
Have nothing to test at hand, off for vacation. But appears to be specific to your setup somehow since CI should complain on a generic fault.

Try removing it and reinstall using pip3.

ayushmankumar7, 2022-08-10:
Hey, it worked! 
Thanks !!!

closed 2022-08-10 by ayushmankumar7

juleq, 2022-08-30:
@mjeri You can post your error or confirm that it looks as the one above.

To provide some confidence: I am on an M1, too. I have never had this issue.

It is likely some environmental issue related to system/user installed python or their respective pips.

The 'which' command is your friend to find out to which installation your shell will resolve to.

mjeri, 2022-08-30:
Yes, fully agree that it's most likely something from the environment that breaks this.

To verify things I've also tried to clone and build the project. I'm appending the log here of the `make`. The error message at the bottom is always the same.

A few more details, just to share, after the `python3 -m pip install cantools` (also tried out with `pip3 install cantools`) I have no access to other commands like candump and such. If I `which cantools` it resolves to `/opt/homebrew/bin/cantools` which first made me think I have a duplicate installation via brew but if I `pip uninstall` the cantools then bin is gone from the homebrew directory :3 

```
➜  cantools git:(master) make
python3 setup.py test
running test
WARNING: Testing via this command is deprecated and will be removed in a future version. Users looking for a generic test entry point independent of test runner are encouraged to use tox.
running egg_info
writing cantools.egg-info/PKG-INFO
writing dependency_links to cantools.egg-info/dependency_links.txt
writing entry points to cantools.egg-info/entry_points.txt
writing requirements to cantools.egg-info/requires.txt
writing top-level names to cantools.egg-info/top_level.txt
reading manifest file 'cantools.egg-info/SOURCES.txt'
reading manifest template 'MANIFEST.in'
adding license file 'LICENSE'
writing manifest file 'cantools.egg-info/SOURCES.txt'
running build_ext
Traceback (most recent call last):
  File "/Users/marco/work/cantools/setup.py", line 14, in <module>
    setup(name='cantools',
  File "/opt/homebrew/lib/python3.9/site-packages/setuptools/__init__.py", line 155, in setup
    return distutils.core.setup(**attrs)
  File "/opt/homebrew/lib/python3.9/site-packages/setuptools/_distutils/core.py", line 148, in setup
    return run_commands(dist)
  File "/opt/homebrew/lib/python3.9/site-packages/setuptools/_distutils/core.py", line 163, in run_commands
    dist.run_commands()
  File "/opt/homebrew/lib/python3.9/site-packages/setuptools/_distutils/dist.py", line 967, in run_commands
    self.run_command(cmd)
  File "/opt/homebrew/lib/python3.9/site-packages/setuptools/_distutils/dist.py", line 986, in run_command
    cmd_obj.run()
  File "/opt/homebrew/lib/python3.9/site-packages/setuptools/command/test.py", line 224, in run
    self.run_tests()
  File "/opt/homebrew/lib/python3.9/site-packages/setuptools/command/test.py", line 227, in run_tests
    test = unittest.main(
  File "/opt/homebrew/Cellar/python@3.9/3.9.12_1/Frameworks/Python.framework/Versions/3.9/lib/python3.9/unittest/main.py", line 100, in __init__
    self.parseArgs(argv)
  File "/opt/homebrew/Cellar/python@3.9/3.9.12_1/Frameworks/Python.framework/Versions/3.9/lib/python3.9/unittest/main.py", line 147, in parseArgs
    self.createTests()
  File "/opt/homebrew/Cellar/python@3.9/3.9.12_1/Frameworks/Python.framework/Versions/3.9/lib/python3.9/unittest/main.py", line 158, in createTests
    self.test = self.testLoader.loadTestsFromNames(self.testNames,
  File "/opt/homebrew/Cellar/python@3.9/3.9.12_1/Frameworks/Python.framework/Versions/3.9/lib/python3.9/unittest/loader.py", line 220, in loadTestsFromNames
    suites = [self.loadTestsFromName(name, module) for name in names]
  File "/opt/homebrew/Cellar/python@3.9/3.9.12_1/Frameworks/Python.framework/Versions/3.9/lib/python3.9/unittest/loader.py", line 220, in <listcomp>
    suites = [self.loadTestsFromName(name, module) for name in names]
  File "/opt/homebrew/Cellar/python@3.9/3.9.12_1/Frameworks/Python.framework/Versions/3.9/lib/python3.9/unittest/loader.py", line 191, in loadTestsFromName
    return self.loadTestsFromModule(obj)
  File "/opt/homebrew/lib/python3.9/site-packages/setuptools/command/test.py", line 57, in loadTestsFromModule
    tests.append(self.loadTestsFromName(submodule))
  File "/opt/homebrew/Cellar/python@3.9/3.9.12_1/Frameworks/Python.framework/Versions/3.9/lib/python3.9/unittest/loader.py", line 154, in loadTestsFromName
    module = __import__(module_name)
  File "/Users/marco/work/cantools/tests/test_monitor.py", line 15, in <module>
    from cantools.subparsers.monitor import Monitor
  File "/Users/marco/work/cantools/cantools/__init__.py", line 7, in <module>
    from . import tester
  File "/Users/marco/work/cantools/cantools/tester.py", line 28, in <module>
    class Listener(can.Listener):
AttributeError: module 'can' has no attribute 'Listener'
make: *** [test] Error 1
```

juleq, 2022-08-30:
What version of python-can do you have? Maybe try to uninstall and reinstall after checking its version. I saw that the python-can version is not pinned and has no minimum required version…

mjeri, 2022-08-30:
It's `4.0.0`

I have checked with `pip list` and found a `can` module in version 0.0.0 which I don't yet understand where it's coming from. Could it be related to this one? Removing this one manually gave me now different error messages, so it's an indicator that it's somehow related to this module. Also: a `pip3 install cantools` doesn't add the can package again. Sorry if this is distracting information, I'm not so much into python. Do you have such a `can` module on your system?

The error without the `can` package on my system:
```
➜  work cantools --version
Traceback (most recent call last):
  File "/opt/homebrew/bin/cantools", line 5, in <module>
    from cantools.__init__ import _main
  File "/opt/homebrew/lib/python3.9/site-packages/cantools/__init__.py", line 7, in <module>
    from . import tester
  File "/opt/homebrew/lib/python3.9/site-packages/cantools/tester.py", line 7, in <module>
    import can
ModuleNotFoundError: No module named 'can'
```

juleq, 2022-08-30:
Try removing can, python-can and cantools. Then install cantools using pip3.

mjeri, 2022-08-30:
That seemed to have worked \o/ 

Wow, thank you so much for the support!

juleq, 2022-08-30:
For later reference: There is a pypi package called "can" (color automator or something) which could be installed accidentally by a user (not sure anyone ever really wanted to install that). If installed, it will have a naming collision with the ever so popular python-can package. cantools (this project) uses python-can for CAN bus access. If it finds the "can" package instead, above error will be reported.