cantools/cantools #466: getting a simple DBC file to be interpreted

Todaychris, 2022-08-11:
Have a simple dbc file and small log file i am trying to validate that the dbc was assembled correctly and i must be missing something because I am unable interpret any frames.  Any help or guidance would be terrific.  Alarm_Warning is a dbc file. 
[Alarm_Warning.txt](https://github.com/cantools/cantools/files/9306237/Alarm_Warning.txt)
[8-9-2022_Last_Log.csv](https://github.com/cantools/cantools/files/9306244/8-9-2022_Last_Log.csv)

juleq, 2022-08-11:
On mobile, so just took a quick glance. I think your can ids might be off. Have you checked them?

Example for message Channel_Faults:
- Dec in dbc: 2566859434
- Hex: 0x98FF2AAA
- Hex minus 0x80000000: 0x18FF2AAA

Todaychris, 2022-08-11:
Thanks juleq,  I will try that now and let you know. I originally developed the dbc from a table I was supplied from my work team.   If I still have issues will share the table details.

Todaychris, 2022-08-11:
Thanks juleq,  appears I somehow modified the can ids.  Channel_fault message dec in dbc is 16722264,  hex 0xff2a20.  I originally used Kvaser dbc editor to create.  I developed from the attached few lines on table using PGN dec 65322.  Either way still must have something wrong as I am unable to interpret frames.   I am using savvy v204 which in DBC file manager shows its loaded.  My apologies for providing wrong dbc yesterday  in my question.  DBC is alarm_warning.
[Alarm_Warning.txt](https://github.com/cantools/cantools/files/9312572/Alarm_Warning.txt)
[Alarm_Warning_Table.xlsx](https://github.com/cantools/cantools/files/9312578/Alarm_Warning_Table.xlsx)

[8-9-2022_Last_Log.csv](https://github.com/cantools/cantools/files/9312575/8-9-2022_Last_Log.csv)
.

juleq, 2022-08-12:
There might still be something off with your id construction.

You can try an online calculator to check, like this one (there might be others, too):

[J1939 ID to PGN calculator](https://michael.chtoen.com/convert-j1939-id-to-pgn.php)

closed 2023-01-04 by juleq