cantools/cantools #469: dump re-direct to file is raising an error

Ahelion, 2022-08-19:
Hello,
While dump a dbc file(with the nice format) to file, i get this:
LINES value must be >= 2 and <= 609: got 1
initscr(): Unable to create SP

command: cantools dump d:\mydbc.dbc >file.txt.

BTW: I tried to call the layout _tring function from a script and python said it was not part of the message class..

andlaus, 2022-08-19:
weird: I do this all the time and so far it never failed. I'm on Linux, though, so it might be a Windows specific issue?

does `cantools --debug dump d:\mydbc.dbc > foo.txt` produce a more meaningful error?

Ahelion, 2022-09-17:
> weird: I do this all the time and so far it never failed. I'm on Linux, though, so it might be a Windows specific issue?
> 
> does `cantools --debug dump d:\mydbc.dbc > foo.txt` produce a more meaningful error?

Hello, sorry for taking so long to reply. A picture says more than 1000 words, right?
[image]

andlaus, 2022-09-19:
ok. this seems to be an issue of the ncurses bindings on windows (it seems to terminate the process instead of raising a python exception). luckily the `dump` subparser does not really need `ncurses`, cf #482. can you check if it does the trick for you?

Ahelion, 2022-09-19:
> ok. this seems to be an issue of the ncurses bindings on windows (it seems to terminate the process instead of raising a python exception). luckily the `dump` subparser does not really need `ncurses`, cf #482. can you check if it does the trick for you?

Hello,
I had some issues installing for some reason, I changed the version because I was confused what i was installing, that is why the picture shows 36.4.1.
I cannot confirm that it works.
[image]

Ahelion, 2022-09-19:
> Hello, I had some issues installing for some reason, I changed the version because I was confused what i was installing, that is why the picture shows 36.4.1. I cannot confirm that it works. [image]

Again me.

It works, I pulled the master instead of the branch from the PR.
Sorry.
Again, yes! it works, I managed to dump the dbc to file on windows.

closed 2022-09-19 by Ahelion